realsenseai/realsense-ros #2887: how to get position from the l515

naorwaiss, 2023-10-02:
hi, i got L515 and try to get position from this camera to connect is to my drone 
i try use ros2 but i didn`t find topic that provide me pos data or the imu data at list 
thanks 

```
/opt/ros/foxy/bin/ros2:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import load_entry_point
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/infra/camera_info
/camera/infra/image_raw
/camera/infra/metadata
/parameter_events
/rosout
/tf
/tf_static
```

MartyG-RealSense, 2023-10-02:
Hi @naorwaiss  The IMU topics are disabled by default in the RealSense ROS wrapper.  To enable and publish them, please try the launch instruction below.

`ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=2`

MartyG-RealSense, 2023-10-04:
The IMU topic published when using the unite_imu_method parameter should be **/camera/imu**

Are the separate 'gyro' and 'accel' topics listed?

closed 2023-10-19 by MartyG-RealSense