realsenseai/realsense-ros #2888: Issue with initializing camera name with node name

nbhatt2022, 2023-10-02:
In the recent PR (commit no. #5c1aaaa) for ros2_development branch, the camera_name parameter was removed and now is the same as node name, which is giving problems while launching the robot model in rviz, as we need to provide 'frame_prefix' for the links of the robot in the robot_state_publisher node. In order to vizualize that properly in rviz we need to input the 'tf_prefix' in the robot model which is the same as the frame_prefix but without '/' (the tf_prefix inherently adds a '/' after the input) treating it as a namespace for the links.
ex.
frame_prefix = "modified_0/"
tf_prefix = "modified_0"
(default name of example link) link = "link1"
(after tf_prefix is applied) link = "modified_0/link1"

we were working around this by naming the cameras like- "modified_0/camera", but after this PR the camera name is the same as the node name thus we cannot have a "/" in its name. The way to get around that seems to be either explicitly defining 2 different frames in robot model and camera (camera_link) and publishing a static transform between them 
or 
having a camera name parameter separate from the node name.

let me know if this can be solved in a better way? like adding a frame_prefix parameter similar to the robot_state_publisher node, in realsense ros 2 or something else?

MartyG-RealSense, 2023-10-03:
Hi @nbhatt2022  I will highlight your case to my Intel RealSense colleagues on the ROS team.  Thanks very much for your patience!

Arun-Prasad-V, 2023-10-04:
Hi @nbhatt2022, Thanks for reporting. The recent change on the 'camera_name' has been reverted. Please refer https://github.com/IntelRealSense/realsense-ros/pull/2891.

MartyG-RealSense, 2023-10-11:
Hi @nbhatt2022  Do you require further assistance with this case, please?  Thanks!

nbhatt2022, 2023-10-11:
No, thank you for the assistance. We tested the change and it works fine.

closed 2023-10-11 by nbhatt2022

MartyG-RealSense, 2023-10-12:
Hi @nbhatt2022  It's great to hear that your issue is resolved.  Thanks very much for the update!