realsenseai/realsense-ros #2889: [ros2run]: Process exited with failure 3221226505

aliasghar1721, 2023-10-03:
Hi,

I'm trying to get the ROS2 wrapper for librealsense to work on Windows 11. I'm using ROS2 Humble and RealSense SDK 2.51.1. I am also able to build the realsense-ros wrapper with no errors using colcon. However, when I execute the node, it instantly crashes with the following output:

`[ros2run]: Process exited with failure 3221226505`

To investigate the source of this crash, I debugged the `realsense2_camera_node` project in the `realsense2_camera.sln` and found the following exception:

[image]

This exception is thrown by kernelbase.dll.

I am not sure if this is a realsense-ros or rclcpp issue, or some version mismatch. Your help will be appreciated.

Thank you

MartyG-RealSense, 2023-10-04:
Hi @aliasghar1721  If you are using SDK 2.51.1 then there would be a mismatch with the latest version of the ros2-development wrapper, which was designed for use with SDK **2.54.1** at the time of writing this.

You could try installing SDK 2.54.1 and then building the ROS2 wrapper again with colcon.

If 2.54.1 is installed then the firmware driver installed in your computer should be version **5.15.0.2**.

aliasghar1721, 2023-10-04:
Hi @MartyG-RealSense, thank you for your reply.

Just to clarify, I am currently using librealsense v2.51.1 with a custom wrapper for an application with no issues regarding operating my Realsense camera (D457).

Regardless, I built the realsense-ros wrapper's ros2-development branch with librealsense v2.54.1 using colcon. Unfortunately, I still have the exact same problem.

MartyG-RealSense, 2023-10-04:
Usually, building the ROS2 wrapper on Windows involves using the WLS2 Linux interface within Windows and following a long procedure of manually installing components that does not always work.  

I believe that you are the first RealSense user to take this ROS2 wrapper installation approach with VS and whilst it's excellent that the installation worked for you without errors, there are no references available for solving problems that occur during launch with such an installation, unfortunately.  I do apologize.

Have you tried both types of launch instruction?

**ros2 run**
`ros2 run realsense2_camera realsense2_camera_node`

**ros2 launch**
`ros2 launch realsense2_camera rs_launch.py`

aliasghar1721, 2023-10-04:
That's unfortunate. Yes I've tried both the launch instructions. The `ros2 launch realsense2_camera rs_launch.py` command gives me the following output:

`
[INFO] [launch]: All log files can be found below C:\Users\mariGPU\.ros\log\2023-10-04-12-18-45-106756-Ali-Asghar-20676
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node.EXE-1]: process started with pid [39988]
[ERROR] [realsense2_camera_node.EXE-1]: process has died [pid 39988, exit code 3221226505, cmd 'c:\opt\realsense_ros2\install\lib\realsense2_camera\realsense2_camera_node.EXE --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file C:\Users\mariGPU\AppData\Local\Temp\launch_params_l0ve98dk --params-file C:\Users\mariGPU\AppData\Local\Temp\launch_params_wvrs619k'].
`

I've tried the WSL2 method as well (I didn't run into much trouble while building it). While realsense-viewer is able to access my camera from WSL2, the ros2 node cannot find my camera, despite it being listed by the `lsusb` command.

I am not sure what my next step should be.

MartyG-RealSense, 2023-10-05:
A user of Windows and ROS Humble described the method that they used to achieve a successful build of the wrapper with colcon build at https://github.com/IntelRealSense/realsense-ros/issues/2660#issuecomment-1488870871

closed 2023-10-11 by aliasghar1721