realsenseai/realsense-ros #2892: Frames didn't arrived within 5 seconds Warning

jolags313, 2023-10-04:
Hello, I am using an Intel D435i camera with Intel® RealSense SDK 2.0 (v2.50.0) with the recommended firmware 5.13.0.50 on Ubuntu 20.04 with ROS Noetic. Whenever I try to launch the camera, I get a warning saying that "Frames didn't arrived within 5 seconds". The launch command I am using is `roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true enable_imu:=true enable_gyro:=true enable_accel:=true`, and variations of this run into the same issue.

I did not encounter this issue before when using the latest Intel® RealSense SDK 2.0 (v2.54.2) with whichever firmware is installed when following the Linux Distribution document, but I had to uninstall and install my current configuration as I was not able to get the depth stream. I used the above Intel® RealSense SDK 2.0 version as it is listed as the supported version for realsense-ros, which I overlooked when installing for the first time. 

I looked at other issues, with answers there saying that it may be an issue with the USB cable or use of a hub, or that unplugging and plugging the camera back in may help, but none of these did anything. I am a bit confused since nothing has changed besides installing the recommended versions of Intel® RealSense SDK and the corresponding firmware. The terminal output I get is below.

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
[... 74 lines cut ...]
[ INFO] [1696459423.375276888]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1696459423.377683236]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1696459423.382945876]: gyro stream is enabled - fps: 200
[ INFO] [1696459423.383007561]: accel stream is enabled - fps: 63
[ INFO] [1696459423.383061542]: setupPublishers...
[ INFO] [1696459423.388996380]: Expected frequency for depth = 30.00000
[ INFO] [1696459423.430188015]: Expected frequency for infra1 = 30.00000
[ INFO] [1696459423.460679450]: Expected frequency for infra2 = 30.00000
[ INFO] [1696459423.487393781]: Expected frequency for color = 30.00000
[ INFO] [1696459423.515962771]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1696459423.544190568]: Start publisher IMU
[ INFO] [1696459423.546589723]: setupStreams...
 04/10 17:43:43,554 WARNING [140227567793920] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ WARN] [1696459423.663911407]: 
[ INFO] [1696459423.668558382]: SELECTED BASE:Depth, 0
[ INFO] [1696459423.706375675]: RealSense Node Is Up!
 04/10 17:43:48,625 WARNING [140226588571392] (backend-v4l2.cpp:1127) Frames didn't arrived within 5 seconds
 04/10 17:43:53,629 WARNING [140226588571392] (backend-v4l2.cpp:1127) Frames didn't arrived within 5 seconds
 04/10 17:43:58,633 WARNING [140226588571392] (backend-v4l2.cpp:1127) Frames didn't arrived within 5 seconds
 04/10 17:44:03,637 WARNING [140226588571392] (backend-v4l2.cpp:1127) Frames didn't arrived within 5 seconds
^C[camera/realsense2_camera-2] killing on exit
[camera/realsense2_camera_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2023-10-05:
Hi @jolags313  Your versions match up correctly (librealsense 2.50.0, ROS1 wrapper 2.3.2, firmware driver 5.13.0.50).  So there is no problem there.

Does the **Frames didn't arrived within 5 seconds** warning still occur if you perform a minimal launch with only depth and color enabled but IMU and infrared 1 and 2 disabled?
 
`roslaunch realsense2_camera rs_camera.launch`

**************

Does resetting the camera at launch with **initial_reset:=true** make a difference?

`roslaunch realsense2_camera rs_camera.launch initial_reset:=true`

*********

Does the warning occur if you perform an **rs_rgbd** launch instead of rs_camera?  You can test this with a 2-step procedure.

1.  Install support for an RGBD launch with the command below.

`sudo apt-get install ros-noetic-rgbd-launch`

2.  Imput the rs_rgbd launch instruction.

`roslaunch realsense2_camera rs_rgbd.launch`

jolags313, 2023-10-05:
Hello @MartyG-RealSense 

Your first suggestion works, but there are still some warnings (I am not sure if these are relevant or not, though). Those are shown below.

...
[ WARN] [1696521580.232066077]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1696521580.238313987]: Done Setting Dynamic reconfig parameters.
[ INFO] [1696521580.239317542]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1696521580.241369061]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1696521580.241408930]: setupPublishers...
[ INFO] [1696521580.244412942]: Expected frequency for depth = 30.00000
[ INFO] [1696521580.290142681]: Expected frequency for color = 30.00000
[ INFO] [1696521580.319098443]: setupStreams...
[ INFO] [1696521580.368545006]: SELECTED BASE:Depth, 0
[ INFO] [1696521580.379985615]: RealSense Node Is Up!
[ WARN] [1696521580.511490651]: 

As for resetting the camera at launch, I had tried this along with my previous launch command, but that did not work, and I am met with additional errors. The output I get when running my original launch command with the reset is as follows.

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
[... 115 lines cut ...]
 05/10 11:07:51,250 ERROR [139788609910528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 05/10 11:07:51,283 ERROR [139788609910528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 05/10 11:07:51,317 ERROR [139788609910528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 05/10 11:07:51,353 ERROR [139788609910528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 05/10 11:07:51,383 ERROR [139788609910528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 05/10 11:07:51,416 ERROR [139788609910528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 05/10 11:07:51,450 ERROR [139788609910528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 05/10 11:07:51,483 ERROR [139788609910528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 05/10 11:07:51,516 ERROR [139788609910528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 05/10 11:07:51,550 ERROR [139788609910528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
[... cut ...]
 05/10 11:07:55,625 WARNING [139788618303232] (backend-v4l2.cpp:1127) Frames didn't arrived within 5 seconds
 05/10 11:08:00,630 WARNING [139788618303232] (backend-v4l2.cpp:1127) Frames didn't arrived within 5 seconds
 05/10 11:08:05,633 WARNING [139788618303232] (backend-v4l2.cpp:1127) Frames didn't arrived within 5 seconds
^C[camera/realsense2_camera-2] killing on exit
[camera/realsense2_camera_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Lastly, for the rs_rgbd launch, that works just fine.

MartyG-RealSense, 2023-10-05:
I'm pleased to hear that the rs_rgbd lanch method works fine!

In regard to the rs_camera errors, **Out of frame resources** could suggest that the computer is struggling to keep up with the volume of data arriving at the USB port from the camera.  If the topics are being published normally in RViz though despite the error messages then they can likely be ignored.

jolags313, 2023-10-05:
Sorry, I meant that the rs_rgbd launch worked fine with just the command you gave me (not with the original launch command parameters I need). 

When I did try it with those using `roslaunch realsense2_camera rs_rgbd.launch unite_imu_method:=linear_interpolation enable_infra:=true enable_infra1:=true enable_infra2:=true enable_imu:=true enable_gyro:=true enable_accel:=true`, I found that although the 'Frames didn't arrived within 5 seconds' warning was fixed, there was no longer an infrared stream/feed, and it no longer shows up in the rostopic list. The output can be seen below.

SUMMARY
========

PARAMETERS
 * /camera/color_rectify_color/queue_size: 5
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: -1
[... 64 lines cut ...]
[ INFO] [1696525103.102269878]: Sync Mode: On
[ INFO] [1696525103.102422703]: Device Sensors: 
[ INFO] [1696525103.115564220]: Stereo Module was found.
[ INFO] [1696525103.144211560]: RGB Camera was found.
[ INFO] [1696525103.145278612]: Motion Module was found.
[ INFO] [1696525103.145377643]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1696525103.147981795]: num_filters: 1
[ INFO] [1696525103.148165712]: Setting Dynamic reconfig parameters.
[ WARN] [1696525103.201824515]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1696525103.212557846]: Done Setting Dynamic reconfig parameters.
[ INFO] [1696525103.213795885]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1696525103.216311505]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1696525103.218870650]: gyro stream is enabled - fps: 200
[ INFO] [1696525103.219056081]: accel stream is enabled - fps: 63
[ INFO] [1696525103.219192440]: setupPublishers...
[ INFO] [1696525103.221797170]: Expected frequency for depth = 30.00000
[... cut ...] ed frequency for color = 30.00000
[ INFO] [1696525103.283884520]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1696525103.313668576]: setupStreams...
 05/10 11:58:23,320 WARNING [139912921741056] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ INFO] [1696525103.374521985]: SELECTED BASE:Depth, 0
[ INFO] [1696525103.389459667]: RealSense Node Is Up!
[ WARN] [1696525103.459243881]: 

Is there any way to fix this new issue?

MartyG-RealSense, 2023-10-05:
Please try removing `enable_infra:=true` from the instruction as it can cause infra1 and infra2 to stop working.  Use enable_infra1 and enable_infra2 only.

jolags313, 2023-10-05:
While it does cause the `camera/infra1/image_rect_raw` to now appear, the topic itself is still empty, and I am unable to see the infrared stream/feed. There are also new warnings/errors output as well. The output is below.

SUMMARY
========

PARAMETERS
 * /camera/color_rectify_color/queue_size: 5
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: -1
 * /camera/realsense2_camera/infra_height: -1
 * /camera/realsense2_camera/infra_rgb: False
[... 67 lines cut ...]
[ INFO] [1696534788.237463372]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1696534788.239829688]: num_filters: 1
[ INFO] [1696534788.239874122]: Setting Dynamic reconfig parameters.
[ WARN] [1696534788.287078249]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1696534788.295930604]: Done Setting Dynamic reconfig parameters.
[ INFO] [1696534788.297120795]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1696534788.299326187]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1696534788.302098881]: gyro stream is enabled - fps: 200
[ INFO] [1696534788.302174921]: accel stream is enabled - fps: 63
[ INFO] [1696534788.302228618]: setupPublishers...
[ INFO] [1696534788.305990864]: Expected frequency for depth = 30.00000
[ INFO] [1696534788.336914976]: Expected frequency for color = 30.00000
[ INFO] [1696534788.365042685]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1696534788.393405933]: setupStreams...
 05/10 14:39:48,400 WARNING [140498542454528] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ INFO] [1696534788.455123907]: SELECTED BASE:Depth, 0
[ INFO] [1696534788.468641626]: RealSense Node Is Up!
[ WARN] [1696534788.543701931]: 
[... cut ...]
 05/10 14:40:38,936 WARNING [140497641273088] (backend-v4l2.cpp:1108) Video frame dropped, video and metadata buffers inconsistency

I also tried unplugging the camera and plugging it back in to see if that would help, but after doing that, I could no longer see the topic anymore. I then tried to reset the camera at launch, but wasn't able to see the topic then, either. 

I'm not sure why this happened, since I didn't change anything between removing `enable_infra:=true` and unplugging and plugging the camera back in.

MartyG-RealSense, 2023-10-05:
It is unusual that you received the **UVC metadata payloads not available** warning as that should appear if librealsense was built from source code and a patch script had not been applied to the Linux kernel.

If librealsense is installed from packages with the distribution_linux instructions or by using the ROS wrapper's 'Method 1' installation method then UVC metadata should be supported.

jolags313, 2023-10-05:
I did build librealsense from the source code since I wanted to install the ROS1 compatible version. I am a bit new to all of this, so I was not sure how to install the older commit based on the installation from packages instructions, and I did not use the ROS wrapper's Method 1 since I have to edit the source code. In any case, the warning in question only appreared recently, and I am not sure why it would have since I haven't changed any of the code besides parameters in the launch files.

Also, do you have any other thoughts on the infrared stream issue? I believe I installed everything correctly, so you think it might be a hardware problem?

MartyG-RealSense, 2023-10-05:
If you built librealsense from source code with CMake and included the build flag **-DBUILD_GRAPHICAL_EXAMPLES=TRUE** in the CMake build instruction then you should be able to run the **realsense-viewer** tool that will let you test whether the infrared streams are working.

It appears that the infrared streams have not been published as they are missing from the list of published streams just above the **RealSense Node is Up!** message.

[ INFO] [1696534788.297120795]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1696534788.299326187]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1696534788.302098881]: gyro stream is enabled - fps: 200
[ INFO] [1696534788.302174921]: accel stream is enabled - fps: 63

[ INFO] [1696534788.305990864]: Expected frequency for depth = 30.00000
[ INFO] [1696534788.336914976]: Expected frequency for color = 30.00000
[ INFO] [1696534788.365042685]: Expected frequency for aligned_depth_to_color = 30.00000

jolags313, 2023-10-05:
From what I can see in the RealSense Viewer, the infrared streams are working properly, so it doesn't look like a hardware problem after all. Do you have any advice on how to make it so that they are published properly? I haven't changed any of the code, so I'm not sure what else to do.

MartyG-RealSense, 2023-10-06:
I cannot see an obvious reason why the rgbd launch is not publishing infrared.  Let's try defining it specifically in a launch.

`roslaunch realsense2_camera rs_rgbd.launch unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true enable_imu:=true enable_gyro:=true enable_accel:=true infra1_width:=640 infra1_height:=480  infra2_width:=640 infra2_height:=480 infra_fps:=30`

jolags313, 2023-10-10:
Sorry for the late response, I've been busy with other things. I tried the command, and it worked the first time I used it. However, on subsequent tries, it was hit or miss and wouldn't work about half of the time. When it does, the stream stops very soon afterwards and I get the following errors

[ INFO] [1696950939.596667608]: RealSense Node Is Up!
[ WARN] [1696950939.648180433]: 
 10/10 10:16:31,064 ERROR [140516731254528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/10 10:16:31,741 ERROR [140516731254528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/10 10:16:35,734 ERROR [140516731254528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/10 10:16:39,271 ERROR [140516731254528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/10 10:16:39,337 ERROR [140516731254528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/10 10:16:39,370 ERROR [140516731254528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/10 10:16:39,905 ERROR [140516731254528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/10 10:16:39,981 ERROR [140516731254528] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!

MartyG-RealSense, 2023-10-11:
synthetic-stream errors are one of the most difficult types to diagnose unfortunately as they are equal to a vague "something went wrong" message.  It is a positive that the launch is capable of working at least sometimes though.

If everything else tried has not worked then uninstalling and reinstalling can sometimes correct a problem.  Given that you have performed installs and uninstalls with both packages and source code, it is possible that there is something remaining on your computer that is causing a conflict.

All librealsense and ROS wrapper package files can be uninstalled from a computer using the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

After that, try launching again and if it still does not work then you can proceed to uninstall source code builds of librealsense and the ROS wrapper.

To uninstall a source code build of librealsense, go to the 'build' directory of your librealsense source code folder and run the following command:

`sudo make uninstall && make clean`

To uninstall a source code build of the ROS wrapper, simply delete the entire **catkin_ws/src** catkin workspace folder.

jolags313, 2023-10-11:
When I try to uninstall the package files using the command you have, I get the following

dpkg: error: --purge needs at least one package name argument

Type dpkg --help for help about installing and deinstalling packages [*];
Use 'apt' or 'aptitude' for user-friendly package management;
Type dpkg -Dhelp for a list of dpkg debug flag values;
Type dpkg --force-help for a list of forcing options;
Type dpkg-deb --help for help about manipulating *.deb files;

Options marked [*] produce a lot of output - pipe it through 'less' or 'more' !

Was I meant to add another argument?

MartyG-RealSense, 2023-10-12:
I cannot see any problems with the uninstall command.  Perhaps there are not currently any RealSense packages on your computer.  You could test whether this is the case by attempting to uninstall a specific package with the command below.

`sudo apt-get purge librealsense2`

jolags313, 2023-10-12:
I thought I had installed everything correctly according to the instructions, but when I try, I it says that `Package 'librealsense2' is not installed, so not removed`. Do you recommend running`sudo make uninstall && make clean` in the librealsense build directory anyway and just going back through the instructions here (https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md) again?

jolags313, 2023-10-12:
Also would this have changed anything? When going through Step 3 of the Install librealsense2 section in the link above, I get the following output when using `./scripts/patch-realsense-ubuntu-lts.sh`

Unsupported kernel version 5.15.0-84-generic . The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only

MartyG-RealSense, 2023-10-13:
If the librealsense2 package is not installed on your computer then you likely only have a source code build of librealsense on it and so do not have to be concerned about cleaning up old package installations.

The **lts.sh** patch script only supports up to kernel 5.11.  For kernel 5.13 and 5.15 support, the **patch-realsense-ubuntu-lts-hwe.sh** patch script should be used.

jolags313, 2023-10-13:
There is no lts-hwe script for the commit I am using (the one for v2.50.0/commit c94410a). The installation instructions just say to use `./scripts/patch-realsense-ubuntu-lts.sh`. If the commit I am using does not have kernel 5.15 support, am I just unable to use it on my current machine?

MartyG-RealSense, 2023-10-13:
It is possible to use librealsense on Jetson without applying a patch script.  https://github.com/IntelRealSense/librealsense/issues/10722#issuecomment-1230250762 is an example of this.

jolags313, 2023-10-13:
I see, so would I just navigate to the existing librealsense build directory and run `cmake ../ -DFORCE_RSUSB_BACKEND=TRUE` and then `sudo make uninstall && make clean && make && sudo make install` afterwards? Because when I run the first command, I see this output and warning

-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.50.0
-- Setting Unix configurations
-- Building libcurl enabled
-- using RS2_USE_LIBUVC_BACKEND
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMake/opengl_config.cmake:1 (find_package)
  examples/CMakeLists.txt:14 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMake/opengl_config.cmake:1 (find_package)
  tools/CMakeLists.txt:32 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Check for updates capability added to realsense-viewer
-- Check for updates capability added to realsense-depth-quality
-- Building with TM2
-- Fetching recommended firmwares:
-- D4XX_FW_VERSION: 5.13.0.50
-- SR3XX_FW_VERSION: 3.26.1.0
-- T26X_FW_VERSION: 0.2.0.951
-- L51X_FW_VERSION: 1.5.8.1
-- L53X_FW_VERSION: 3.5.5.1
-- https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-5.13.0.50.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for D4XX_FW_Image-5.13.0.50.bin
-- https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for SR3XX_FW_Image-3.26.1.0.bin
-- https://librealsense.intel.com/Releases/TM2/FW/target/0.2.0.951/target-0.2.0.951.mvcmd
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for target-0.2.0.951.mvcmd
-- https://librealsense.intel.com/Releases/L5xx/FW/L51X_FW_Image-1.5.8.1.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for L51X_FW_Image-1.5.8.1.bin
-- https://librealsense.intel.com/Releases/L5xx/FW/L53X_FW_Image-3.5.5.1.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for L53X_FW_Image-3.5.5.1.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/conlab-v/librealsense-2.50.0/build

MartyG-RealSense, 2023-10-13:
https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692 has a good example of the RSUSB installation process.  

`cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

As well as the RSUSB flag, including **-DBUILD_GRAPHICAL_EXAMPLES=true** builds librealsense with OpenGL graphics support and **-DCMAKE_BUILD_TYPE=release** builds librealsense in Release mode (which includes optimizations) instead of the default developer Debug mode.

jolags313, 2023-10-13:
I followed the steps and rebuilt, and the infrared camera now seems to be streaming properly. However, I am seeing new errors/warnings in the camera launch output as seen below

SUMMARY
========

PARAMETERS
 * /camera/color_rectify_color/queue_size: 5
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
[... 113 lines cut ...]
 13/10 18:48:29,420 WARNING [140648805385984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 13/10 18:48:29,471 WARNING [140648805385984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 13/10 18:48:29,522 WARNING [140648805385984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 13/10 18:48:29,573 WARNING [140648805385984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 13/10 18:48:29,623 WARNING [140648805385984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 13/10 18:48:29,674 WARNING [140648805385984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 13/10 18:48:29,926 WARNING [140648805385984] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ WARN] [1697240909.936232522]: Hardware Notification:Depth stream start failure,1.69724e+12,Error,Hardware Error
 13/10 18:48:29,977 WARNING [140648805385984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 13/10 18:49:59,442 WARNING [140648805385984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 13/10 18:49:59,493 WARNING [140648805385984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 13/10 18:50:09,551 WARNING [140648805385984] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

Are these anything to worry about?

MartyG-RealSense, 2023-10-14:
The **control_transfer returned** warning, if generating continuously in the log, can indicate that there is a communication problem between the camera and computer.  However, if the published streams appear to be working normally in RViz then the warnings can be ignored.

jolags313, 2023-10-16:
Marty, if that is the case, then I believe I am fine for now and that my streams are working properly. Thank you very much for your help!

MartyG-RealSense, 2023-10-17:
You are very welcome.  As you are fine, I will close this case.  Thanks again!

closed 2023-10-17 by MartyG-RealSense

jolags313, 2023-10-24:
@MartyG-RealSense Unfortunately, I have run into another issue and could use some more help. I recently had to move to and reinstall on another machine (Ubuntu 18.04 with ROS Melodic, RealSense SDK 2.0 v2.50.0, firmware v5.13.00.50), and this time, I found that I am once again unable to see my depth stream when using the rs_camera launch. The depth topics appear, and in the camera launch terminal output, it says that the depth stream is enabled. However, nothing is actually being published. The output I see is below

<details open>
<summary>Camera launch terminal output</summary>
<br>
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
[... 85 lines cut ...]
[ INFO] [1698166990.382972655]: Expected frequency for depth = 30.00000
[ INFO] [1698166990.444623883]: Expected frequency for infra1 = 30.00000
[ INFO] [1698166990.508845956]: Expected frequency for infra2 = 30.00000
[ INFO] [1698166990.565475188]: Expected frequency for color = 30.00000
[ INFO] [1698166990.607536973]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1698166990.658296147]: Start publisher IMU
[ INFO] [1698166990.664329733]: setupStreams...
 24/10 12:03:10,854 WARNING [140100625233664] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ WARN] [1698166990.923413678]: 
[ WARN] [1698166990.923846007]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 24/10 12:03:11,090 WARNING [140100323227392] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1698166991.140629647]: SELECTED BASE:Depth, 0
[ INFO] [1698166991.167781135]: RealSense Node Is Up!
 24/10 12:03:11,343 WARNING [140100323227392] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 24/10 12:09:46,058 WARNING [140100323227392] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
</details>

I've tried doing an initial reset, but that did not fix the issue. I then tried to use the rs_rgbd launch as you suggested before, and while that did work in getting me the depth stream, unite_imu_method is not present in the launch. Thus, I am unable to unite all of the IMU information like is possible with the rs_camera launch. Is there any way to fix this? The project I am working on requires all of the IMU information in a single topic.

MartyG-RealSense, 2023-10-25:
If you are able to edit the rs_rgbd.launch file then you should be able to add code to import the unite_imu_method parameter from the rs_cameralaunch file with an `include `instruction.

```
    <include unless="$(arg offline)" 
        file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="unite_imu_method" value="linear_interpolation"/>
    </include>
```

This technique of importing parameters from rs_camera.launch is demonstrated by the **opensource_tracking.launch** file.

https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/launch/opensource_tracking.launch#L3-L8

jolags313, 2023-10-25:
Is including those lines in the rs_rgbd.launch file all I need to do? If it matters, I placed them right after all of the args for 'Processing Modules' and before the line `<group ns="$(arg camera)">`

But when I run the command to launch `roslaunch realsense2_camera rs_rgbd.launch unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true enable_imu:=true enable_gyro:=true enable_accel:=true infra_width:=640 infra_height:=480 infra_fps:=30  depth_width:=640 depth_height:=480 depth_fps:=30  color_width:=640 color_height:=480 color_fps:=30`, I get the following error

`RLException: roslaunch file contains multiple nodes named [/camera/realsense2_camera_manager].
Please check all <node> 'name' attributes to make sure they are unique.
Also check that $(anon id) use different ids.
The traceback for the exception was written to the log file`

Did I need to edit something else?

MartyG-RealSense, 2023-10-26:
I cannot see anything obviously missing.  I would recommend trying placing the lines **after** the `<group ns="$(arg camera)">` line because all the` <include> </include>` blocks seem to be nested together under that line.

jolags313, 2023-10-26:
I tried that and didn't get the error I got last time, so that was nice. However, I no longer get the gyroscope and accelerometer readings, even when using the same command and setting the 'enable_gyro' and 'enable_accel' to true in the command and both the rs_rgbd and rs_camera launch files. Their topics, as well as that of the IMU, are all empty. I am not sure why this is the case- shouldn't the only effect of adding those lines be the inclusion of the unite_imu_method?

MartyG-RealSense, 2023-10-27:
A line that I omitted from the include was **  <arg name="linear_accel_cov" value="1.0"/>** as it did not seem relevant.  What happens if you add this line to the include along with unite_imu_method and align_depth please?

jolags313, 2023-10-27:
I tried what you suggested, and the color, infrared, and IMU streams are now working. **However, now the depth streams are empty- there is nothing publishing on the depth/image_rect_raw and aligned_depth_to_color_image_raw topics.** 

The rostopics published are also a bit strange- they all begin with /camera/camera instead of /camera as shown in the below examples

/camera/camera/aligned_depth_to_color/image_raw
/camera/camera/infra1/image_rect_raw
/camera/camera/imu

I don't think this will be much of an issue though, I just want to note it. 

Finally, I am now getting the following warnings/errors, with the block after the control_transfer returned errors appearing repeatedly (the Asic Temperature errors did not repeat, but they appear quite often as well)

 27/10 13:46:12,557 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:12,565 ERROR [140491492402944] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 131
 27/10 13:46:12,608 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:12,659 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:12,709 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:12,760 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:12,810 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:12,861 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:12,912 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:12,962 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:13,013 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:13,063 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:13,114 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:13,164 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:13,215 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:13,266 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:13,316 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:13,367 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:13,721 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 13:46:13,872 WARNING [140492689598208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1698432373.941089033]:  
[ WARN] [1698432373.953720525]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1698432373.953795042]: The requested device with  is NOT found. Will Try again.
 27/10 13:46:13,953 ERROR [140275873269504] (handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 27/10 13:46:13,953 ERROR [140275892381440] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 13:46:13,953 WARNING [140275892381440] (rs.cpp:310) null pointer passed for argument "device"
 27/10 13:46:13,984 ERROR [140491324647168] (ds5-options.cpp:88) Asic Temperature value is not valid!
[ WARN] [1698432374.182001709]: Hardware Notification:Depth stream start failure,1.69843e+12,Error,Hardware Error
 27/10 13:46:14,989 ERROR [140491324647168] (ds5-options.cpp:88) Asic Temperature value is not valid!
 27/10 13:46:15,995 ERROR [140491324647168] (ds5-options.cpp:88) Asic Temperature value is not valid!
[ INFO] [1698432379.995200664]:  
 27/10 13:46:20,007 ERROR [140275604379392] (handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
[ WARN] [1698432380.007573510]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1698432380.007611946]: The requested device with  is NOT found. Will Try again.
 27/10 13:46:20,007 ERROR [140275892381440] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 13:46:20,007 WARNING [140275892381440] (rs.cpp:310) null pointer passed for argument "device"
[ INFO] [1698432386.103988198]:  

The only thing I changed from before was adding that one line. Is there any way to fix these new issues? 

Also, is using this camera/these packages always this buggy? I thought I had followed all of the instructions and requirements properly.

jolags313, 2023-10-27:
I also get the following when using realsense-viewer

 27/10 14:08:03,235 ERROR [140458231584512] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 27/10 14:08:03,235 ERROR [140458941773568] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 14:08:03,235 WARNING [140458941773568] (sensor.cpp:1173) Failed to add Backlight Compensation control for RGB Camera
 27/10 14:08:03,236 ERROR [140458231584512] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 27/10 14:08:03,237 ERROR [140458941773568] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 14:08:03,237 WARNING [140458941773568] (sensor.cpp:1173) Failed to add Brightness control for RGB Camera
 27/10 14:08:03,237 ERROR [140458231584512] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 27/10 14:08:03,237 ERROR [140458941773568] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 14:08:03,237 WARNING [140458941773568] (sensor.cpp:1173) Failed to add Contrast control for RGB Camera
 27/10 14:08:03,237 ERROR [140458231584512] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 27/10 14:08:03,237 ERROR [140458941773568] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 14:08:03,237 WARNING [140458941773568] (sensor.cpp:1173) Failed to add Exposure control for RGB Camera
 27/10 14:08:03,237 ERROR [140458231584512] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 27/10 14:08:03,237 ERROR [140458941773568] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 14:08:03,237 WARNING [140458941773568] (sensor.cpp:1173) Failed to add Gamma control for RGB Camera
 27/10 14:08:03,237 ERROR [140458231584512] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 27/10 14:08:03,238 ERROR [140458941773568] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 14:08:03,238 WARNING [140458941773568] (sensor.cpp:1173) Failed to add Hue control for RGB Camera
 27/10 14:08:03,238 ERROR [140458231584512] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 27/10 14:08:03,238 ERROR [140458941773568] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 14:08:03,238 WARNING [140458941773568] (sensor.cpp:1173) Failed to add Saturation control for RGB Camera
 27/10 14:08:03,238 ERROR [140458231584512] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 27/10 14:08:03,238 ERROR [140458941773568] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 14:08:03,238 WARNING [140458941773568] (sensor.cpp:1173) Failed to add Sharpness control for RGB Camera
 27/10 14:08:03,238 ERROR [140458231584512] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 27/10 14:08:03,238 ERROR [140458941773568] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 14:08:03,238 WARNING [140458941773568] (sensor.cpp:1173) Failed to add White Balance control for RGB Camera
 27/10 14:08:03,238 ERROR [140458231584512] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 27/10 14:08:03,238 ERROR [140458941773568] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 14:08:03,238 WARNING [140458941773568] (sensor.cpp:1173) Failed to add Enable Auto Exposure control for RGB Camera
 27/10 14:08:03,238 ERROR [140458231584512] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 27/10 14:08:03,238 ERROR [140458941773568] (sensor.cpp:572) acquire_power failed: failed to set power state
 27/10 14:08:03,238 WARNING [140458941773568] (sensor.cpp:1173) Failed to add Enable Auto White Balance control for RGB Camera

After turning on the Stereo Module,

 27/10 14:08:19,796 WARNING [140458843956992] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 14:08:20,054 WARNING [140458843956992] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 14:08:20,105 WARNING [140458843956992] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 14:08:20,155 WARNING [140458843956992] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 14:08:20,206 WARNING [140458843956992] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 14:08:20,257 WARNING [140458843956992] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 14:08:20,307 WARNING [140458843956992] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 14:08:20,358 WARNING [140458843956992] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 14:08:20,409 WARNING [140458843956992] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 27/10 14:08:29,535 WARNING [140458843956992] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/10 14:08:29,586 WARNING [140458843956992] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

I've also been seeing a problem where the laptop I am using will sometimes just completely freeze whenever I use the camera, both when using the viewer and when using the ROS launch.

jolags313, 2023-10-27:
As another update, I ran the same command (the one I have been using this entire time) twice in a row without changing anything. The first time, all of the streams (color, depth, infrared, gyro, accel) all worked properly, but the IMU topic was missing. The topics were also all named properly. The second time I ran the command, which was right afterwards and without changing anything, the topic names all began with /camera/camera, and while the IMU topic appeared and was working the depth stream was empty again.

Both of these times, I did not encounter the errors/warnings I mentioned above, besides the control_transfer returned error. 

Additionally,

MartyG-RealSense, 2023-10-28:
The topic being /camera/camera or just /camera should not be a problem.

The **Asic Temperature value is not valid!** warning is not serious.  It just means that when the current internal temperature value of the camera hardware was requested, the camera was not ready to respond to the query.

If the **control_transfer returned** error repeats continuously in the log then it can indicate a communication problem between the camera and computer, such as a USB issue (e.g data arriving from the camera at a faster rate than the computer can process it).  However, it is sometimes possible for the enabled streams to behave normally in RViz even when these warnings are continuing to generate.

**Failed to set power state** can indicate that the computer is having problems with accessing the camera, perhaps because the camera is being supplied with insufficient power by the USB port to meet all of the power that the camera needs to draw for its various streams.  In that situation, using a mains electricity powered USB port may help if it is possible for the project to do so (for example, if it is not a mobile robot that needs to be able to move around freely).  Mains plug socket power is more stable than the power provided by a computer's USB port or by an unpowered USB hub.

jolags313, 2023-10-30:
Thank you for explaining the errors; however, is there any way to get the depth topics working?

MartyG-RealSense, 2023-10-31:
Does the depth topic appear if you set align_depth to false?

`roslaunch realsense2_camera rs_rgbd.launch align_depth:=false unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true enable_imu:=true enable_gyro:=true enable_accel:=true infra_width:=640 infra_height:=480 infra_fps:=30  depth_width:=640 depth_height:=480 depth_fps:=30  color_width:=640 color_height:=480 color_fps:=30`

jolags313, 2023-11-02:
The depth topic worked the first time I ran it, but the IMU didn't work. I then tried to add initial_reset:=true, which led to the IMU working but the depth topic no longer working. When I run the command with align_depth set to true, the depth now works (???) but now the IMU does not. 

With how buggy and inconsistent all of this is, I think I will just give up on using the D435i. I believe I went through all of the installation correctly, and I haven't changed any of the actual code, but it still isn't working even with all of this troubleshooting. Thank you for the help, though.

MartyG-RealSense, 2023-11-02:
You are very welcome.  I'm sincerely sorry that you did not achieve the desired outcome from this case.