realsenseai/realsense-ros #2896: Unable to get Aligned Point cloud even when Aligned depth is enabled.

minku1219, 2023-10-06:
Hi, I am working on a pose estimation pipeline and it requires a Ordered Pointcloud aligned with rgb.
While using realsense d435 and d455 i have observed that when enabling the ordered_pc flag to true it gives point cloud that is not aligned to rgb.
I have attached a screenshot which mentions that when using aligned depth you will get the pointcloud in fram of aligned depth. But mine is shifted.
[image]

MartyG-RealSense, 2023-10-06:
Hi @minku1219  Is a depth-color aligned pointcloud generated correctly with ordered_pc = true if you do **not** use align_depth:=true?  RGB should be mapped to a depth pointcloud when RGB is enabled and the pointcloud filter is used.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 rgb_camera.profile:=1280x720x30 pointcloud.enable:=true pointcloud.ordered_pc:=true`

adamwhats, 2023-10-07:
Looks like this the same issue as #2595?

MartyG-RealSense, 2023-10-08:
@adamwhats  The case of @minku1219 does look similar to https://github.com/IntelRealSense/realsense-ros/issues/2595 though the RealSense ROS developer team created a fix for the alignment issue in that case - see https://github.com/IntelRealSense/realsense-ros/pull/2775 - though the RealSense user in that case did not provide feedback about whether the fix had resolved their issue.

MartyG-RealSense, 2023-10-25:
Hi @minku1219 and @adamwhats  Do either of you require further assistance with this case, please?  Thanks!

closed 2023-11-01 by MartyG-RealSense