realsenseai/realsense-ros #2899: Unable to change exposure in HDR mode: hdr_config::send_sub_preset_to_fw() - hw command failed

Finn2708, 2023-10-06:
Hi,

I'm trying to utilize the HDR mode of my D435i to handle high dynamic range environments. However, I'm unable to modify `stereo_module/exposure/1` and `stereo_module/exposure/2` through `rosrun rqt_reconfigure rqt_reconfigure` as described in the readme.

My workflow looks like this:
1. `roslaunch realsense2_camera rs_camera.launch filters:=hdr_merge`
```
... logging to /home/weithoff/.ros/log/d61b4b84-6444-11ee-9a6e-7785b070c095/roslaunch-weithoff-CELSIUS-H7510-47262.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://weithoff-CELSIUS-H7510:33137/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: hdr_merge
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
[... 106 lines cut ...]
 06/10 14:35:36,162 WARNING [140330003330816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 06/10 14:35:36,212 WARNING [140330003330816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 06/10 14:35:36,263 WARNING [140330003330816] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
```

2. `rosrun rqt_reconfigure rqt_reconfigure`
3. Set `sequence_id` from `UVC (0)` to `1 (1)` (sequence_name: 0, sequence_size: 2):
```
[ INFO] [1696595778.899847823]: call to set stereo_module/set_parameters/gain to 16 succeeded
[ INFO] [1696595778.900906304]: call to set stereo_module/set_parameters/exposure to 7500 succeeded
 06/10 14:36:18,899 WARNING [140330894071552] (hdr-config.cpp:364) In hdr_config::send_sub_preset_to_fw() - hw command failed: std::exception
 06/10 14:36:18,900 WARNING [140330894071552] (hdr-config.cpp:364) In hdr_config::send_sub_preset_to_fw() - hw command failed: std::exception
 06/10 14:36:26,315 WARNING [140330894071552] (hdr-config.cpp:364) In hdr_config::send_sub_preset_to_fw() - hw command failed: std::exception
```

4. Set `exposure` to `5000`:
```
06/10 14:36:31,191 WARNING [140330894071552] (hdr-config.cpp:364) In hdr_config::send_sub_preset_to_fw() - hw command failed: std::exception
```

It seems like the parameters can not be written to the camera. I think, I configured the HDR mode correctly, as the infra1 image is flickering as expected. I can also switch between over- and underexposed images via `sequence_id_filter` in `rqt_reconfigure`. Setting the exposure via launch file also works as expected, but I would prefer to set it dynamically.

Is this a known bug or perhaps some hardware issue on my side?

I am on Ubuntu 20.04 (noetic), running the latest commit from ros1-legacy branch.

Finn2708, 2023-10-06:
On further exploration, setting the exposure of the camera in HDR mode through the realsense-viewer results in a similar error:
```
hdr-config.cpp:371 - In hdr_config::send_sub_preset_to_fw() - hw command failed: std::exception
Setting Exposure to 5715
```
The new exposure is not updated immediately. The stream has to be toggled off and on again for changes to take effect.
Is this the expected behaviour?

MartyG-RealSense, 2023-10-06:
Hi @Finn2708  If you are using librealsense SDK version 2.50.0 then your camera firmware driver version should be downgraded to **5.13.0.50**, as 5.15.0.2 is designed for use with SDK version 2.54.1.  Using a firmware with an SDK version older than the one that it is recommended for can result in errors.

In regard to setting manual exposure dynamically, does the instruction below work for you please?

`rosrun dynamic_reconfigure dynparam set /camera/stereo_module exposure 5715`

Finn2708, 2023-10-06:
Hi @MartyG-RealSense, thanks for the quick reply. 
Looks like I have to versions of the SDK installed, which is probably not good. I will look into it.

When in HDR mode, `rosrun dynamic_reconfigure dynparam set /camera/stereo_module exposure 5715` yields:
```
06/10 17:02:30,209 WARNING [139870281180928] (ds5-options.cpp:619) The control - Depth Exposure (usec) - is locked while HDR mode is active.
```
When I disable HDR, `rosrun dynamic_reconfigure dynparam set /camera/stereo_module exposure 5715` works as expected and without any output in the terminal.

Finn2708, 2023-10-06:
Downgrading the device firmware to 5.13.0.50 resolved the issue. I am now able to dynamically set the exposure through rqt_reconfigure when in HDR mode. 

Thanks :+1:

MartyG-RealSense, 2023-10-06:
Excellent!  Thanks very much for the update  :)

MartyG-RealSense, 2023-10-12:
Case closed due to solution achieved and no further comments received.

closed 2023-10-12 by MartyG-RealSense

souljaboy764, 2024-06-03:
I was still having this issue in ROS1. It happened because I was trying to set the parameters via the `json_file_path` argument. 

```xml
 <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="align_depth" value="true"/>
    <arg name="tf_prefix" value="camera_right"/>
    <arg name="filters" value="decimation,hdr_merge,disparity,spatial,temporal"/>
    <arg name="enable_pointcloud" value="true"/>
    <arg name="clip_distance" value="1"/>
    <arg name="json_file_path" value="$(find data_recorder)/config/realsense_d405.json"/>
</include>
```

The JSON itself was generated by exporting the configuration from `realsense-viewer` for Depth Accuracy.

I also got an error when trying to change the preset in realsense-viewer. Something like
```
the control uvc image gain is locked while HDR mode is active
```

I could only do it by switching off the streaming and then changing the preset.

Setup Information:
Camera Model D405
Firmware version 5.16.0.1
librealsense version 2.55.1
Ubuntu 20.04
ROS noetic

Based on this [comment](https://github.com/IntelRealSense/realsense-ros/issues/2899#issuecomment-1750891577), After I downgraded the firmware to 5.13.0.50 and the librealsense version to 2.50.0, I was able to load the parameters using the `json_file_path` argument.

This unfortunately led to flickering in the image due the `hdr_merge` filter

[image]

and the following warning in my console
```
 03/06 22:38:41,087 WARNING [140492795930368] (hdr-merge.cpp:38) HDR Merge filter cannot process frames because relevant metadata params are missing
```

If I immediately remove the `hdr_merge` filter from the launch file, then the image still flickers, but then once I unplug and replug the USB cable, the image comes normally.

I tried different USB ports on my PC as well but still had this issue

For now I have deactivated the `hdr_merge` filter since it doesn't affect the depth output significantly.

MartyG-RealSense, 2024-06-04:
Hi @souljaboy764  When HDR is enabled in the librealsense SDK then the flickering can be stopped by configuring a post-processing filter called **Filter By Sequence id**.   This filter is not included in the list of post-processing filters supported by the ROS wrapper though.  

disparity_filter
spatial_filter
temporal_filter
hole_filling_filter
decimation_filter

So disabling HDR as you have done is likely to be the best course of action.