realsenseai/realsense-ros #2900: How to start camera node with yaml file?

WundtAsim, 2023-10-08:
# environment
ubuntu2204
ROS2 humble
distro:`sudo apt install ros-humble-realsense2-*`
# question
I tried:
```
ros2 launch realsense2_camera rs_launch.py config_file:="/home/yangqi/.realsense/config.yaml"
```
The content of the 'config.yaml':
```
enable_color: true
enable_sync: true
rgb_camera.profile: 1280x720x30
depth_module.profile: 1280x720x30
depth_module.enable_auto_exposure: true
enable_depth: true
align_depth.enable: true

publish_tf: true
tf_publish_rate: 1.0

pointcloud.enable: true
decimation_filter,enable: true
spatial_filter.enable: true
temporal_filter.enable: true
disparity_filter.enable: true
```
the output of terminal is:
```
[INFO] [launch]: All log files can be found below /home/yangqi/.ros/log/2023-10-08-18-07-39-706370-PC-45691
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [45692]
[realsense2_camera_node-1] [INFO] [1696759660.023251414] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1696759660.023329198] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1696759660.023341533] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1696759660.035028812] [camera.camera]: Device with serial number 308322060611 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1696759660.035078908] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1696759660.035088875] [camera.camera]: Device with name Intel RealSense D415 was found.
[realsense2_camera_node-1] [INFO] [1696759660.035248225] [camera.camera]: Device with port number 2-3 was found.
[realsense2_camera_node-1] [INFO] [1696759660.035259953] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1696759660.036343026] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1696759660.036492318] [camera.camera]: Publishing dynamic camera transforms (/tf) at 1 Hz
[realsense2_camera_node-1] [INFO] [1696759660.036849200] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1696759660.036864743] [camera.camera]: Device Name: Intel RealSense D415
[realsense2_camera_node-1] [INFO] [1696759660.036873287] [camera.camera]: Device Serial No: 308322060611
[realsense2_camera_node-1] [INFO] [1696759660.036881183] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1696759660.036889353] [camera.camera]: Device FW version: 5.15.1
[realsense2_camera_node-1] [INFO] [1696759660.036896870] [camera.camera]: Device Product ID: 0x0AD3
[realsense2_camera_node-1] [INFO] [1696759660.036905863] [camera.camera]: Sync Mode: On
[realsense2_camera_node-1] [INFO] [1696759660.058392218] [camera.camera]: Set Exposure.1 to 7500
[realsense2_camera_node-1] [INFO] [1696759660.081797220] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1696759660.084185649] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1696759660.094407904] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1696759660.094518047] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1696759660.105195693] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1696759660.108384502] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1696759660.109364361] [camera.camera]: RealSense Node Is Up!
```
While, there is nothing in rviz2:
[image]
By the way, this works fine: `ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true align_depth.enable:=true`

MartyG-RealSense, 2023-10-08:
Hi @WundtAsim  The best reference available for using a yaml configuration file with ROS2 is at https://github.com/IntelRealSense/realsense-ros/issues/1640 and lists the contents of the yaml file.

That case related to the previous ROS2 wrapper (now called **ros2-legacy**) rather than the current 4.54.1 wrapper but the case may contain useful insights.

WundtAsim, 2023-10-08:
Hi @MartyG-RealSense  
Thanks for your quick answer. I tested my YAML file and got a conclusion:
after I delete this line: `disparity_filter.enable: true`, I got the right pointcloud in rviz.
a little confused...
anyway, my problem has been solved, but I do think there should be better documentation about how to use the parameters in yaml file.
Thanks for your help.

MartyG-RealSense, 2023-10-08:
You are very welcome!

MartyG-RealSense, 2023-10-14:
Case closed due to solution achieved and no further comments received.

closed 2023-10-14 by MartyG-RealSense

liushuya7, 2024-12-13:
I still have issue launching the realsense camera with yaml file. It was working with ros2-legacy branch, but not working with the ros2-master branch.

MartyG-RealSense, 2024-12-14:
Hi @liushuya7  The current ros2-master ROS2 wrapper provides a tutorial for using a yaml file with it at the link below, with a demo launch file called `rs_launch_get_params_from_yaml.py`

https://github.com/IntelRealSense/realsense-ros/tree/ros2-master/realsense2_camera/examples/launch_params_from_file