realsenseai/realsense-ros #2902: L515 stops working when moved in z-direction

shervinkoushan, 2023-10-10:
We have multiple L515 cameras that all drop out (stops publishing data) when the camera is moved up and down in the z-direction. See https://youtube.com/shorts/xxIAm4CA2pE?feature=share for a demo 

 

The same movement does not cause any harm when it is performed in x or y direction. Could this be related to the IMU?

We see that the IR light stops shining for a second when the camera drops out.

 

This was with realsense2_camera/rs_camera.launch

Error messages:

```
10/10 11:40:47,173 WARNING [140140175984384] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 10/10 11:41:24,108 ERROR [140139581912832] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 10/10 11:41:24,109 ERROR [140139347048192] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 131
 10/10 11:41:24,256 WARNING [140140175984384] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [/camera/realsense2_camera_manager] [1696930884.988854669]: Hardware Notification:Fatal error occurred (15),1.69693e+12,Error,Hardware Error
```

Same issue with realsense viewer

MartyG-RealSense, 2023-10-10:
Hi @shervinkoushan  I used your side-on shaking movement with an L515 but was not able to repeat the problem that you experienced.  The official trailer video for L515 shows that it is capable of being used during high-speed travel.

https://www.youtube.com/watch?v=Rx-75ZEAMeg&t=36s

Could you first try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

shervinkoushan, 2023-10-10:
> Hi @shervinkoushan I used your side-on shaking movement with an L515 but was not able to repeat the problem that you experienced. The official trailer video for L515 shows that it is capable of being used during high-speed travel.

Thank you for the prompt response. Note that the issue only occurs when the camera is abruptly moved in the z-direction. 

> Could you first try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end). Does doing so make any difference?

Unfortunately, the problem persists with the other side of the cable as well.

MartyG-RealSense, 2023-10-10:
Does it occur if you move the camera quickly only in one direction and not alternating in two directions?

shervinkoushan, 2023-10-10:
> Does it occur if you move the camera quickly only in one direction and not alternating in two directions? For example, holding the camera high above your head and then plunging it down.

I would say it is easiest to reproduce by moving steadily up and down. I tried plunging it from both directions but that did not cause any issues.

MartyG-RealSense, 2023-10-10:
If your real-world use of the camera will not involve the camera rapidly moving up and down then it may be worth waiting to see whether a problem occurs with fast travel in one direction, as the issue may not ever occur during normal use.

shervinkoushan, 2023-10-10:
> If your real-world use of the camera will not involve the camera rapidly moving up and down then it may be worth waiting to see whether a problem occurs with fast travel in one direction, as the issue may not ever occur during normal use.

Unfortunately it occurs quite often as the camera is attached to a drone

MartyG-RealSense, 2023-10-10:
Does it still occur if you move the camera up-down at a slower speed, about half the speed that it is shaken in the video?

shervinkoushan, 2023-10-10:
> Does it still occur if you move the camera up-down at a slower speed, about half the speed that it is shaken in the video?
Yes, also at slightly lower speeds.

> Will the camera be in that orientation in real-world use, as it is being held side-on (normal upright orientation is with the tripod on the bottom facing downwards)

It will not be sideways, but tilted at a 45 degree angle. Like this, but more tilted forward:

[image]

It is attached under the drone, pointing downwards at 45 degrees.

Any idea if this is a software or firmware issue? I have sometimes had to force shut down my PC after this happens.

shervinkoushan, 2023-10-10:
Here is a new video with another camera. If you have any other suggestions (firmware, software, building from source etc I would be happy to test it)

https://github.com/IntelRealSense/realsense-ros/assets/59822141/0e356352-1779-4dc7-931b-dc0dc37b8c8c

shervinkoushan, 2023-10-10:
> In the above video the firmware version is 1.5.4.1. There have been a couple of firmware updates since then and it is recommended to update to the latest available one, 1.5.8.1. You can use the 'Install' button offered in the pop-up message in the top corner of the Viewer to do so.
> 
> [image]

Thanks, but in the first video the version was 1.5.8.1 :/

MartyG-RealSense, 2023-10-10:
If you left-click on a small arrow icon at the bottom corner of the Viewer window to expand open the Viewer's debug log then you can see which error messages appear at the moment that the image stops updating.

[image]

shervinkoushan, 2023-10-11:
> If you left-click on a small arrow icon at the bottom corner of the Viewer window to expand open the Viewer's debug log then you can see which error messages appear at the moment that the image stops updating.

This is the output:
[image]

MartyG-RealSense, 2023-10-11:
This message will have a similar cause to the **control_transfer returned** warnings in ROS, indicating that the camera has not provided new image frames for a period of more than 5 seconds.  

If the L515 camera is still listed in the side-panel of the Viewer during these log warnings then it suggests that the camera has not become disconnected by the rapid up-down motion.

In regard to the red **Fatal error occurred** error, the fact that the camera is restorable to normal operation after it has occurred during rapid up-down movement indicates that it is a temporary glitch rather than a permanent camera hardware fault.

During further tests, I was able to replicate the error in the Viewer and the accompanying depth freeze with some difficulty by wildly swinging the camera up and down and in circular movements for a period of about 10 seconds.  Stopping the camera and re-enabling it cleared the problem and enabled the camera to continue functioning normally.  The error could not be generated again after the restart when repeating the same motions.

Does the issue occur if you first go to the **More** option at the top of the Viewer side-panel and select **Hardware Reset** from its menu to reset the camera and then enable the stream afterwards and shake the camera?

shervinkoushan, 2023-10-11:
I noticed this parameter that might be relevant: `freefall_detection_enabled`

shervinkoushan, 2023-10-11:
I have not been able to reproduce the issue after setting it to false

MartyG-RealSense, 2023-10-11:
It sounds logical that disabling freefall detection would affect how the camera deals with rapid motion.  This option shuts the sensor down when a free-fall is detected.  It is enabled by default.

shervinkoushan, 2023-10-11:
We see a lot less dropouts when running on the drone as well now. 

Still, sometimes we lose data (even when the drone is hovering, so another issue). I wonder if it is related to noise

MartyG-RealSense, 2023-10-12:
If severe vibration is being transmitted through the drone's frame to the camera during flight then that could disrupt the connection.  if that is the case then adding vibration damping pads to the camera mounting points could help with this.

MartyG-RealSense, 2023-10-19:
Hi @shervinkoushan  Do you require further assistance with this case, please?  Thanks!

shervinkoushan, 2023-10-19:
> Hi @shervinkoushan Do you require further assistance with this case, please? Thanks!

We will try to solve the remaining issues by adding more damping. Thanks for the help!

closed 2023-10-19 by shervinkoushan