realsenseai/realsense-ros #2903: D435i base_frame + camera_link definition

tflueckiger, 2023-10-12:
Hi, 

I want to use the D435i with OpenVINS. I struggle to find what I am supposed to set for the parameter base_frame_id. In the documentation, it is mentionned "for example the camera_link". What is the camera_link? I have seen this term many times. It is said that all the others static transforms refers to, so my question is how should I define it so that the others transforms are correct ? What does it represent physically ? Can I define it wrt to a tf from my state estimator (openVINS) ? 

Thank you

MartyG-RealSense, 2023-10-12:
Hi @tflueckiger  **camera_link** corresponds to the center-line of the camera's left infrared sensor (which is the origin point of depth) and **base_link** corresponds to the tripod screw-thread hole on the bottom of the camera.

[image]

There is not much information available though about using OpenVINS with RealSense cameras other than the links below.

https://github.com/rpng/open_vins/issues/102

https://github.com/ethz-asl/kalibr/wiki/Calibrating-Intel-RealSense-Cameras

MartyG-RealSense, 2023-10-12:
1.  The D435i coordinate system is described at the documentation link below.  The D455, which has an IMU, uses thesame coordinate system.

https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md

For all camera models, the axis system is X horizontal, Y vertical and Z forward.  This differs from the ROS axis system, which is X horizontal, Y forward and Z vertical.  The RealSense ROS wrapper therefore automatically converts the RealSense camera axes to the standard ROS axes.

2.  The default for base_frame_id will be **camera_link**, which.does indeed correspond to the left infrared sensor.

3.  My knowledge of TF is admittedly limited, so I do not know the answer to this question unfortunately.  I do apologize.