realsenseai/realsense-ros #2904: How to directly extract depth images from .bag files?

hushimin-create, 2023-10-13:
I used ROSBAG to simultaneously record depth maps and color maps of multiple cameras. I hope to extract the color maps and depth maps of multiple cameras into images for subsequent image processing. The color map extraction was successful. How can I extract depth maps of different cameras?

MartyG-RealSense, 2023-10-13:
Hi @hushimin-create  If you used ROS and **rosbag record** to record topics from all attached cameras into a single rosbag file then you could try using the RealSense SDK's bag extraction tool **rs-convert** to convert the depth and color frames into various formats.  

The rs-convert tool was intended for bags recorded in RealSense SDK tools such as the RealSense Viewer but you could give it a try to see how it responds to your rosbag.

Alternatively, you could try the ROS bag tool **extract_stereo_images**

http://wiki.ros.org/bag_tools

MartyG-RealSense, 2023-10-19:
Hi @hushimin-create  Do you require further assistance with this case, please?  Thanks!

hushimin-create, 2023-10-19:
Thank you very much for your concern！
Since the rs-convert tool needs to be used in windows, I want to use ubuntu, so I did not choose this method. I installed bag_tools, but it couldn't extract the depth map I needed.My topic are /camera1/depth/image_rect_raw and /camera2/depth/image_rect_raw.

The following situation occurred when I used extract_stereo_images  to extract my bag：
hu@hu:~/bag_hu$ /home/hu/catkin_ws/devel/lib/bag_tools/extract_stereo_images /tmp jpg /stereo_down HU.bag
Starting processing HU.bag, 0 messages to process.

0%   10   20   30   40   50   60   70   80   90   100%
|----|----|----|----|----|----|----|----|----|----|
Finished processing HU.bag
This may be due to different topics.So I'm still looking for something else method.

MartyG-RealSense, 2023-10-19:
rs-convert can be used with Ubuntu too.  If the librealsense SDK has been built with its tools and examples included then a pre-made executable version of rs-convert should be in the **usr/local/bin** folder of Ubuntu.

hushimin-create, 2023-10-20:
Thank you very much, I found rs-convert. But I don't know how to use it to parse the depth map of different cameras in a bag. Is there any python code that can be used directly? Thank you ！

MartyG-RealSense, 2023-10-20:
The RealSense SDK has a Python example called **read_bag_example.py** for reading a bag file at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py

A bag recording stores the serial number of the camera.  So you should be able to access a specific camera's data via its serial number with the instruction **config.enable_device** as shown in the Python script at https://github.com/IntelRealSense/librealsense/issues/4968#issuecomment-546837805

closed 2023-11-01 by MartyG-RealSense