realsenseai/realsense-ros #2907: [ROS2] D455 unable to start with depth auto exposure when outdoors

drapado, 2023-10-17:
Hi using this command:
```ros2 launch realsense2_camera rs_launch.py serial_no:="'318122302162'" camera_name:=D455_top initial_reset:=true enable_color:=true enable_depth:=true depth_module.enable_auto_exposure:=true pointcloud.enable:=true depth_module.profile:=848x480x30 rgb_camera.profile:=848x480x30 pointcloud.ordered_pc:=true align_depth.enable:=true  depth_module.hdr_enabled:=true
```
both of my D455 cameras don't start in depth auto-exposure mode when outdoors (they do when indoors). The infrared images are completely overblown. I also get this warning: `[realsense2_camera_node-1] [WARN] [1697538565.749702035] [D455_top.D455_top]: re-enable the stream for the change to take effect.`

How can I solve it?

MartyG-RealSense, 2023-10-17:
Hi @drapado  The launch command that you used did not display in the opening message above.  Could you repeat it please?  Thanks!

drapado, 2023-10-17:
Hi, I updated the original message (I forgot to add a new line before the actual code). You should see it now.

MartyG-RealSense, 2023-10-17:
Thanks very much for the update!

I note that you have **hdr_enabled** set to true.  If HDR is true then auto-exposure becomes false, as auto-exposure cannot be used with HDR mode.  

Perhaps auto-exposure also becomes false indoors but it is not as noticable indoors under artificial lighting as it would be outdoors in variable lighting conditions.

MartyG-RealSense, 2023-10-18:
Thanks very much for your detailed feedback.  Given that HDR mode does not use auto-exposure because it alternates between two different exposure values, using manual exposure is the most suitable approach.

drapado, 2023-10-18:
Hi @MartyG-RealSense, re-tested indoors right now. The depth auto_exposure mode doesn't start automatically as it should. After using the following command:
```bash
ros2 launch realsense2_camera rs_launch.py serial_no:="'318122302479'" \
camera_name:=D455_bottom \
initial_reset:=true \
enable_color:=true \
enable_depth:=true \
depth_module.enable_auto_exposure:=true \
pointcloud.enable:=true \
depth_module.profile:=848x480x30 \
rgb_camera.profile:=848x480x30 \
pointcloud.ordered_pc:=true \
align_depth.enable:=true
```

If I run `ros2 param get /D455_bottom/D455_bottom depth_module.auto_exposure_mode`, I get `Boolean value is: False`.

This shouldn't be happening right?

MartyG-RealSense, 2023-10-18:
That is correct, if `depth_module.enable_auto_exposure:=true` is used in the launch instruction then depth auto-exposure should be enabled after launch has completed.

Auto-exposure can disable automatically if HDR mode is enabled or the exposure or gain value is changed.

drapado, 2023-10-18:
Any idea on how can this be solved? Is there a parameter wrongly configured in the default rs_launch.py? I believe HDR is disabled by default

MartyG-RealSense, 2023-10-18:
There is not anything in rs_launch.py that should be disabling auto-exposure (which is enabled by default) and HDR is disabled by default.  

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_launch.py#L51

What is the auto-exposure status if you launch with **ros2 run** instead of a launc file, please?

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p initial_reset:=true -p serial_no:="'318122302479'"`

drapado, 2023-10-19:
I'll try it this evening or tomorrow morning. I don't have the cameras with me right now.

MartyG-RealSense, 2023-10-19:
Okay, thanks very much for the update.  I look forward to your next report.  Good luck!

drapado, 2023-10-20:
Using
```bash
ros2 run realsense2_camera realsense2_camera_node --ros-args -p initial_reset:=true -p depth_module.enable_auto_exposure:=true -p serial_no:="'318122302479'"
```
also launch with auto-exposure disable. Running:
```bash
ros2 param get /camera depth_module.auto_exposure_mode
```
also returns: `Boolean value is: False`

MartyG-RealSense, 2023-10-20:
Have you tried using **param set** instead of param get to set auto_exposure_enabled to true?

`ros2 param set /camera depth_module.auto_exposure_enabled 1`

drapado, 2023-10-20:
Running the provide command doesn't work:
```bash
> ros2 param set /camera depth_module.auto_exposure_enabled 1
Setting parameter failed: parameter 'depth_module.auto_exposure_enabled' cannot be set because it was not declared
```

If I run the following:
```bash
> ros2 param set /camera depth_module.auto_exposure_mode true
Set parameter successful
```
However, in the terminal running the node I get:
```bash
Failed to set value: setting this option during streaming is not allowed!
```
Using 1 instead of true doesn't work:
```bash
Failed to set value: parameter 'depth_module.auto_exposure_mode' has invalid type: expected [bool] got [integer]
Failed to set value: setting this option during streaming is not allowed!
```

MartyG-RealSense, 2023-10-20:
Thank you for the information.

Next, please try holding your hand over the front of the D405 to block its lenses when you are launching it.  This will help to eliminate the possibility that something in the environment is causing the auto-exposure to disengage when launch has completed.

drapado, 2023-10-20:
Still, both of my D455 start with depth auto-exposure disabled when a hand is in front of the lens

MartyG-RealSense, 2023-10-20:
There was a past case at https://github.com/IntelRealSense/realsense-ros/issues/2507#issuecomment-1411214372 where auto-exposure was disabled in the ROS2 wrapper.  A RealSense team member investigated it and found that a **depth_module.exposure.1** line in the rs_launch.py launch file that set depth exposure to '1' was disabling auto-exposure at launch.  This is because when a manual exposure value is set, auto-exposure automatically disables.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_launch.py#L52

The RealSense team member discusses this at https://github.com/IntelRealSense/realsense-ros/issues/2507#issuecomment-1411214372

They suggested in that past case to try the following launch command:

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_color:=true -p enable_infra1:=false -p enable_infra2:=false`

drapado, 2023-10-20:
Thanks for the information, unfortunately that command doesn't solve the issue neither.

The other issue is still open, was it solved?

MartyG-RealSense, 2023-10-26:
Hi @drapado  Has there been any progress made with this problem or do you require further assistance, please?  Thanks!

drapado, 2023-10-27:
Can you replicate the issue?

MartyG-RealSense, 2023-10-27:
I am not able to replicate the issue, unfortunately.

drapado, 2023-10-28:
Hi @MartyG-RealSense, then I'm closing the issue for now. I'll create a clean ubuntu install and try again. If the issue persist, or if I gain further insights into it I'll re-open it.

Thanks for the help!

closed 2023-10-28 by drapado

MartyG-RealSense, 2023-10-28:
You are very welcome.  Thanks very much for your patience and understanding!