realsenseai/realsense-ros #2909: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link"

cgroves3, 2023-10-21:
I'm experiencing an issue similar to https://github.com/IntelRealSense/realsense-ros/issues/2215. Here is my launch script. Please bear with me as I'm a little new to ros2
```
from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare
from launch.substitutions import PathJoinSubstitution

def generate_launch_description():
    # realsense
    rs_camera_launch = LaunchDescription([
        Node(
            package="realsense2_camera",
            executable="realsense2_camera_node",
            name="realsense_camera",
            namespace="camera",
            parameters=[{
                "align_depth.enable": True,
                "linear_accel_cov": 1.0,
                "unite_imu_method": 1,  # linear_interpolation
                "enable_rgbd": True,
                "enable_sync": True,
                "enable_color": True,
                "enable_depth": True,
                "initial_reset": True
            }]
        )
    ])

    rtabmap_launch = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            PathJoinSubstitution([
                FindPackageShare("rtabmap_launch"),
                "launch",
                "rtabmap.launch.py"
            ])
        ),
        launch_arguments={
            "args": "--delete_db_on_start",
            "rgb_topic": "/camera/color/image_raw",
            "depth_topic": "/camera/aligned_depth_to_color/image_raw",
            "camera_info_topic": "/camera/color/camera_info",
            "depth_camera_info_topic": "/camera/depth/camera_info",
            "rtabmapviz": "False",
            "rviz": "False",
            "wait_imu_to_init": "True"
        }.items()
    )

    # imu_filter_madgwick
    imu_launch = LaunchDescription(
        [
            Node(
                package='imu_filter_madgwick',
                executable='imu_filter_madgwick_node',
                name='imu_filter',
                parameters=[{
                    "use_mag": False,
                    "publish_tf": True,
                    "world_frame": "enu"
                }],
                remappings=[
                    ('/imu/data_raw', '/camera/imu'),
                ]
            )
        ]
    )

    # robot_localization
    robot_loc_launch = LaunchDescription(
        [
            Node(
                package="robot_localization",
                executable="ukf_node",
                name="ukf_filter_node",
                parameters=[
                    {
                        "frequency": 300.0,
                        "odom0": "/rtabmap/odom",
                        "odom0_config": [True, True, True,
                                         True, True, True,
                                         True, True, True,
                                         True, True, True,
                                         True, True, True],
                        "odom0_relative": True,
                        "odom0_pose_rejection_threshold": 10000000.0,
                        "odom0_twist_rejection_threshold": 10000000.0,
                        "imu0": "/imu/data",
                        "imu0_config": [False, False, False,
                                        True, True, True,
                                        True, True, True,
                                        True, True, True,
                                        True, True, True, ],
                        "imu0_differential": True,
                        "imu0_relative": False,
                        "use_control": False
                    }
                ]
            )
        ]
    )

    base_link_to_camera = LaunchDescription([
        Node(
            package="tf2_ros",
            executable="static_transform_publisher",
            output="screen",
            arguments=["0", "0", "1", "0", "0", "0", "base_link", "camera_link"]
        )
    ])

    return LaunchDescription([
        imu_launch,
        rtabmap_launch,
        rs_camera_launch,
        base_link_to_camera,
        robot_loc_launch
    ])
```

I'm getting the following error messages:
[realsense2_camera_node-4] [INFO] [1697910508.551350146] [camera.realsense_camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-4] [INFO] [1697910508.567269853] [camera.realsense_camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-4] [INFO] [1697910508.653729059] [camera.realsense_camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-4] [INFO] [1697910508.653857093] [camera.realsense_camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-4] [INFO] [1697910508.653883589] [camera.realsense_camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-4] [INFO] [1697910508.654059240] [camera.realsense_camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-4] [INFO] [1697910508.662696770] [camera.realsense_camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-4] [INFO] [1697910508.676393751] [camera.realsense_camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-4] [INFO] [1697910508.676623995] [camera.realsense_camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-4] [INFO] [1697910508.684884718] [camera.realsense_camera]: Starting Sensor: Motion Module
[realsense2_camera_node-4] [INFO] [1697910508.691061468] [camera.realsense_camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 63
[realsense2_camera_node-4] [INFO] [1697910508.691175166] [camera.realsense_camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[realsense2_camera_node-4] [INFO] [1697910508.694376247] [camera.realsense_camera]: RealSense Node Is Up!
[realsense2_camera_node-4] [WARN] [1697910508.709285409] [camera.realsense_camera]: 
[imu_filter_madgwick_node-1] [INFO] [1697910508.869217380] [imu_filter]: First IMU message received.
[rgbd_odometry-2] [ERROR] [1697910509.072118500] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910508.850821
[rgbd_odometry-2] [ERROR] [1697910509.274238295] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910508.856719
[rgbd_odometry-2] [ERROR] [1697910509.477363644] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910508.991390
[rgbd_odometry-2] [ERROR] [1697910509.680161339] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910509.137486
[rgbd_odometry-2] [ERROR] [1697910509.883610982] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910509.253501
[rgbd_odometry-2] [ERROR] [1697910510.087743900] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910509.507531
[rgbd_odometry-2] [ERROR] [1697910510.291350698] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910509.698227
[rgbd_odometry-2] [ERROR] [1697910510.494141992] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910509.929486
[rgbd_odometry-2] [ERROR] [1697910510.697950425] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910510.115256
[rgbd_odometry-2] [ERROR] [1697910510.900790041] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910510.337561

MartyG-RealSense, 2023-10-22:
Hi @cgroves3  What happens if you change unite_imu_method from '1' to '2'?  

2 is the value for Linear Interpolation.  The other setting is Copy.

cgroves3, 2023-10-22:
I get the same error unfortunately. Btw, I've included my tf_tree in case that helps
[image]

MartyG-RealSense, 2023-10-23:
I note that in the section of your launch code where you define which streams are enabled (depth and color), there are not commands for enabling the IMU streams (gyro and accel).  Setting unite_imu_method alone does not enable IMU, as the IMU topics are disabled by default. Please try adding gyro and accel enable instructions to your parameters list.

```
enable_accel
enable_gyro
```

cgroves3, 2023-10-23:
After enabling them, I get
```
[INFO] [launch]: All log files can be found below /home/oxenbot/.ros/log/2023-10-23-18-23-13-898258-oxenbot-desktop-6235
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [imu_filter_madgwick_node-1]: process started with pid [6237]
[INFO] [rgbd_odometry-2]: process started with pid [6239]
[INFO] [rtabmap-3]: process started with pid [6241]
[INFO] [realsense2_camera_node-4]: process started with pid [6243]
[INFO] [static_transform_publisher-5]: process started with pid [6245]
[INFO] [ukf_node-6]: process started with pid [6247]
[imu_filter_madgwick_node-1] [INFO] [1698099795.492423825] [imu_filter]: Starting ImuFilter
[static_transform_publisher-5] [INFO] [1698099795.492931211] [static_transform_publisher_R8uULLyDDOrFzqvx]: Spinning until killed publishing transform from 'base_link' to 'camera_link'
[imu_filter_madgwick_node-1] [INFO] [1698099795.495024692] [imu_filter]: Using dt computed from message headers
[imu_filter_madgwick_node-1] [INFO] [1698099795.495143674] [imu_filter]: The gravity vector is kept in the IMU message.
[imu_filter_madgwick_node-1] [INFO] [1698099795.495765178] [imu_filter]: Imu filter gain set to 0.100000
[imu_filter_madgwick_node-1] [INFO] [1698099795.495871071] [imu_filter]: Gyro drift bias set to 0.000000
[imu_filter_madgwick_node-1] [INFO] [1698099795.495897632] [imu_filter]: Magnetometer bias values: 0.000000 0.000000 0.000000
[ukf_node-6] [WARN] [1698099795.660668916] [ukf_filter_node]: Warning: Some linear velocity entries in parameter imu0_config are listed true, but an sensor_msgs/Imu contains no information about linear velocities
[realsense2_camera_node-4] [INFO] [1698099798.198277482] [camera.realsense_camera]: RealSense ROS v4.51.1
[realsense2_camera_node-4] [INFO] [1698099798.199177591] [camera.realsense_camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-4] [INFO] [1698099798.199216473] [camera.realsense_camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-4] [INFO] [1698099798.244924277] [camera.realsense_camera]: Device with serial number 943222070496 was found.
[realsense2_camera_node-4] 
[realsense2_camera_node-4] [INFO] [1698099798.245122207] [camera.realsense_camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.1/2-1.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-4] [INFO] [1698099798.245145248] [camera.realsense_camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-4] [INFO] [1698099798.245647097] [camera.realsense_camera]: Device with port number 2-1.1 was found.
[realsense2_camera_node-4] [INFO] [1698099798.245682715] [camera.realsense_camera]: Device USB type: 3.2
[realsense2_camera_node-4] [INFO] [1698099798.249314287] [camera.realsense_camera]: getParameters...
[realsense2_camera_node-4] [INFO] [1698099798.252699191] [camera.realsense_camera]: JSON file is not provided
[realsense2_camera_node-4] [INFO] [1698099798.253112779] [camera.realsense_camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-4] [INFO] [1698099798.253139821] [camera.realsense_camera]: Device Serial No: 943222070496
[realsense2_camera_node-4] [INFO] [1698099798.253158606] [camera.realsense_camera]: Device physical port: /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.1/2-1.1:1.0/video4linux/video0
[realsense2_camera_node-4] [INFO] [1698099798.253171694] [camera.realsense_camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-4] [INFO] [1698099798.253184975] [camera.realsense_camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-4] [INFO] [1698099798.253200592] [camera.realsense_camera]: Sync Mode: On
[realsense2_camera_node-4] [WARN] [1698099798.414780387] [camera.realsense_camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-4] [INFO] [1698099798.468539983] [camera.realsense_camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-4] [INFO] [1698099798.592596860] [camera.realsense_camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-4] [INFO] [1698099798.688282889] [camera.realsense_camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-4] [INFO] [1698099798.688520404] [camera.realsense_camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-4] [INFO] [1698099798.688546582] [camera.realsense_camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-4] [INFO] [1698099798.688786178] [camera.realsense_camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-4] [INFO] [1698099798.703111496] [camera.realsense_camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-4] [INFO] [1698099798.721649953] [camera.realsense_camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-4] [INFO] [1698099798.721995250] [camera.realsense_camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-4] [INFO] [1698099798.734334838] [camera.realsense_camera]: Starting Sensor: Motion Module
[realsense2_camera_node-4] [INFO] [1698099798.741539644] [camera.realsense_camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 63
[... cut ...]
[rgbd_odometry-2] [ERROR] [1698099801.435971296] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1698099801.022243
[rgbd_odometry-2] [ERROR] [1698099801.641300880] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1698099801.205724
[rtabmap-3] [INFO] [1698099801.842124036] [rtabmap.rtabmap]: rtabmap: Database version = "0.21.1".
```

MartyG-RealSense, 2023-10-24:
As you are using rtabmap, please try changing the **rtabmapviz** parameter in your launch arguments to **true** (it is currently false) and leaving the **rviz** parameter as false.

```
 "rtabmapviz": "True",
 "rviz": "False",
```

cgroves3, 2023-10-29:
A new window opened up with some axes, but the axes stayed stationary when I rotated the camera around.

MartyG-RealSense, 2023-10-30:
Thank you.  And what happens with the opposite settings?

```
 "rtabmapviz": "False",
 "rviz": "True",
```

cgroves3, 2023-11-01:
Interesting, so I followed the steps in the video in https://github.com/IntelRealSense/realsense-ros/issues/2563 and it appears to be working?
[image]
I can see the axes moving now, but I'm still getting the same error message in the terminal
```
[rgbd_odometry-2] [ERROR] [1698841311.351014817] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1698841310.902930
[rgbd_odometry-2] [ERROR] [1698841311.553271344] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1698841311.102226
[rgbd_odometry-2] [ERROR] [1698841311.756216708] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1698841311.306489
[rgbd_odometry-2] [ERROR] [1698841311.958628980] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1698841311.505753
[rgbd_odometry-2] [ERROR] [1698841312.162618767] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1698841311.709985
[rgbd_odometry-2] [ERROR] [1698841312.365009183] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1698841311.914226
```

MartyG-RealSense, 2023-11-01:
Does this message cause any problems for your project other than the inconvenience of the log filling up with the messages, please?

cgroves3, 2023-11-06:
> Does this message cause any problems for your project other than the inconvenience of the log filling up with the messages, please?

It causes some issues when using RTABMAP for SLAM since it can't find the transform. Today, I found the final missing piece was that I think I needed to change the `fixed_frame = camera_link` for the imu_filter_madgwick. This fixed the tf tree. 

Thank you for your time and effort @MartyG-RealSense !

[image]

closed 2023-11-06 by cgroves3

MartyG-RealSense, 2023-11-07:
You are very welcome.  It's great to hear that you found the solution.  Thanks very much for the update!

JTShuai, 2024-07-22:
Hi @cgroves3, can you please provide the final launch file? Thx!

cgroves3, 2024-07-24:
> Hi @cgroves3, can you please provide the final launch file? Thx!

@JTShuai I think you just needed this part in addition to the file.

```
    # imu_filter_madgwick
    imu_launch = LaunchDescription(
        [
            Node(
                package='imu_filter_madgwick',
                executable='imu_filter_madgwick_node',
                name='imu_filter',
                parameters=[{
                    "use_mag": False,
                    "publish_tf": True,
                    "world_frame": "enu",
                    "fixed_frame": "camera_link"
                }],
                remappings=[
                    ('/imu/data_raw', '/camera/imu'),
                ]
            )
        ]
    )
```

I'd prefer not to show to whole file as I'm working on a project that I plan to commercialize

JTShuai, 2024-07-30:
Ah, I think I should not set `camera_imu_frame` to `fixed_frame` as the transformation between imu_optical frame and imu_frame should be static. So I set it to `camera_link` now. Then I have a new question if the `imu_filter_madgwick` will compute the original transformation chain `camera_link --> camera_gyro_frame --> camera_imu_frame --> camera_imu_optical_frame` in the new publishing transformation `camera_link --> camera_imu_optical_frame`?

I also opened a new issue #3173 . Any help would be greatly appreciated!