realsenseai/realsense-ros #2912: Can't use rs_camera.py script with D457 using serial_no option in ROS2

maksimmasalski, 2023-10-25:
Hi,
I have GMSL camera and I'm trying to make it work in ROS. Usually with USB 3.0 Realsense cameras like D435i, I can easily provide `serial_no` option to choose camera that I want to use in ROS2 (I use Humble), like this 
`ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true enable_sync:=true init_reset:=true pointcloud.enable:=true serial_no:="'947522071000'"`

with D457 the same style causes an error "No RealSense devices were found!:"
`ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true enable_sync:=true init_reset:=true pointcloud.enable:=true serial_no:="'220422302342'"`

Full Error log:
```
ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true enable_sync:=true init_reset:=true pointcloud.enable:=true serial_no:="'220422302342'"
[INFO] [launch]: All log files can be found below /home/intel15w-02/.ros/log/2023-10-25-13-30-32-190788-intel15w-02-22673
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [22674]
[realsense2_camera_node-1] [INFO] [1698233432.306589411] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1698233432.306648299] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1698233432.306655556] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1]  25/10 13:30:32,310 ERROR [140485947545152] (backend-v4l2.cpp:922) UVC meta-node with no video streaming node encountered: id- /dev/video-rs-depth-0
[realsense2_camera_node-1] vid- 8086
[realsense2_camera_node-1] pid- abcd
[realsense2_camera_node-1] mi- 0
[realsense2_camera_node-1] unique_id- PCI:pci:pci0000:00-0-0
[realsense2_camera_node-1] path- /dev/video-rs-depth-0
[realsense2_camera_node-1] UVC capabilities- 4a01000
[realsense2_camera_node-1] UVC specification- 0
[realsense2_camera_node-1] 
[realsense2_camera_node-1]  25/10 13:30:32,310 ERROR [140485947545152] (backend-v4l2.cpp:922) UVC meta-node with no video streaming node encountered: id- /dev/video-rs-depth-md-0
[realsense2_camera_node-1] vid- 8086
[realsense2_camera_node-1] pid- abcd
[realsense2_camera_node-1] mi- 3
[realsense2_camera_node-1] unique_id- PCI:pci:pci0000:00-0-0
[realsense2_camera_node-1] path- /dev/video-rs-depth-md-0
[realsense2_camera_node-1] UVC capabilities- 4a00000
[realsense2_camera_node-1] UVC specification- 0
[realsense2_camera_node-1] 
[realsense2_camera_node-1]  25/10 13:30:32,310 ERROR [140485947545152] (backend-v4l2.cpp:922) UVC meta-node with no video streaming node encountered: id- /dev/video-rs-color-0
[realsense2_camera_node-1] vid- 8086
[realsense2_camera_node-1] pid- abcd
[realsense2_camera_node-1] mi- 0
[realsense2_camera_node-1] unique_id- PCI:pci:pci0000:00-0-0
[realsense2_camera_node-1] path- /dev/video-rs-color-0
[realsense2_camera_node-1] UVC capabilities- 4a01000
[realsense2_camera_node-1] UVC specification- 0
[realsense2_camera_node-1] 
[realsense2_camera_node-1]  25/10 13:30:32,310 ERROR [140485947545152] (backend-v4l2.cpp:922) UVC meta-node with no video streaming node encountered: id- /dev/video-rs-color-md-0
[realsense2_camera_node-1] vid- 8086
[realsense2_camera_node-1] pid- abcd
[realsense2_camera_node-1] mi- 3
[realsense2_camera_node-1] unique_id- PCI:pci:pci0000:00-0-0
[realsense2_camera_node-1] path- /dev/video-rs-color-md-0
[realsense2_camera_node-1] UVC capabilities- 4a00000
[realsense2_camera_node-1] UVC specification- 0
[realsense2_camera_node-1] 
[realsense2_camera_node-1]  25/10 13:30:32,310 ERROR [140485947545152] (backend-v4l2.cpp:922) UVC meta-node with no video streaming node encountered: id- /dev/video-rs-imu-0
[realsense2_camera_node-1] vid- 8086
[realsense2_camera_node-1] pid- abcd
[realsense2_camera_node-1] mi- 4
[realsense2_camera_node-1] unique_id- PCI:pci:pci0000:00-0-0
[realsense2_camera_node-1] path- /dev/video-rs-imu-0
[realsense2_camera_node-1] UVC capabilities- 4a01000
[realsense2_camera_node-1] UVC specification- 0
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [WARN] [1698233432.310871285] [camera.camera]: No RealSense devices were found!
```

What I am doing wrong or is it a bug?

MartyG-RealSense, 2023-10-25:
Hi @maksimmasalski  Is your D457 set up for use on a **GMSL** connection, with the required accompanying components such as a special software driver and a **deserializer** board?  If you do not have either of these then the D457 can be used on a USB connection by moving a **mode switch** in a recess within the camera casing to the 'USB' position (the switch's default position is 'MIPI').  Instructions for doing so can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/14840675121043/comments/14843232242835

maksimmasalski, 2023-10-26:
> Hi @maksimmasalski Is your D457 set up for use on a **GMSL** connection, with the required accompanying components such as a special software driver and a **deserializer** board? If you do not have either of these then the D457 can be used on a USB connection by moving a **mode switch** in a recess within the camera casing to the 'USB' position (the switch's default position is 'MIPI'). Instructions for doing so can be found at the link below.
> 
> https://support.intelrealsense.com/hc/en-us/community/posts/14840675121043/comments/14843232242835
> 
> In USB mode the camera will be detected as **D455**, with the same feature-set and performance as a D455 but with the added protection against dust and water provided by the IP65-rated camera casing.
> 
> In USB mode the camera is 'plug and play' like the other RealSense 400 Series models and does not require the special driver or the deserializer board.

Yes, I have all the equipment like deserializer board and driver.

MartyG-RealSense, 2023-10-26:
Support for D457 in the ROS2 compatibility 'wrapper' for RealSense was added in version 4.54.1 of the wrapper, which is designed for use with version 2.54.1 of the RealSense SDK and ROS2 Humble or Iron.  So this part of your system should be fine.

https://github.com/IntelRealSense/realsense-ros/pull/2557

Intel's MIPI reference driver is designed for use with an Jetson AGX Xavier computer and Intel's own brand of deserializer board, otherwise then the reference driver must be converted (ported) for use with the computer and deserializer component that you have chosen.  Intel are unfortunately unable to assist with the porting process, so it must be done by the user's own development team.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

The Intel deserializer is only supplied for very large volume orders, so most D457 users who are using GMSL will need to obtain a third-party deserializer and port the driver for their particular hardware.  An exception to this is the D457-compatible **Connect Tech** system, which does not require a deserializer and has its own driver.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/10984#issuecomment-1363709490

closed 2023-11-09 by MartyG-RealSense