realsenseai/realsense-ros #2913: Missing namespace or camera name from topics

dorinclisu, 2023-10-26:
### Version
``` 
$ sudo apt show ros-iron-realsense2-camera

Package: ros-iron-realsense2-camera
Version: 4.54.1-2jammy.20230908.184837
Priority: optional
Section: misc
Maintainer: LibRealSense ROS Team <librs.ros@intel.com>
Installed-Size: 1.388 kB
...
```

### Using ros2 launch
```
$ ros2 launch realsense2_camera rs_launch.py camera_namespace:=robot1 camera_name:=D455_1

[INFO] [launch]: All log files can be found below /home/dorin/.ros/log/2023-10-26-11-38-44-678852-Ubuntu22-21249
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [21252]
[realsense2_camera_node-1] [INFO] [1698309524.955549850] [D455_1.D455_1]: RealSense ROS v4.54.1
...
```
```
$ ros2 topic list

/D455_1/color/camera_info
/D455_1/color/image_raw
/D455_1/color/metadata
/D455_1/depth/camera_info
/D455_1/depth/image_rect_raw
/D455_1/depth/metadata
/D455_1/extrinsics/depth_to_color
/D455_1/imu
```
Notice that "robot1" is missing from topic names.

### Using ros2 run
```
$ ros2 run realsense2_camera realsense2_camera_node --ros-args -r __node:=D455_1 -r __ns:=robot1

[WARN] [1698309775.885258279] [rcl]: Namespace not remapped to a fully qualified name (found: robot1)
[ERROR] [1698309775.885311470] [rcl]: Failed to parse global arguments
terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
  what():  failed to initialize rcl: Couldn't parse remap rule: '-r __ns:=robot1'. Error: Expected lexeme type 19, got 18 at index 6, at ./src/rcl/lexer_lookahead.c:244, at ./src/rcl/arguments.c:371
[ros2run]: Aborted
```

### Using ros2 run with fully qualified namespace
```
$ ros2 run realsense2_camera realsense2_camera_node --ros-args -r __node:=D455_1 -r __ns:=/robot1
[INFO] [1698309842.862752407] [robot1.D455_1]: RealSense ROS v4.54.1
...
```
```
$ ros2 topic list

/robot1/color/camera_info
/robot1/color/image_raw
/robot1/color/metadata
/robot1/depth/camera_info
/robot1/depth/image_rect_raw
/robot1/depth/metadata
/robot1/extrinsics/depth_to_color
/robot1/extrinsics/depth_to_infra1
/robot1/extrinsics/depth_to_infra2
/robot1/imu
/robot1/infra1/camera_info
/robot1/infra1/image_rect_raw
/robot1/infra1/metadata
/robot1/infra2/camera_info
/robot1/infra2/image_rect_raw
/robot1/infra2/metadata
```

Notice that "D455_1" is missing from topic names.

MartyG-RealSense, 2023-10-26:
Hi @dorinclisu  How long has it been since you installed the 4.54.1 wrapper, please?  The camera namespace system described in the wrapper documentation was added to the wrapper on August 29 2023.  Before that August 29 update, topics were listed with the camera_namespace but without the camera_name, as described at https://github.com/IntelRealSense/realsense-ros/pull/2857 in 'before and after' topic lists.

dorinclisu, 2023-10-27:
I installed it 3 days ago. But if it was a problem in 4.54.1, should it be fixed in 4.54.2? This is the whole point of tags / releases so that there is no time ambiguity.

MartyG-RealSense, 2023-10-27:
There is not a 4.54.2 ROS wrapper.  librealsense SDK version 2.54.2 was a minor 'patch' release focused only on bug-fixes, so wrapper 4.54.1 should work fine with 2.54.2 even though it was designed for 2.54.1.

If you installed the wrapper 3 days ago then the camera namespace changes will already be in the wrapper that has been installed on your computer and the topics should have been displaying with both the camera name and namespace.

MartyG-RealSense, 2023-11-10:
Case closed due to no further comments received.

dorinclisu, 2023-11-13:
> If you installed the wrapper 3 days ago then the camera namespace changes will already be in the wrapper that has been installed on your computer and the topics should have been displaying with both the camera name and namespace.

You are telling me what "should" be, I'm telling you what is. Why don't you try a fresh install and see for yourself?

MartyG-RealSense, 2023-11-13:
I will re-open the issue and have highlighted it to the RealSense ROS development team to seek their advice.  Thanks very much for your patience.

MartyG-RealSense, 2023-11-15:
Hi @dorinclisu  The RealSense ROS team informed me that a package that includes the camera_name feature has not been released at the time of writing this, but you can build the wrapper from source code and re-try running the same commands.

closed 2023-12-04 by MartyG-RealSense