realsenseai/realsense-ros #2917: What are the compatible versions of SDK, Wrapper, and Firmware for D435i camera when working with ROS2 Foxy, Ubuntu LTS 20.04 and Krenel 5.15.0.87.97~20.04.45 (Generic)?

mehradmrt, 2023-10-30:
I am struggling with using the right versions of SDK, Wrapper and Firmware on my system so that  I can properly deploy the D435i camera in the ROS2 environment.

 I can run the camera but have some warnings and errors which might be due to unstable or incompatible versions of drivers. I would appreciate getting some advice on what to do to get the most stable result on my system. 

Using /camera/aligned_depth_to_color/image_raw, I receive the depth frame which is aligned RGB but the depth frames often drop from 1280-720 to the default which is 848-480.

> ros2 launch realsense2_camera rs_launch.py align_depth:=true
> [INFO] [launch]: All log files can be found below /home/ddd/.ros/log/2023-10-22-15-21-53-856657-ddd-NUC11TNHi7-94159
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [realsense2_camera_node-1]: process started with pid [94167]
> [realsense2_camera_node-1] [INFO] [1698013314.101627775] [camera.camera]: RealSense ROS v3.2.3
> [realsense2_camera_node-1] [INFO] [1698013314.101784214] [camera.camera]: Built with LibRealSense v2.51.1
> [realsense2_camera_node-1] [INFO] [1698013314.101806659] [camera.camera]: Running with LibRealSense v2.51.1
> [realsense2_camera_node-1] [INFO] [1698013314.129668598] [camera.camera]: Device with serial number 153222071475 was found.
> [realsense2_camera_node-1] 
> [realsense2_camera_node-1] [INFO] [1698013314.129719427] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0 was found.
> [realsense2_camera_node-1] [INFO] [1698013314.129728834] [camera.camera]: Device with name Intel RealSense D435I was found.
> [realsense2_camera_node-1] [INFO] [1698013314.129864013] [camera.camera]: Device with port number 4-3 was found.
> [realsense2_camera_node-1] [INFO] [1698013314.129877663] [camera.camera]: Device USB type: 3.2
> [realsense2_camera_node-1] [INFO] [1698013314.132804362] [camera.camera]: getParameters...
> [realsense2_camera_node-1] [INFO] [1698013314.133405190] [camera.camera]: setupDevice...
> [realsense2_camera_node-1] [INFO] [1698013314.133415803] [camera.camera]: JSON file is not provided
> [realsense2_camera_node-1] [INFO] [1698013314.133426723] [camera.camera]: Device Name: Intel RealSense D435I
> [realsense2_camera_node-1] [INFO] [1698013314.133433608] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0
> [realsense2_camera_node-1] [INFO] [1698013314.133440430] [camera.camera]: Device FW version: 05.13.00.55
> [realsense2_camera_node-1] [INFO] [1698013314.133446901] [camera.camera]: Device Product ID: 0x0B3A
> [realsense2_camera_node-1] [INFO] [1698013314.133452456] [camera.camera]: Enable PointCloud: Off
> [realsense2_camera_node-1] [INFO] [1698013314.133458035] [camera.camera]: Align Depth: On
> [realsense2_camera_node-1] [INFO] [1698013314.133463434] [camera.camera]: Sync Mode: On
> [realsense2_camera_node-1] [INFO] [1698013314.133523147] [camera.camera]: Device Sensors: 
> [realsense2_camera_node-1] [INFO] [1698013314.139005792] [camera.camera]: Stereo Module was found.
> [realsense2_camera_node-1] [INFO] [1698013314.148603916] [camera.camera]: RGB Camera was found.
> [realsense2_camera_node-1] [INFO] [1698013314.148972087] [camera.camera]: Motion Module was found.
> [realsense2_camera_node-1] [INFO] [1698013314.148992337] [camera.camera]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
> [realsense2_camera_node-1] [INFO] [1698013314.148998847] [camera.camera]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
> [realsense2_camera_node-1] [INFO] [1698013314.149004561] [camera.camera]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
> [realsense2_camera_node-1] [INFO] [1698013314.149010044] [camera.camera]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
> [realsense2_camera_node-1] [INFO] [1698013314.149015918] [camera.camera]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
> [realsense2_camera_node-1] [INFO] [1698013314.149021824] [camera.camera]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
> [realsense2_camera_node-1] [INFO] [1698013314.150019454] [camera.camera]: num_filters: 1
> [realsense2_camera_node-1] [INFO] [1698013314.150036053] [camera.camera]: Setting Dynamic reconfig parameters.
> [realsense2_camera_node-1] [WARN] [1698013314.161984362] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
> [realsense2_camera_node-1] [INFO] [1698013314.162183740] [camera.camera]: Done Setting Dynamic reconfig parameters.
> [realsense2_camera_node-1] [INFO] [1698013314.162630752] [camera.camera]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
> [realsense2_camera_node-1] [INFO] [1698013314.163654634] [camera.camera]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
> [realsense2_camera_node-1] [INFO] [1698013314.164559361] [camera.camera]: setupPublishers...
> [realsense2_camera_node-1] [INFO] [1698013314.166327345] [camera.camera]: setupStreams...
> [realsense2_camera_node-1] [INFO] [1698013314.205854660] [camera.camera]: SELECTED BASE:Depth, 0
> [realsense2_camera_node-1] [INFO] [1698013314.206992632] [camera.camera]: Device Serial No: 153222071475
> [realsense2_camera_node-1] [INFO] [1698013314.207020347] [camera.camera]: RealSense Node Is Up!
> [realsense2_camera_node-1] [WARN] [1698013314.337834508] [camera.camera]: 
> [realsense2_camera_node-1]  22/10 15:22:15,707 WARNING [140420707489536] (ds5-timestamp.cpp:69) UVC metadata payloads not available. Please refer to the installation chapter for details.
> [realsense2_camera_node-1]  22/10 15:22:15,801 ERROR [140420707489536] (backend-v4l2.cpp:417) Non-sequential Video and Metadata v4l buffers
> [realsense2_camera_node-1]  22/10 15:22:15,801 WARNING [140420707489536] (backend-v4l2.cpp:1165) Video frame dropped, video and metadata buffers inconsistency
> [realsense2_camera_node-1]  22/10 15:22:15,801 ERROR [140420707489536] (backend-v4l2.cpp:417) Non-sequential Video and Metadata v4l buffers
> [realsense2_camera_node-1]  22/10 15:22:15,801 WARNING [140420707489536] (backend-v4l2.cpp:1165) Video frame dropped, video and metadata buffers inconsistency
> ^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)

MartyG-RealSense, 2023-10-30:
Hi @mehradmrt  The match-up of librealsense SDK version 2.51.1 and ROS2 wrapper 3.2.3 should be compatible.  Your camera firmware driver version is also fine for use with 2.51.1.

The UVC metadata messages suggest that camera metadata may not be enabled in your librealsense installation.  If you built librealsense from source code with CMake and did not include the build flag **-DFORCE_RSUSB_BACKEND=TRUE** in the CMake build instruction then you should patch the 5.15 kernel using the kernel patch script **patch-realsense-ubuntu-lts-hwe.sh**.  This patch script adds metadata support for kernels 5.13 and 5.15.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-lts-hwe.sh

If librealsense is built from DKMS packages or if it is compiled from source code with the build flag **-DFORCE_RSUSB_BACKEND=TRUE** included then metadata support will be enabled.

*********

It is also worth mentioning that the 3.2.3 ROS2 wrapper is obsolete.  The '3.' series of wrappers was replaced by the '4.' series.  The appropriate ROS2 wrapper for use with Ubuntu 20.04, Foxy and librealsense 2.51.1 is **4.51.1**

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.51.1

mehradmrt, 2023-10-31:
@MartyG-RealSense Thanks a lot for your quick clarification. I reinstalled everything from scratch using the dkms package with `sudo apt install librealsense2=2.51.1-0~realsense0.7526 librealsense2-gl=2.51.1-0~realsense0.7526 librealsense2-net=2.51.1-0~realsense0.7526 librealsense2-utils=2.51.1-0~realsense0.7526`. And the wrapper with `sudo apt install ros-foxy-realsense2-*` . I was not sure how to include `-DFORCE_RSUSB_BACKEND=TRUE` during the dkms installation process. 
running the `realsense-viewer` I got:

>WARNING [140205984024320] (hdr-merge.cpp:38) HDR Merge filter cannot process frames because relevant metadata params are missing <

in my terminal log, which  possibly is related to what you previously mentioned about -DFORCE_RSUSB_BACKEND=TRUE.

Additionally, I still have a few Warnings from my ros2 package. are these warnings big issues? 

> ros2 launch realsense2_camera rs_launch.py align_depth.enable:=True
[INFO] [launch]: All log files can be found below /home/ddd/.ros/log/2023-10-31-12-14-36-098405-ddd-NUC11TNHi7-225944
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [225946]
[realsense2_camera_node-1] [INFO] [1698779676.256890722] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1698779676.256958431] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1698779676.256964667] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1698779676.316306277] [camera.camera]: Device with serial number 153222071475 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1698779676.316395363] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1698779676.316413233] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1698779676.316730013] [camera.camera]: Device with port number 4-3 was found.
[realsense2_camera_node-1] [INFO] [1698779676.316750959] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1698779676.319025560] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1698779676.319454216] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1698779676.319495978] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1698779676.319518857] [camera.camera]: Device Serial No: 153222071475
[realsense2_camera_node-1] [INFO] [1698779676.319540978] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1698779676.319565660] [camera.camera]: Device FW version: 05.13.00.55
[realsense2_camera_node-1] [INFO] [1698779676.319588868] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1698779676.319619193] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1698779676.332032018] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
**[realsense2_camera_node-1] [WARN] [1698779676.333028596] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.**
[realsense2_camera_node-1] [INFO] [1698779676.336627738] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1698779676.336975138] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1698779676.336996153] [camera.camera]: Set ROS param accel_fps to default: 100
**[realsense2_camera_node-1] [WARN] [1698779676.340608626] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.**
[realsense2_camera_node-1] [INFO] [1698779676.341177575] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1698779676.342634510] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1698779676.351424560] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1698779676.351521427] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1698779676.352343077] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1698779676.363939973] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1698779676.364275878] [camera.camera]: RealSense Node Is Up!
**[realsense2_camera_node-1] [WARN] [1698779676.484953324] [camera.camera]:**

mehradmrt, 2023-11-07:
I removed previous builts and built them again as you suggested. The packages seem to be working well and are stable. As you emphasized, I wouldn't care about the power line frequency and hdr at this point. 

Another problem that I am currently dealing with is related to pointcloud. I desire to get an organized pointcloud with the same height and width as the rgb or depth frames (1280x720) but still get unorganized points with height=1. Reviewed some other issues discussed in the repository but still no luck. If I should close this and initiate a different issue, will be happy to do so. Here are the terminal logs:

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 rgb_camera.profile:=1280x720x30 pointcloud.enable:=true pointcloud.ordered_pc:=true`

> [INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [313004]
[realsense2_camera_node-1] [INFO] [1699400447.460262147] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1699400447.460425352] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1699400447.460453394] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1699400447.472660953] [camera.camera]: Device with serial number 153222071475 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1699400447.472719401] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1699400447.472734229] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1699400447.473171133] [camera.camera]: Device with port number 4-3 was found.
[realsense2_camera_node-1] [INFO] [1699400447.473214679] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1699400447.475451178] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1699400447.475908253] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1699400447.475954158] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1699400447.475980865] [camera.camera]: Device Serial No: 153222071475
[realsense2_camera_node-1] [INFO] [1699400447.476019960] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1699400447.476045278] [camera.camera]: Device FW version: 05.13.00.55
[realsense2_camera_node-1] [INFO] [1699400447.476068663] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1699400447.476090251] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [**WARN] [1699400447.488699329] [camera.camera]: re-enable the stream for the change to take effect.**
[realsense2_camera_node-1] [WARN] [1699400447.489612141] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1699400447.493141842] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1699400447.493168851] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] **[WARN] [1699400447.496704620] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.**
[realsense2_camera_node-1] [INFO] [1699400447.497270202] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1699400447.498615620] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1699400447.508201148] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1699400447.508269947] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1699400447.508835379] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1699400447.520354438] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1699400447.520786828] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1699400447.646344974] [camera.camera]:

`ros2 topic echo /camera/depth/color/points`:

> header:
  stamp:
    sec: 1699397627
    nanosec: 227046912
  frame_id: camera_color_optical_frame
height: 1
width: 388199
fields:
 name: x
  offset: 0
  datatype: 7
  count: 1
 name: y
  offset: 4
  datatype: 7
  count: 1
 name: z
  offset: 8
  datatype: 7
  count: 1
 name: rgb
  offset: 16
  datatype: 7
  count: 1
is_bigendian: false
point_step: 20
row_step: 7763980
data:
172
...

mehradmrt, 2023-11-08:
Added **align_depth.enable:=true** but no difference was observed.

> header:
  stamp:
    sec: 1699470102
    nanosec: 836443136
  frame_id: camera_color_optical_frame
height: 1
width: 851445
fields:
name: x
...

mehradmrt, 2023-11-08:
I resolved the issue by setting the ordered_pc parameter to true for a second time during the runtime:
`ros2 param set /camera/camera pointcloud.ordered_pc true`

`ros2 topic echo /camera/depth/color/points`:
> header:
  stamp:
    sec: 1699472043
    nanosec: 138851840
  frame_id: camera_color_optical_frame
**height: 720
width: 1280**
...

mehradmrt, 2023-11-08:
However, I run many other packages alongside realsense, so this might not be an efficient approach. If there is another way to avoid setting parameters during runtime, that would be great.

MartyG-RealSense, 2023-11-09:
Do you still have to perform **ros2 param set** during runtime if you reset the camera during the launch process by adding **initial_reset:=true** onto the end of your `ros2 launch` instruction?

mehradmrt, 2023-11-09:
I tried it but did not enable the ordered_pc, still setting the parameter during the runtime is needed.

MartyG-RealSense, 2023-11-10:
The ROS2 documentation link below suggests that if you set the parameter during runtime then you can then save it to a param file and call it during launch so that you do not need to enter the parameter command again during runtime.

https://docs.ros.org/en/eloquent/Tutorials/Parameters/Understanding-ROS2-Parameters.html#ros2-param-dump

It looks as though to use this method you would have to launch the ROS2 wrapper with **ros2 run** instead of ros2 launch.  for example:

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p align_depth.enable:=true -p pointcloud.enable:=true `

ordered_pc would be set inside the param dump file rather than the launch instruction.

mehradmrt, 2023-11-15:
Thank you @MartyG-RealSense for your responses. I need to work with the launch file to input some other parameters such as json_file_path. However, I will close this issue since it is no longer relevant to the topic and the previous solutions you provided are working well. Will open new issues if needed!

closed 2023-11-15 by mehradmrt