realsenseai/realsense-ros #2919: RGB Frame not Working

100ws, 2023-10-31:
[ INFO] [1698736173.236593332]: Motion Module was found.
[ INFO] [1698736173.236691297]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1698736173.236774204]: num_filters: 0
[ INFO] [1698736173.236846362]: Setting Dynamic reconfig parameters.
[ INFO] [1698736175.582563625]: Done Setting Dynamic reconfig parameters.
[ INFO] [1698736175.584436516]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1698736175.586079604]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1698736175.586116919]: setupPublishers...
[ INFO] [1698736175.590209085]: Expected frequency for depth = 30.00000
[ INFO] [1698736175.622856650]: Expected frequency for color = 30.00000
[ INFO] [1698736175.639086747]: setupStreams...
 31/10 12:39:35,777 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1698736175.876283179]: SELECTED BASE:Depth, 0
[ INFO] [1698736175.914868204]: RealSense Node Is Up!
 31/10 12:39:35,930 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ WARN] [1698736176.005265824]: 
 31/10 12:39:36,005 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:39:36,056 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:39:36,207 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:39:38,366 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:39:39,417 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:39:41,470 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:39:42,522 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:39:44,574 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:39:48,629 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:39:52,683 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:41:11,802 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:41:15,856 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:42:58,998 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:42:59,049 WARNING [139848511784512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 31/10 12:43:22,649 ERROR [139847713375808] (ds5-options.cpp:88) Asic Temperature value is not valid!

MartyG-RealSense, 2023-10-31:
Hi @100ws  Which version of the RealSense ROS wrapper are you using, please?  And are you using librealsense **2.38.1** with the ROS wrapper as your other issue at https://github.com/IntelRealSense/librealsense/issues/12339 suggests?

If you are using librealsense 2.38.1 then the ROS wrapper version that should be matched to it is ROS1 wrapper **2.2.17** or ROS2 wrapper **3.1.1**.

The camera firmware driver version that should be used with 2.38.1 is **5.12.7.100**.

100ws, 2023-10-31:
I am using the same version that you said 2.38.1 and 5.12.7.100.
> Hi @100ws Which version of the RealSense ROS wrapper are you using, please? And are you using librealsense **2.38.1** with the ROS wrapper as your other issue at [IntelRealSense/librealsense#12339](https://github.com/IntelRealSense/librealsense/issues/12339) suggests?
> 
> If you are using librealsense 2.38.1 then the ROS wrapper version that should be matched to it is ROS1 wrapper **2.2.17** or ROS2 wrapper **3.1.1**.
> 
> The camera firmware driver version that should be used with 2.38.1 is **5.12.7.100**.

MartyG-RealSense, 2023-10-31:
When the **control_transfer returned** warning is generating continuously in the ROS terminal, it indicates a communication problem between the camera and computer.  The **Resource temporarily unavailable, number: 11** type of control_transfer warning can suggest that either new frames are not arriving - like in your RealSense Viewer problem at https://github.com/IntelRealSense/librealsense/issues/12339 - or the computer cannot keep up with processing the volume of data arriving from the camera at the computer's USB port.

Do the warnings still occur if you set a lower stream configuration of 848x480 and 15 FPS for both depth and color using the launch instruction below?

`roslaunch realsense2_camera rs_camera.launch depth_width:=848 depth_height:=480 depth_fps:=15 color_width:=848 color_height:=480 color_fps:=15`

100ws, 2023-11-01:
> When the **control_transfer returned** warning is generating continuously in the ROS terminal, it indicates a communication problem between the camera and computer. The **Resource temporarily unavailable, number: 11** type of control_transfer warning can suggest that either new frames are not arriving - like in your RealSense Viewer problem at [IntelRealSense/librealsense#12339](https://github.com/IntelRealSense/librealsense/issues/12339) - or the computer cannot keep up with processing the volume of data arriving from the camera at the computer's USB port.
> 
> Do the warnings still occur if you set a lower stream configuration of 848x480 and 15 FPS for both depth and color using the launch instruction below?
> 
> `roslaunch realsense2_camera rs_camera.launch depth_width:=848 depth_height:=480 depth_fps:=15 color_width:=848 color_height:=480 color_fps:=15`

process[camera/realsense2_camera_manager-2]: started with pid [42573]
process[camera/realsense2_camera-3]: started with pid [42574]
[ INFO] [1698815511.379559741]: Initializing nodelet with 6 worker threads.
[ INFO] [1698815511.753688138]: RealSense ROS v2.3.2
[ INFO] [1698815511.753710320]: Built with LibRealSense v2.50.0
[ INFO] [1698815511.753719277]: Running with LibRealSense v2.50.0
[ INFO] [1698815511.811307246]:  
[ INFO] [1698815512.065501226]: Device with serial number 052522070765 was found.

[ INFO] [1698815512.065533388]: Device with physical ID 2-8-2 was found.
[ INFO] [1698815512.065542630]: Device with name Intel RealSense D435I was found.
[ INFO] [1698815512.065800503]: Device with port number 2-8 was found.
[ INFO] [1698815512.065826544]: Device USB type: 3.2
[ INFO] [1698815512.067729921]: getParameters...
[ INFO] [1698815512.095086647]: setupDevice...
[ INFO] [1698815512.095105090]: JSON file is not provided
[ INFO] [1698815512.095112814]: ROS Node Namespace: camera
[ INFO] [1698815512.095121590]: Device Name: Intel RealSense D435I
[ INFO] [1698815512.095127931]: Device Serial No: 052522070765
[ INFO] [1698815512.095138569]: Device physical port: 2-8-2
[ INFO] [1698815512.095145526]: Device FW version: 05.15.01.00
[ INFO] [1698815512.095151619]: Device Product ID: 0x0B3A
[ INFO] [1698815512.095157385]: Enable PointCloud: Off
[ INFO] [1698815512.095165968]: Align Depth: Off
[ INFO] [1698815512.095172108]: Sync Mode: Off
[ INFO] [1698815512.095203163]: Device Sensors: 
[ INFO] [1698815512.196512882]: Stereo Module was found.
[ INFO] [1698815512.219673286]: RGB Camera was found.
[ INFO] [1698815512.219881577]: Motion Module was found.
[ INFO] [1698815512.219972164]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1698815512.220005397]: num_filters: 0
[ INFO] [1698815512.220028397]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1698815515.222596174]: Done Setting Dynamic reconfig parameters.
[ INFO] [1698815515.224434539]: depth stream is enabled - width: 848, height: 480, fps: 15, Format: Z16
[ INFO] [1698815515.224950366]: color stream is enabled - width: 848, height: 480, fps: 15, Format: RGB8
[ INFO] [1698815515.224990405]: setupPublishers...
[ INFO] [1698815515.228580907]: Expected frequency for depth = 15.00000
[ INFO] [1698815515.260653580]: Expected frequency for color = 15.00000
[ INFO] [1698815515.276139329]: setupStreams...
 01/11 10:41:55,504 WARNING [140193159964224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1698815515.557055623]: SELECTED BASE:Depth, 0
[ INFO] [1698815515.584771063]: RealSense Node Is Up!
 01/11 10:41:55,658 WARNING [140193159964224] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ WARN] [1698815515.679591109]: 
 01/11 10:41:55,708 WARNING [140193159964224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/11 10:41:55,759 WARNING [140193159964224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/11 10:41:55,809 WARNING [140193159964224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/11 10:41:55,860 WARNING [140193159964224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

Same error is coming

MartyG-RealSense, 2023-11-01:
Does the same error occur if you add **initial_reset:=true** to the launch instruction to reset the camera at launch?

100ws, 2023-11-02:
> Does the same error occur if you add **initial_reset:=true** to the launch instruction to reset the camera at launch?

roslaunch realsense2_camera rs_camera.launch depth_width:=848 depth_height:=480 depth_fps:=15 color_width:=848 color_height:=480 color_fps:=15 initial_reset:=true 
... logging to /home/computer-vision/.ros/log/58c06124-7939-11ee-af6a-9f721a77197e/roslaunch-computervision-MS-7A38-4894.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://computervision-MS-7A38:41723/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 848
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 848
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 98 lines cut ...]
[ INFO] [1698899781.131073193]: setupStreams...
 02/11 10:06:21,295 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1698899781.346093091]: SELECTED BASE:Depth, 0
[ INFO] [1698899781.376232650]: RealSense Node Is Up!
[ WARN] [1698899781.473315032]: 
 02/11 10:06:21,473 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/11 10:06:21,524 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/11 10:06:21,574 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/11 10:06:21,625 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/11 10:06:22,077 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/11 10:06:22,127 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/11 10:06:22,279 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/11 10:06:22,330 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[... cut ...]
 02/11 10:06:23,138 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 02/11 10:07:57,474 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/11 10:07:57,525 WARNING [140388354471488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

Same error is coming

MartyG-RealSense, 2023-11-02:
https://github.com/IntelRealSense/realsense-ros/issues/2535#issuecomment-1305386638 describes how in past cases of the **control_transfer returned error, index: 768, error: Success, number: 0** warning, RealSense user's found that building the librealsense SDK in RSUSB = **false** mode resolved the problem.

MartyG-RealSense, 2023-11-09:
Hi @100ws  Do you require further assistance with this case, please?  Thanks!

100ws, 2023-11-09:
> Hi @100ws Do you require further assistance with this case, please? Thanks!

Yes How should I process it for return if I buy it from amazon

MartyG-RealSense, 2023-11-09:
You can log in to your Amazon account and go to the **Your Orders** section of your account settings to find the list of items that you have ordered. Find the listing for the item that you want to return and there should be a 'Return Items' option for returning an item.

https://www.amazon.com/b?ie=UTF8&node=18726306011

The settings and their names may vary depending on which country's Amazon site is being used.

100ws, 2023-11-16:
> You can log in to your Amazon account and go to the **Your Orders** section of your account settings to find the list of items that you have ordered. Find the listing for the item that you want to [...]

But in amazon return there is only 1 week replacement of product how should I contact to intel store to exchange this within warranty period

MartyG-RealSense, 2023-11-16:
The official Intel RealSense store is only responsible for returns of products that were purchased from it and not from other retailers, unfortunately.  

You could try contacting Amazon through their customer service **Help** page to talk to an Amazon customer support representative directly.

https://www.amazon.com/gp/help/customer/display.html?nodeId=508510&ref_=footer_gw_m_b_he

closed 2023-12-05 by MartyG-RealSense