realsenseai/realsense-ros #2920: Converting image_raw topic from a ros2 bag to video using ros2bag2video.py script gets stuck for a short video capture

monajalal, 2023-11-01:
I am using this command and it gets stuck after seeing this message:
```
(base) mona@ada:~/rosbag2video$ ./ros2bag2video.py -t /camera/camera/color/image_raw ~/rosbag2_2023_11_01-15_52_42/
FPS (int) =  25
Rate (float) =  1.0
Topic (str) =  /camera/camera/color/image_raw
Output File (str) =  output.mp4
Verbose (bool) =  False
Reading info from bag file: /home/mona/rosbag2_2023_11_01-15_52_42/
ROS Message name =  sensor_msgs/msg/Image
Image count =  1295
msgtype =  <class 'sensor_msgs.msg._image.Image'>
AJB: subscribing to msg:  <class 'sensor_msgs.msg._image.Image'> on topic:  /camera/camera/color/image_raw
Starting ROS bag playback...
[INFO] [1698871197.021297370] [rosbag2_storage]: Opened database '/home/mona/rosbag2_2023_11_01-15_52_42/rosbag2_2023_11_01-15_52_42_0.db3' for READ_ONLY.
[INFO] [1698871197.021376685] [rosbag2_player]: Set rate to 1
[INFO] [1698871197.023444441] [rosbag2_player]: Adding keyboard callbacks.
[INFO] [1698871197.023484055] [rosbag2_player]: Press SPACE for Pause/Resume
[INFO] [1698871197.023497198] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
[INFO] [1698871197.023505395] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
[INFO] [1698871197.023510402] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%
[INFO] [1698871197.023983667] [rosbag2_storage]: Opened database '/home/mona/rosbag2_2023_11_01-15_52_42/rosbag2_2023_11_01-15_52_42_0.db3' for READ_ONLY.

```

Could you please help me to know if something was wrong with the ros2 recording? and also how should I fix this?
```
(base) mona@ada:~$ ros2 bag info rosbag2_2023_11_01-15_52_42/

Files:             rosbag2_2023_11_01-15_52_42_0.db3
Bag size:          2.6 GiB
Storage id:        sqlite3
Duration:          86.343s
Start:             Nov  1 2023 15:52:42.240 (1698868362.240)
End:               Nov  1 2023 15:54:08.583 (1698868448.583)
Messages:          10373
Topic information: Topic: /goal_pose | Type: geometry_msgs/msg/PoseStamped | Count: 0 | Serialization Format: cdr
                   Topic: /camera/camera/extrinsics/depth_to_color | Type: realsense2_camera_msgs/msg/Extrinsics | Count: 1 | Serialization Format: cdr
                   Topic: /events/write_split | Type: rosbag2_interfaces/msg/WriteSplitEvent | Count: 0 | Serialization Format: cdr
                   Topic: /camera/camera/color/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1295 | Serialization Format: cdr
                   Topic: /tf_static | Type: tf2_msgs/msg/TFMessage | Count: 1 | Serialization Format: cdr
                   Topic: /camera/camera/aligned_depth_to_color/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1295 | Serialization Format: cdr
                   Topic: /camera/camera/color/metadata | Type: realsense2_camera_msgs/msg/Metadata | Count: 1295 | Serialization Format: cdr
                   Topic: /camera/camera/depth/image_rect_raw | Type: sensor_msgs/msg/Image | Count: 1295 | Serialization Format: cdr
                   Topic: /rosout | Type: rcl_interfaces/msg/Log | Count: 10 | Serialization Format: cdr
                   Topic: /parameter_events | Type: rcl_interfaces/msg/ParameterEvent | Count: 0 | Serialization Format: cdr
                   Topic: /camera/camera/depth/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1295 | Serialization Format: cdr
                   Topic: /initialpose | Type: geometry_msgs/msg/PoseWithCovarianceStamped | Count: 0 | Serialization Format: cdr
                   Topic: /camera/camera/color/image_raw | Type: sensor_msgs/msg/Image | Count: 1295 | Serialization Format: cdr
                   Topic: /camera/camera/aligned_depth_to_color/image_raw | Type: sensor_msgs/msg/Image | Count: 1295 | Serialization Format: cdr
                   Topic: /camera/camera/depth/metadata | Type: realsense2_camera_msgs/msg/Metadata | Count: 1295 | Serialization Format: cdr
                   Topic: /camera/camera/extrinsics/depth_to_depth | Type: realsense2_camera_msgs/msg/Extrinsics | Count: 1 | Serialization Format: cdr
                   Topic: /clicked_point | Type: geometry_msgs/msg/PointStamped | Count: 0 | Serialization Format: cdr
```

When I do the same command for the depth stream, I actually get an error.

My goal is to convert the aligned depth video and color raw image from a ros bag I recorded (using Intel RealSense D435 camera) to video format (each a separate format).

monajalal, 2023-11-02:
so we did use echo today and we see this topic has content in it however, we still get this error:
```
(base) mona@ada:~/rosbag2video$ /usr/bin/python3.10 ros2bag2video.py -t /camera/camera/color/image_raw ~/ros2bagmona/
FPS (int) =  25
Rate (float) =  1.0
Topic (str) =  /camera/camera/color/image_raw
Output File (str) =  output.mp4
Verbose (bool) =  False
Reading info from bag file: /home/mona/ros2bagmona/
ROS Message name =  sensor_msgs/msg/Image
Image count =  198
msgtype =  <class 'sensor_msgs.msg._image.Image'>
AJB: subscribing to msg:  <class 'sensor_msgs.msg._image.Image'> on topic:  /camera/camera/color/image_raw
Starting ROS bag playback...
[INFO] [1698940471.288059558] [ros2bag2videos]: Image Received [1/198]
Traceback (most recent call last):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/cv_bridge/core.py", line 141, in compressed_imgmsg_to_cv2
    res = cvtColor2(im, 'bgr8', desired_encoding)
RuntimeError: OpenCV(4.5.4) ./modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mona/rosbag2video/ros2bag2video.py", line 430, in <module>
    main(sys.argv)
  File "/home/mona/rosbag2video/ros2bag2video.py", line 422, in main
    rclpy.spin(videowriter)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
    executor.spin_once()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 713, in spin_once
    raise handler.exception()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
    await call_coroutine(entity, arg)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 343, in _execute_subscription
    await await_or_execute(sub.callback, msg)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
    return callback(*args)
  File "/home/mona/rosbag2video/ros2bag2video.py", line 355, in listener_callback
    img = self.bridge.compressed_imgmsg_to_cv2(msg, self.msg_fmt)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/cv_bridge/core.py", line 143, in compressed_imgmsg_to_cv2
    raise CvBridgeError(e)
cv_bridge.core.CvBridgeError: OpenCV(4.5.4) ./modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

[INFO] [1698940471.417141092] [rosbag2_storage]: Opened database '/home/mona/ros2bagmona/ros2bagmona_0.db3' for READ_ONLY.
[INFO] [1698940471.417225980] [rosbag2_player]: Set rate to 1
[INFO] [1698940471.419538419] [rosbag2_player]: Adding keyboard callbacks.
[INFO] [1698940471.419576491] [rosbag2_player]: Press SPACE for Pause/Resume
[INFO] [1698940471.419591379] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
[INFO] [1698940471.419599541] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
[INFO] [1698940471.419609794] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%
[INFO] [1698940471.420004717] [rosbag2_storage]: Opened database '/home/mona/ros2bagmona/ros2bagmona_0.db3' for READ_ONLY.
(base) mona@ada:~/rosbag2video$ Error in tcsetattr old_term_settings. errno = 5
Caught exception: "Error in read(). errno = 5"

```

I really appreciate it if you could provide some guidance on this.

MartyG-RealSense, 2023-11-03:
Hi @monajalal  Posting your question on the **rosbag2video** issues forum as you did at https://github.com/mlaiacker/rosbag2video/issues/36 was likely the best course of action for obtaining a solution for your question, as the tool is not an Intel product.

A simpler solution than using rosbag2video may be to use a **video capture** program, where you start bag playback running and then activate the video capture program to capture the live contents of the screen as a video file.  The links below have advice about video capture of the screen in Ubuntu.

https://beebom.com/how-record-screen-ubuntu/

https://help.ubuntu.com/stable/ubuntu-help/screen-shot-record.html

monajalal, 2023-11-03:
@MartyG-RealSense  thanks a lot for your response. I also asked the question in much greater detail here in ROS StackExchange in the hope we get some response from the community. I'll keep you posted when I find a solution.

https://robotics.stackexchange.com/questions/105191/ros2-humble-converting-a-color-raw-bag-file-to-a-series-of-consecutive-images-o

MartyG-RealSense, 2023-11-04:
You are very welcome.  Thanks very much for the update!

monajalal, 2023-11-06:
@MartyG-RealSense  I ended up using `rosbags` package instead and it worked for me.

closed 2023-11-06 by monajalal

MartyG-RealSense, 2023-11-07:
It's great to hear that you found a solution that worked for you.  Thanks very much for the update!