realsenseai/realsense-ros #2921: Launch Description launching realsense does not enable hole fillinng filter

hanschanhs, 2023-11-03:
I have written a launch file which should just realsense launch file that is shipped with this repo
`def generate_launch_description():
    camera_launch = IncludeLaunchDescription(
                        PythonLaunchDescriptionSource([
                            PathJoinSubstitution([
                                FindPackageShare('realsense2_camera'), 'launch', 'rs_launch.py'
                                ])
                            ]),
                        launch_arguments={
                            'align_depth.enable': 'true',
                            'hole_filling_filter.enable': 'true',
                            }.items(),
                        )
   return LaunchDescription([
        camera_launch,
        ])
`
Expected behaviour: 
hole filling filter and align depth filter both enabled when launch file containing the above launch description is run

Observed behaviour
Only the align_depth_enable is correctly enabled. If align_depth_enable is set to false or commented out (default config), the parameter is set to false as expected - but the hole filling setting in the launch arguments seem to have made no difference.

Thanks in advance!

MartyG-RealSense, 2023-11-04:
Hi @hanschanhs Is the hole-filling filter applied successfully if you set it with a **ros2 param set** command after launch has completed, using a command such as the one below?

`ros2 param set /camera depth_module.hole_filling_filter true`

hanschanhs, 2023-11-04:
Yes - it does turn on if I manually change it after launch has been run

hanschanhs, 2023-11-04:
I can also confirm that if i run ros2 launch with the argument hole_filling_enable:=true that the hole filling filter is also not enabled

MartyG-RealSense, 2023-11-04:
Let's try adding the **Spatial Filter** to your launch file and setting its **Hole Filling** parameter to '5' (unlimited-strength hole filling).

```
'spatial_filter.enable': 'true',
'spatial_filter.hole_filling': '5',
```

Does that enable the Spatial Filter and make a difference to the image?

MartyG-RealSense, 2023-11-13:
Hi @hanschanhs  Do you require further assistance with this case, please?  Thanks!

TemugeB, 2023-11-16:
Hi. I'm having a similar problem with ROS2 Humble. Trying to turn on temporal filter fails with this command:

```ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true temporal_filter.enable:=true```
Also tried
```ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true filters:=temporal```

The pointcloud is enabled fine but temporal filter does not turn on. The temporal filter can be turned on with ros2 param:
```ros2 param set /camera/camera temporal_filter.enable true```

The fix appears to be to add the parameter in the ``rs_launch.py`` file. Specifically, I added 
```{'name': 'temporal_filter.enable', 'default': 'false', 'description': 'enable temporal filter'}```
to ```configurable_parameters``` list in ```rs_launch.py```. 

After this change, temporal filter can be enabled from launch with:
```ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true temporal_filter.enable:=true```

ROS2 Humble has a strange behavior when passing in parameters to a node. For example, if I have ```launchfile_A``` and I include it in ```launchfile_B```, then the default parameters of ```launchfile_A``` is not loaded. In other words, the parameters of ```launchfile_A``` need to be listed again in ```launchfile_B```. Not sure if this is intended behavior or not. But it is how it is currently.

MartyG-RealSense, 2023-11-16:
Thanks so much @TemugeB for sharing your solution!

closed 2023-11-23 by MartyG-RealSense