realsenseai/realsense-ros #2923: Point cloud frequency issue on realsense D435

TopRamen1, 2023-11-08:
I'm running a basic rosrun on an 

Platform: Intel NUC SBC
Realsense-ros: latest ros2-devel
Command: ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true

Outcome:
ros2 run logs:
[image]

ros2 topic hz:
[image]

The expected frequency is 30 hz but I'm getting on average: 11 Hz

Is this a known issue? And if so what could be a solution?

MartyG-RealSense, 2023-11-08:
Hi @TopRamen1  Depending on the capabilities of the particular computer, having depth, color and both infrared topics enabled simultaneously could be a processing burden on the computer's CPU.  

If you do not require the infra1 and infra2 topics that are being published then you can reduce the CPU burden by disabling the infrared topics.  Please try adding **-p enable_infra1:=false -p enable_infra2:=false** to your ros2 run launch instruction.

TopRamen1, 2023-11-08:
Hi, thanks for a quick response.

My actual desired setup is as follows:

ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true -p enable_color:=false -p enable_infra1:=false -p enable_infra2:=false -p enable_infra:=false -p depth_module.profile:=848x480x6

In this case, I disabled both infra and color images since I'm interested only in pointcloud and depth.
Additionally, I configured the driver to publish on 6 Hz since it would be enough for my usecase

In this case, the outcome is even worse:
[image]

It's strange for me because ROS 1 driver with the same cameras is able publish on steady 6 Hz on analogical settings.

So my question now is this a known issue of the driver to have there publishing frequencies differences?

MartyG-RealSense, 2023-11-08:
There can be FPS drops or other issues sometimes when the pointcloud is enabled.  

Is the FPS 30 if you do not enable the pointcloud, please?

TopRamen1, 2023-11-08:
I have chacked the frequency with just the ros run command without any arguments

ros2 run realsense2_camera realsense2_camera_node

Output ros2 topic hz /camera/camera/color/image_raw

[image]

The FPS is still not 30

TopRamen1, 2023-11-09:
The auto_exposure_priority did not improve the FPS when i ran with and without the point cloud enabled

MartyG-RealSense, 2023-11-09:
How does the FPS perform when the pointcloud is enabled with **ros2 launch** instead of ros2 run?

`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true`

TopRamen1, 2023-11-10:
When i run with the ros2 launch command you have provided the fps is even more unstable.

I managed to see 30 FPS shortly after the start of the driver but is declined sharply:

average rate: 29.941
	min: 0.031s max: 0.037s std dev: 0.00091s window: 32
average rate: 29.943
	min: 0.019s max: 0.047s std dev: 0.00258s window: 62
average rate: 29.956
	min: 0.019s max: 0.047s std dev: 0.00216s window: 92
average rate: 29.963
	min: 0.019s max: 0.047s std dev: 0.00190s window: 122
average rate: 29.963
	min: 0.019s max: 0.047s std dev: 0.00171s window: 152
average rate: 29.968
	min: 0.019s max: 0.047s std dev: 0.00158s window: 182
average rate: 29.969
	min: 0.019s max: 0.047s std dev: 0.00149s window: 213
average rate: 29.971
	min: 0.019s max: 0.047s std dev: 0.00140s window: 243
average rate: 29.969
	min: 0.019s max: 0.047s std dev: 0.00139s window: 273
average rate: 29.972
	min: 0.019s max: 0.047s std dev: 0.00133s window: 304
average rate: 29.973
	min: 0.019s max: 0.047s std dev: 0.00127s window: 334
average rate: 29.973
	min: 0.019s max: 0.047s std dev: 0.00122s window: 364
average rate: 29.972
	min: 0.019s max: 0.047s std dev: 0.00118s window: 394
average rate: 29.973
	min: 0.019s max: 0.047s std dev: 0.00120s window: 424
average rate: 29.974
	min: 0.019s max: 0.047s std dev: 0.00116s window: 454
average rate: 29.975
	min: 0.019s max: 0.047s std dev: 0.00114s window: 484
average rate: 27.803
	min: 0.019s max: 1.001s std dev: 0.04576s window: 500
average rate: 23.697
	min: 0.019s max: 2.669s std dev: 0.12629s window: 506
average rate: 19.971
	min: 0.019s max: 4.034s std dev: 0.21746s window: 507
average rate: 18.939
	min: 0.019s max: 4.034s std dev: 0.22576s window: 508
average rate: 17.298
	min: 0.019s max: 4.034s std dev: 0.25222s window: 509
average rate: 16.689
	min: 0.019s max: 4.034s std dev: 0.25643s window: 510
average rate: 15.930
	min: 0.019s max: 4.034s std dev: 0.25903s window: 515
average rate: 15.421
	min: 0.019s max: 4.034s std dev: 0.26303s window: 516
average rate: 14.578
	min: 0.019s max: 4.034s std dev: 0.27621s window: 517
average rate: 14.196
	min: 0.019s max: 4.034s std dev: 0.27714s window: 520
average rate: 13.794
	min: 0.019s max: 4.034s std dev: 0.27781s window: 526
average rate: 12.826
	min: 0.019s max: 4.034s std dev: 0.30279s window: 528
average rate: 12.466
	min: 0.019s max: 4.034s std dev: 0.30689s window: 529
average rate: 11.726
	min: 0.019s max: 4.034s std dev: 0.32174s window: 532
average rate: 11.326
	min: 0.019s max: 4.034s std dev: 0.32803s window: 535
average rate: 11.046
	min: 0.019s max: 4.034s std dev: 0.32913s window: 538
average rate: 10.827
	min: 0.019s max: 4.034s std dev: 0.33034s window: 540
[... 80 lines cut ...]
average rate: 9.428
	min: 0.030s max: 8.475s std dev: 0.52938s window: 562
average rate: 9.592
	min: 0.030s max: 8.475s std dev: 0.52031s window: 582
average rate: 9.841
	min: 0.030s max: 8.475s std dev: 0.50965s window: 607
average rate: 10.030
	min: 0.030s max: 8.475s std dev: 0.50078s window: 629
average rate: 10.139
	min: 0.030s max: 8.475s std dev: 0.49420s window: 646
average rate: 10.266
	min: 0.030s max: 8.475s std dev: 0.48717s window: 665
average rate: 10.490
	min: 0.030s max: 8.475s std dev: 0.47839s window: 690
average rate: 10.732
	min: 0.030s max: 8.475s std dev: 0.46943s window: 717
average rate: 10.941
	min: 0.030s max: 8.475s std dev: 0.46156s window: 742
average rate: 11.068
	min: 0.030s max: 8.475s std dev: 0.45552s window: 762
average rate: 11.167
	min: 0.030s max: 8.475s std dev: 0.45028s window: 780
average rate: 10.571
	min: 0.030s max: 8.475s std dev: 0.47156s window: 781
average rate: 10.611

TopRamen1, 2023-11-10:
I'm running the driver from a ros humble container on an Intel NUC with ubuntu 20.04 installed with kernel: 5.13.0-28-generic

Could this information help you?

MartyG-RealSense, 2023-11-10:
What happens to the FPS if you put your hand over the front of the camera and hold it there?  If the FPS recovers to 30 then it could suggest that something in the environment that the camera is located in is causing the exposure to change incorrectly and therefore making the FPS change too (exposure value and FPS decrease / increase are connected).

TopRamen1, 2023-11-13:
Yes indeed the rate becomes stable when I cover up the camera:

I launched this command with the camera covered and then uncovered it during the test:
```
ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true
```
 
 Results:
 
 average rate: 29.511
	min: 0.024s max: 0.056s std dev: 0.00428s window: 32
average rate: 28.776
	min: 0.024s max: 0.067s std dev: 0.00671s window: 60
average rate: 28.549
	min: 0.024s max: 0.067s std dev: 0.00737s window: 89
average rate: 27.925
	min: 0.024s max: 0.100s std dev: 0.00983s window: 115
average rate: 28.325
	min: 0.024s max: 0.100s std dev: 0.00881s window: 145
average rate: 28.596
	min: 0.024s max: 0.100s std dev: 0.00805s window: 175
average rate: 28.792
	min: 0.024s max: 0.100s std dev: 0.00747s window: 205
average rate: 28.938
	min: 0.024s max: 0.100s std dev: 0.00700s window: 235
average rate: 29.053
	min: 0.024s max: 0.100s std dev: 0.00660s window: 265
average rate: 29.144
	min: 0.024s max: 0.100s std dev: 0.00626s window: 295
average rate: 29.218
	min: 0.024s max: 0.100s std dev: 0.00597s window: 325
average rate: 29.282
	min: 0.024s max: 0.100s std dev: 0.00572s window: 355
average rate: 26.759
	min: 0.024s max: 1.277s std dev: 0.06389s window: 380
average rate: 23.778
	min: 0.024s max: 1.277s std dev: 0.08796s window: 386
average rate: 21.521
	min: 0.024s max: 1.277s std dev: 0.10478s window: 391
average rate: 20.029
	min: 0.024s max: 1.403s std dev: 0.12503s window: 392
average rate: 18.711
	min: 0.024s max: 1.433s std dev: 0.14298s window: 393
average rate: 18.060
	min: 0.024s max: 1.433s std dev: 0.14464s window: 398
average rate: 16.893
	min: 0.024s max: 1.433s std dev: 0.15593s window: 401
average rate: 15.714
	min: 0.024s max: 1.433s std dev: 0.16855s window: 405
average rate: 12.482
	min: 0.024s max: 6.872s std dev: 0.37531s window: 410
average rate: 11.921
	min: 0.024s max: 6.872s std dev: 0.37788s window: 415
average rate: 11.468
	min: 0.024s max: 6.872s std dev: 0.38031s window: 418
average rate: 10.945
	min: 0.024s max: 6.872s std dev: 0.38268s window: 423
average rate: 10.672
	min: 0.024s max: 6.872s std dev: 0.38331s window: 426
average rate: 10.367
	min: 0.024s max: 6.872s std dev: 0.38452s window: 428
average rate: 10.192
	min: 0.024s max: 6.872s std dev: 0.38378s window: 431
average rate: 9.927
	min: 0.024s max: 6.872s std dev: 0.38489s window: 434

**The drop in fps happened exactly when I uncovered the camera**

MartyG-RealSense, 2023-11-13:
Thanks very much for the confirmation that covering the camera stabilizes FPS and uncovering it destabilizes it.

Is the camera being used indoors or outdoors, please?

TopRamen1, 2023-11-13:
Only indoors

TopRamen1, 2023-11-13:
Hi, changing this parameter does not make any noticeable difference. 

Regarding the light sources I have tested and there is no significant difference in performance.
Also in the initial setup there was no direct strong light pointed at the camera.

Additionally I have tested with a smaller window of 10 msgs and observed that the frequency is very  volatile:

average rate: 7.898
	min: 0.063s max: 0.269s std dev: 0.06287s window: 10
average rate: 7.503
	min: 0.056s max: 0.336s std dev: 0.09117s window: 10
average rate: 10.015
	min: 0.061s max: 0.144s std dev: 0.03175s window: 10
average rate: 7.915
	min: 0.066s max: 0.202s std dev: 0.04674s window: 10
average rate: 11.559
	min: 0.065s max: 0.137s std dev: 0.03059s window: 10
average rate: 14.991
	min: 0.059s max: 0.075s std dev: 0.00573s window: 10
average rate: 14.788
	min: 0.061s max: 0.075s std dev: 0.00399s window: 10
average rate: 15.041
	min: 0.059s max: 0.072s std dev: 0.00317s window: 10
average rate: 14.853
	min: 0.060s max: 0.075s std dev: 0.00471s window: 10
average rate: 15.019
	min: 0.060s max: 0.075s std dev: 0.00354s window: 10
average rate: 14.742
	min: 0.059s max: 0.077s std dev: 0.00606s window: 10
average rate: 12.524
	min: 0.060s max: 0.138s std dev: 0.02672s window: 10
average rate: 15.041
	min: 0.061s max: 0.071s std dev: 0.00337s window: 10
average rate: 12.444
	min: 0.060s max: 0.131s std dev: 0.02493s window: 10
average rate: 9.992
	min: 0.063s max: 0.137s std dev: 0.03072s window: 10
average rate: 13.552
	min: 0.063s max: 0.126s std dev: 0.01794s window: 10
average rate: 10.731
	min: 0.062s max: 0.131s std dev: 0.02876s window: 10
average rate: 6.512
	min: 0.061s max: 0.536s std dev: 0.13996s window: 10
average rate: 7.882
	min: 0.059s max: 0.272s std dev: 0.08161s window: 10
average rate: 14.886
	min: 0.058s max: 0.078s std dev: 0.00499s window: 10
average rate: 10.704
	min: 0.066s max: 0.136s std dev: 0.02796s window: 10
average rate: 9.337
	min: 0.063s max: 0.136s std dev: 0.02989s window: 10
average rate: 10.684
	min: 0.056s max: 0.192s std dev: 0.04259s window: 10
average rate: 6.616
	min: 0.064s max: 0.404s std dev: 0.10403s window: 10
average rate: 13.597
	min: 0.064s max: 0.132s std dev: 0.01948s window: 10
average rate: 12.508
	min: 0.063s max: 0.135s std dev: 0.02762s window: 10

My main concern right now is that when I launch the ROS 1 driver the framerate is exactly as expected even when pointed at a direct light source.
Do you have any idea why the ROS 2 driver performance might be worse?

TopRamen1, 2023-11-15:
No it does not have any effect.

MartyG-RealSense, 2023-11-16:
Continuing with the theory that it is an exposure related issue as the camera recovers when covered over and returns to the problem when uncovered: how is the FPS affected if you set the depth exposure manually to its default value of 33000 

`ros2 param set /camera depth_module.exposure 33000`

TopRamen1, 2023-11-20:
Unfortunately changing this param did not have any effect. The fps shows the same peaks and drops as in my last comment

MartyG-RealSense, 2023-11-20:
You mentioned in https://github.com/IntelRealSense/realsense-ros/issues/2923#issuecomment-1805515207 that you were using a ROS2 Humble container.  Have you had the opportunity to test without a container to see whether it is a factor in the variable FPS that you are experiencing in the ROS2 wrapper, please?

TopRamen1, 2023-11-22:
I do not have the option to test the camera with ros installed natively on the NUC.
However, I have tested on a PC (lenovo legion) and there is no difference in running the driver natively or in the same container I'm using on the NUC.

The strange issue I'm running into on the legion is that the framerate seems to be more stable with the setting on 30 FPS than when I tone it down to 6 FPS. 
I have tested with ros1 on the legion as well and the performance is significantly better.

It's strange to me that the FPS issue is not reported more often here.
Is this a known issue you are able to reproduce yourself?

MartyG-RealSense, 2023-11-22:
Historically, 6 FPS is more likely to have problems than the next highest value (15 FPS).  Because new frames are delivered less often at 6 FPS, there is a higher risk of the frame delivery pipeline 'timing out' if the next new frame does not arrive quickly enough.  Such errors tend to disappear at 15 FPS.

TopRamen1, 2023-11-24:
Sadly for us, the 15 FPS is unstable as well. We have encountered at worst one to five-second gaps in the point cloud stream.

MartyG-RealSense, 2023-11-24:
If 30 FPS is more stable, you could try reducing the launch resolution from 848x480 to 640x480 at 30 FPS to see whether you get acceptable performance.

closed 2023-12-28 by MartyG-RealSense