realsenseai/realsense-ros #2928: What may cause different number of depth and RGB frames when we do depth aligned capture?

monajalal, 2023-11-10:
Within everything staying the same, sometimes my captures have different number of frames in "RGB" and "Depth" for depth aligned captures with Intel RealSense D435 camera using the ROS Noetic inside Docker with Ubuntu 20.04 and Python 3.8.
What is the cause?

```
(base) mona@ada:~/depth_aligned_captures$ tree .
.
├── depth
│   └── 1699558775.573845.png
└── rgb
    ├── 1699558772.634131.png
    ├── 1699558772.725263.png
    ├── 1699558772.819511.png
    ├── 1699558772.916873.png
    ├── 1699558773.017349.png
    ├── 1699558773.120940.png
    ├── 1699558773.227645.png
    ├── 1699558773.337465.png
    ├── 1699558773.450398.png
    ├── 1699558773.566447.png
    ├── 1699558773.685609.png
    ├── 1699558773.807886.png
    ├── 1699558773.926256.png
    ├── 1699558774.026151.png
    ├── 1699558774.126045.png
    ├── 1699558774.225940.png
    ├── 1699558774.325835.png
    ├── 1699558774.425729.png
    ├── 1699558774.525624.png
    ├── 1699558774.625519.png
    ├── 1699558774.725413.png
    ├── 1699558774.825308.png
    ├── 1699558774.925202.png
    ├── 1699558775.025097.png
    ├── 1699558775.124991.png
    ├── 1699558775.224886.png
    ├── 1699558775.324781.png
    ├── 1699558775.424675.png
    ├── 1699558775.454372.png
    ├── 1699558775.479725.png
    ├── 1699558775.512069.png
    ├── 1699558775.533233.png
    ├── 1699558775.540731.png
    └── 1699558775.573845.png

2 directories, 35 files

```

```
root@8603c293ad2b:/workspace# rosbag info mona_test_bag_2023-11-09-21-24-16.bag
path:        mona_test_bag_2023-11-09-21-24-16.bag
version:     2.0
duration:    31.7s
start:       Nov 09 2023 21:24:16.44 (1699565056.44)
end:         Nov 09 2023 21:24:48.17 (1699565088.17)
size:        72.1 MB
messages:    941
compression: none [74/74 chunks]
types:       bond/Status                           [eacc84bf5d65b6777d4c50f463dfb9c8]
             diagnostic_msgs/DiagnosticArray       [60810da900de1dd6ddd437c3503511da]
             dynamic_reconfigure/Config            [958f16a05573709014982821e6822580]
             dynamic_reconfigure/ConfigDescription [757ce9d44ba8ddd801bb30bc456f946f]
             realsense2_camera/Extrinsics          [3627b43073f4cd5dd6dc179a49eda2ad]
             realsense2_camera/Metadata            [4966ca002be16ee67fe4dbfb2f354787]
             rosgraph_msgs/Log                     [acffd30cd6b6de30f120938c17c593fb]
             sensor_msgs/CameraInfo                [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage           [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/Image                     [060021388200f6f0f447d0fcd9c64743]
             tf2_msgs/TFMessage                    [94810edda583a504dfda3829e70d7eec]
             theora_image_transport/Packet         [33ac4e14a7cff32e7e0d65f18bb410f3]
[... 21 lines cut ...]
             /camera/depth/image_rect_raw/compressed/parameter_descriptions                      1 msg     : dynamic_reconfigure/ConfigDescription
             /camera/depth/image_rect_raw/compressed/parameter_updates                           1 msg     : dynamic_reconfigure/Config           
             /camera/depth/image_rect_raw/compressedDepth                                       94 msgs    : sensor_msgs/CompressedImage          
             /camera/depth/image_rect_raw/compressedDepth/parameter_descriptions                 1 msg     : dynamic_reconfigure/ConfigDescription
             /camera/depth/image_rect_raw/compressedDepth/parameter_updates                      1 msg     : dynamic_reconfigure/Config           
             /camera/depth/image_rect_raw/theora                                                 3 msgs    : theora_image_transport/Packet        
             /camera/depth/image_rect_raw/theora/parameter_descriptions                          1 msg     : dynamic_reconfigure/ConfigDescription
             /camera/depth/image_rect_raw/theora/parameter_updates                               1 msg     : dynamic_reconfigure/Config           
             /camera/depth/metadata                                                             97 msgs    : realsense2_camera/Metadata           
             /camera/extrinsics/depth_to_color                                                   1 msg     : realsense2_camera/Extrinsics         
             /camera/realsense2_camera_manager/bond                                             64 msgs    : bond/Status                           (2 connections)
             /camera/rgb_camera/auto_exposure_roi/parameter_descriptions                         1 msg     : dynamic_reconfigure/ConfigDescription
             /camera/rgb_camera/auto_exposure_roi/parameter_updates                              1 msg     : dynamic_reconfigure/Config           
             /camera/rgb_camera/parameter_descriptions                                           1 msg     : dynamic_reconfigure/ConfigDescription
             /camera/rgb_camera/parameter_updates                                                1 msg     : dynamic_reconfigure/Config           
             /camera/stereo_module/auto_exposure_roi/parameter_descriptions                      1 msg     : dynamic_reconfigure/ConfigDescription
             /camera/stereo_module/auto_exposure_roi/parameter_updates                           1 msg     : dynamic_reconfigure/Config           
             /camera/stereo_module/parameter_descriptions                                        1 msg     : dynamic_reconfigure/ConfigDescription
             /camera/stereo_module/parameter_updates                                             1 msg     : dynamic_reconfigure/Config           
             /diagnostics                                                                      120 msgs    : diagnostic_msgs/DiagnosticArray      
             /rosout                                                                           147 msgs    : rosgraph_msgs/Log                     (2 connections)
             /rosout_agg                                                                        97 msgs    : rosgraph_msgs/Log                    
             /tf_static                                                                          1 msg     : tf2_msgs/TFMessage

```

MartyG-RealSense, 2023-11-11:
Hi @monajalal  When you say that sometimes your captures have a different number of frames, do you mean that some bag files are good recordings and some are bad?

monajalal, 2023-11-16:
closed with answer provided here https://robotics.stackexchange.com/a/105440/16739

closed 2023-11-16 by monajalal