realsenseai/realsense-ros #2929: [ERROR] [1699653946.712325915]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense' during roslaunch realsense2_camera rs_camera.launch align_depth:=true

monajalal, 2023-11-10:
Is this an error I should ignore or could you please let me know how to resolve it?

```

root@8603c293ad2b:/workspace/catkin_ws# roslaunch realsense2_camera rs_camera.launch align_depth:=true

ETC MESSAGES

[ERROR] [1699653946.712325915]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'

```

[image]

when I first open rviz I see this:

[image]

and I am able to see `color_raw`

[image]

I created `bag_converter` package and I am using this script inside it:
```
(base) mona@ada:~/catkin_ws/src/bag_converter/scripts$ cat convert.py 
#!/usr/bin/python3

import os
import sys
import yaml
from rosbag.bag import Bag
import cv2

import roslib;   #roslib.load_manifest(PKG)
import rosbag
import rospy
import cv2
import numpy as np
import argparse
import os

from sensor_msgs.msg import Image
from cv_bridge import CvBridge
from cv_bridge import CvBridgeError

class ImageCreator():
    def __init__(self, bagfile, rgbpath, depthpath, rgbstamp, depthstamp):
        self.bridge = CvBridge()
        with rosbag.Bag(bagfile, 'r') as bag:
            for topic,msg,t in bag.read_messages():
                print(topic)
                if topic == "/camera/color/image_raw":
                    cv_image = self.bridge.imgmsg_to_cv2(msg,"bgr8")
                    timestr = "%.6f" %  msg.header.stamp.to_sec()
                    image_name = timestr+ ".png" 
                    cv2.waitKey(1);
                    cv2.imwrite(rgbpath + image_name, cv_image) 

                elif topic == "/camera/aligned_depth_to_color/image_raw": 
                    cv_image = self.bridge.imgmsg_to_cv2(msg,"16UC1")
                    timestr = "%.6f" %  msg.header.stamp.to_sec()
                    image_name = timestr+ ".png"
                    
                    cv2.imwrite(depthpath + image_name, (cv_image).astype(np.uint16))

if __name__ == '__main__':
    #ImageCreator('/workspace/2023-11-09-19-39-30.bag', "/workspace/depth_aligned_captures/rgb/", "/workspace/depth_aligned_captures/depth/", 1, 1)
    ImageCreator('/workspace/cheezit_1280_whole.bag', '/workspace/cheezit/rgb/', '/workspace/cheezit/depth/', 1, 1)

```

since this line of script `                    cv_image = self.bridge.imgmsg_to_cv2(msg,"16UC1")` uses `16UC1` I am concerned this error is causing problem.

MartyG-RealSense, 2023-11-11:
Hi @monajalal  Do you see the same color format error if you perform an **RGBD** launch instead of rs_camera.launch?  An RGBD launch can be performed with the steps below.

1.  Install support for an RGBD launch on Noetic with this instruction:

sudo apt-get install ros-noetic-rgbd-launch

2.  Launch the **rs_rgbd** launch file.

`roslaunch realsense2_camera rs_rgbd.launch`

You do not need to use the align_depth instruction with an rs_rgbd launch as it is aready true by default in this launch file.

MartyG-RealSense, 2023-11-16:
Hi @monajalal  Do you require further assistance with this case, please?  Thanks!

monajalal, 2023-11-16:
@MartyG-RealSense I think I am all set. Could you please check this answer and let me know if that makes sense to you?

https://robotics.stackexchange.com/a/105440/16739

closed 2023-11-16 by monajalal

MartyG-RealSense, 2023-11-16:
Yes, the Stack Exchange answer makes sense to me.  Thanks very much for the update!