realsenseai/realsense-ros #2930: Get RTSP Stream from Intel Realsense in parallel to ROS2 Topics?

Scoeerg, 2023-11-17:
Hi.

is there any way to access an RTSP stream (or similar) **in parallel** to ROS2 Topics from Realsense-SDK?

There's [this ROS1 project](https://github.com/CircusMonkey/ros_rtsp) which serves as a bridge ros-topic <-> RTSP. Apparently this
is efficient since it's implemented using ROS components, i.e. communication is fast. But it's involved and ROS1. 

Thanks :)

MartyG-RealSense, 2023-11-17:
Hi @Scoeerg  The RGB camera is not involved in depth processing, as the depth frame is constructed from raw left and right **infrared** images in the camera hardware.  

So if you are not using the RGB stream in ROS2 for purposes such as texturing a pointcloud and you make sure to disable the color topic in the ROS2 wrapper's launch with the launch parameter **enable_color:=false** then the RGB sensor should remain available for use by another application for object recognition without having to publish it to a ROS2 topic.

The rules for accessing a camera's sensors with more than one application are governed by the RealSense SDK's **Multi-Streaming Model** rules at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

Scoeerg, 2023-11-17:
Thanks @MartyG-RealSense I will give it a try and only close this issue when I get it running with some helpful insights for others to read in the future.

MartyG-RealSense, 2023-11-17:
You are very welcome. I look forward to your future update.  Good luck!

MartyG-RealSense, 2024-01-21:
Hi @Scoeerg  Are you still working on the project described in this issue, please?  Thanks!

MartyG-RealSense, 2024-01-28:
Hi @Scoeerg  Do you require further assistance with this case, please?  Thanks!

Scoeerg, 2024-01-29:
Hi @MartyG-RealSense 

my way to deal with this problem for me is: run the realsense-camera as intended, and it publishes all data (including RGB) as a ROS2-topic. Then use a ROS -> RTSP bridge like https://github.com/CircusMonkey/ros_rtsp 

The beforementioned bridge does not support ROS2, hence I needed to use another repo based on this one, which supports ROS2:

https://github.com/45kmh/image2rtsp

There bridges are simply ros2-nodes subscribing to a ROS2-Topic and publish the RTSP-stream. Easy to setup, and it works with all cameras  - not only Intel's ;). Sorry for the late response, closing issue.

closed 2024-01-29 by Scoeerg

MartyG-RealSense, 2024-01-29:
It's no problem at all.  I'm pleased to hear that you achieved a solution.  Thanks very much for the update and for sharing the details of your solution!