realsenseai/realsense-ros #2931: D435i rgbd pointcloud no color

quangmta, 2023-11-19:
Hello
I installed realsense ros in following by this link: https://github.com/IntelRealSense/realsense-ros. 
I used method 1 to install realsense SDK 2.51.1: `sudo apt install ros-humble-librealsense2* `
And install ros realsense wrapper 4.51.1: `sudo apt install ros-humble-realsense2-*`
I run the camera node by this command: `ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true -p colorizer.enable:=true -p depth_module.profile:=424x240x30 -p rgb_camera:=424x240x30 -p pointcloud_texture_stream:=RS2_STREAM_COLOR`
The problem is the pointcloud has no color in it when I open Rviz2.
Color transformer has no RGB8.
My environment system is :
Jetson nano, Ubuntu 20.04.
I also try to install wrapper in my laptop, Ubuntu 20.04, but still this error.
P/s: I tried to install wraper by cloning the source code and build it. In my computer everything works well. But in Jetson nano, when I try to enable pointcloud, the nothing  is shown in rviz2 and rgbd image is stuck.
Could you help me to fix this? 
Thank you!

MartyG-RealSense, 2023-11-20:
Hi @quangmta  Do you have the same problem with a **ros2 launch** instead of ros2 run?

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=424x240x30 rgb_camera.profile:=424x240x30  pointcloud.enable:=true`

quangmta, 2023-11-21:
Hi @MartyG-RealSense Using `ros2 launch` solved the problem! Thank you very much!

MartyG-RealSense, 2023-11-21:
That's great news, @quangmta - you are very welcome.  Thanks very much for the update!

closed 2023-11-28 by MartyG-RealSense