realsenseai/realsense-ros #2938: CMake failure couldn't find package configuration (realsense2_camera_msgsConfig.cmake, realsense2_camera_msgs-config.cmake)

Lexi2go, 2023-11-28:
maren@maren-Yoga-Slim-7-14ARE05:catkin_make
Base path: /home/maren/catkin_ws
Source space: /home/maren/catkin_ws/src
Build space: /home/maren/catkin_ws/build
Devel space: /home/maren/catkin_ws/devel
Install space: /home/maren/catkin_ws/install
####
#### Running command: "cmake /home/maren/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/maren/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/maren/catkin_ws/install -G Unix Makefiles" in "/home/maren/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/maren/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/maren/catkin_ws/devel;/opt/ros/noetic
-- This workspace overlays: /home/maren/catkin_ws/devel;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/maren/catkin_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Could NOT find realsense2_camera_msgs (missing: realsense2_camera_msgs_DIR)
-- Could not find the required component 'realsense2_camera_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "realsense2_camera_msgs" with any of the following names:

    realsense2_camera_msgsConfig.cmake
    realsense2_camera_msgs-config.cmake

  Add the installation prefix of "realsense2_camera_msgs" to
  CMAKE_PREFIX_PATH or set "realsense2_camera_msgs_DIR" to a directory
  containing one of the above files.  If "realsense2_camera_msgs" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:63 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/maren/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/maren/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

MartyG-RealSense, 2023-11-29:
Hi @Lexi2go  Have you followed the series of instructions below for an Ubuntu installation of the ROS1 wrapper for Noetic please?

1.  Create a catkin workspace.

```
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src/
```

2.  Clone the latest Intel® RealSense™ ROS from here into 'catkin_ws/src/'

```
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
cd ..
```

3.  Make sure all dependent packages are installed. You can check .travis.yml file for reference.  Specifically, make sure that the ros package ddynamic_reconfigure is installed.  If ddynamic_reconfigure cannot be installed using APT, you may clone it into your workspace 'catkin_ws/src/' from https://github.com/pal-robotics/ddynamic_reconfigure/tree/kinetic-devel

```
catkin_init_workspace
cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
```

4.  Source the bash file.

```
echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc
```

*************

If you have followed these instructions and the build process still fails, it may be easier to build the librealsense SDK and the ROS1 wrapper together at the same time from packages by using the command below.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

This method does not install the RealSense Viewer tool, so if you require the Viewer then you will need to build the librealsense SDK and the ROS1 wrapper separately (the SDK first, then the wrapper secondly afterwards).

Lexi2go, 2023-12-01:
Perfect, thank you

closed 2023-12-01 by Lexi2go