realsenseai/realsense-ros #2939: imge_rect_raw data from depth-frame value

Aki1991, 2023-11-29:
Hi all,
I  am working on project where I need realsense topic data (/camera/depth/image_rect_raw) on ROS2 and the depth_frame from pyrealsense for detection model. (I am running a python script and realsense_ros both together). Now my problem is that both script can not use same camera. So what I tried is to convert the depth_frame data from pyrealsense into data needed by the image_rect_raw topic. But I am not sure what data are there on the "/camera/depth/image_recr_raw". I tried searching in the git repo but I didnt find which data are published on this topic.

Can anyone help me with that? Is there any way to convert the depth_frame,  that we get from pyrealsense into the /camera/depth/imagr_rect_raw data? 

@MartyG-RealSense Can you give me some insight on this?

Aki1991, 2023-11-29:
Let me explain a bit. I know the topic type for "/camera/depth/image_rect_raw". I know all the parameters or values needed by this topic. I got all the values except the "data []". If I am not wrong, this has the depth values for all pixels. 

The number of values I am getting from pyrealsense depth frame is "480 x 640" but the number of values from the "/camera/depth/image_rect_raw" is "480 x 640 x 2". This confuses me more.

MartyG-RealSense, 2023-11-29:
Hi @Aki1991  Instead of converting the pyrealsense2 depth frame to the `/camera/depth/image_rect_raw` topic, perhaps you could do the reverse of this and use a Python **node script** in the ROS2 wrapper to read the topic.  The wrapper has an example of a pyrealsense2 node script called **show_center_depth.py** that demonstrates this.

 https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/scripts/show_center_depth.py#L85

Aki1991, 2023-12-07:
Hey @MartyG-RealSense, sorry for no reply. 

I did find that useful. I have yet to solve that error fully. I got some other urgent work and had to do that first.  Once I will solve the problem, I will add further comments.

MartyG-RealSense, 2023-12-07:
I look forward to your next report.  Thanks very much for the update!

MartyG-RealSense, 2023-12-22:
Hi @Aki1991  Do you have an update about this case that you can provide, please?  Thanks!

Aki1991, 2023-12-27:
Hi @MartyG-RealSense, again sorry for the late reply. I found the solution from the link you sent. In the link, they have used `imgmsg_to_cv2` function. I looked at the library functions and found that there is also a function `cv2_to_imgmsg`. That function is what I needed to convert the pyrealsense img to the ros2 topic. The only difference is the dimension, which I can set as per my use-case.

Thank you so much. :)

closed 2023-12-27 by Aki1991

MartyG-RealSense, 2023-12-27:
You are very welcome.  It's excellent to hear that you have the solution.  Thanks very much for the update!