realsenseai/realsense-ros #2942: Echo of Node => Unable to communicate with master!

Lexi2go, 2023-12-01:
1. Terminal
maren@maren-Yoga-Slim-7-14ARE05: roscore
maren@maren-Yoga-Slim-7-14ARE05: roslaunch realsense2_camera rs_camera.launch
maren@maren-Yoga-Slim-7-14ARE05:rosrun my_realsense_pkg realsense_publisher.py
---------------------------------------------------------------------------------------------------------------------
realsense_publisher.py

#!/usr/bin/env python3

import rospy
import pyrealsense2 as rs
from sensor_msgs.msg import Image
from cv_bridge import CvBridge
import cv2

def publish_realsense_data():
    rospy.init_node('realsense_publisher', anonymous=True)
    rate = rospy.Rate(10)  # 10 Hz

    bridge = CvBridge()
    image_pub = rospy.Publisher('/Live_Camera', Image, queue_size=10)

    # RealSense Pipeline initialisieren
    pipeline = rs.pipeline()
    config = rs.config()
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

    profile = pipeline.start(config)

    try:
        while not rospy.is_shutdown():
            # RealSense Frame abrufen
            frames = pipeline.wait_for_frames()
            color_frame = frames.get_color_frame()
            
            if not color_frame:
                continue

            # RealSense Frame in OpenCV-Format umwandeln
            frame = np.asanyarray(color_frame.get_data())

            # Frame in ROS-Bildnachricht umwandeln
            image_msg = bridge.cv2_to_imgmsg(frame, "bgr8")

            # Bildnachricht veröffentlichen
            image_pub.publish(image_msg)

            rate.sleep()

    finally:
        pipeline.stop()

if __name__ == '__main__':
    try:
        publish_realsense_data()
    except rospy.ROSInterruptException:
        pass

------------------------------------------------------------------------------------------------------------------
2. Terminal
maren@maren-Yoga-Slim-7-14ARE05:~/catkin_ws$ rostopic echo /Live_Camera
ERROR: Unable to communicate with master!

MartyG-RealSense, 2023-12-02:
Hi @Lexi2go  As you started roscore in the first terminal, what happens if you input the rostopic command in the first terminal instead of the second?

`:~/catkin_ws$ rostopic echo /Live_Camera`

Lexi2go, 2023-12-07:
Unfortunately the same problem, but I have an idea. I will try to subscribe the topic and write this in another node

MartyG-RealSense, 2023-12-28:
Hi @Lexi2go  Do you require further assistance with this case, please?  Thanks!

Lexi2go, 2023-12-28:
I am sorry for my late reply.
I have managed to communicate with the master, through a self written launch file.
roslaunch  my_realsense_pkg realsense_publisher.launch

closed 2023-12-28 by Lexi2go

MartyG-RealSense, 2023-12-29:
It's no problem at all.  I'm pleased to hear that you achieved a solution.  Thanks very much for the update!