realsenseai/realsense-ros #2947: When I try "rosbag record -a",the realsense-ros camera code "rs_camera.launch" error.

collarz, 2023-12-15:
The Error:
[ERROR] [1702629067.095244071]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1702629067.095959997]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: mono8).
[ERROR] [1702629067.108349659]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: mono8).
[ERROR] [1702629067.133162794]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1702629067.133854084]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: mono8).

MartyG-RealSense, 2023-12-15:
Hi @collarz  A RealSense ROS user with the same problem found that the advice in the link below solved the issue for them.

https://robotics.stackexchange.com/questions/105438/depth-aligned-capture-doesnt-yield-same-number-of-rgb-and-depth-frames-cv-br/105440#105440

collarz, 2023-12-20:
Thank for your help.The solution is very useful.

MartyG-RealSense, 2023-12-20:
You are very welcome.  I'm pleased that I could help.  :)

MartyG-RealSense, 2024-01-05:
Case closed due to no further comments received.

closed 2024-01-05 by MartyG-RealSense

Rhymakers, 2024-08-01:
I have the same problem. Can't I fix it so that the compressed topic doesn't come out?

MartyG-RealSense, 2024-08-01:
Hi @Rhymakers  Compressed topics are provided by the **image_transport** plugin.  Disabling it is discussed at https://github.com/ros-perception/image_transport_plugins/issues/160

Rhymakers, 2024-08-01:
Hi @MartyG-RealSense, Thank you for your response. I don't think there's a way to turn it off from Humble.. What should I do?

MartyG-RealSense, 2024-08-01:
A ROS2 user at https://github.com/ros-perception/image_common/pull/264 suggested 'blacklisting' image_transport with the command below.

`ros2 run usb_cam usb_cam_node_exe --ros-args -p image_raw.disable_pub_plugins:=["image_transport/compressed"]`