realsenseai/realsense-ros #2951: D405 roslaunch failing

FadiGEBRAYEL, 2023-12-20:
I'm working on a ROS project with a RealSense camera.
Initially, everything was running smoothly using the D435 camera. However, when I switched to a D405 RealSense camera, I encountered the following error: [FATAL] [1703074414.798278625]: Unsupported device! Product ID: 0x0B5B.

Here's the output from the terminal:
 
 setting /run_id to 3237a120-9f31-11ee-8167-657b29b0e20a
process[rosout-1]: started with pid [69319]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [69326]
process[camera/realsense2_camera-3]: started with pid [69327]
process[main_image_processing-4]: started with pid [69328]
process[main_icp_alignment_vs-5]: started with pid [69333]
process[main_compute_command-6]: started with pid [69335]
process[rviz-7]: started with pid [69336]
[ INFO] [1703074414.477523775]: visual sevoing mode: PBVS
Lambda gain initialization1
VisualServoingCommand Initialization Done
[ INFO] [1703074414.483480894]:  IcpAlignmentVS Begin initialization
[ INFO] [1703074414.504645020]: Initializing nodelet with 20 worker threads.
[ INFO] [1703074414.535860861]: [INFO] PointCloudConstructor initialization
[ INFO] [1703074414.537921284]: subscribing on Images topic
[ INFO] [1703074414.564928555]: RealSense ROS v2.3.2
[ INFO] [1703074414.564951496]: Built with LibRealSense v2.50.0
[ INFO] [1703074414.564959803]: Running with LibRealSense v2.50.0
[ INFO] [1703074414.585367073]:  
[ INFO] [1703074414.797573793]: Device with serial number 127122270485 was found.

[ INFO] [1703074414.797646733]: Device with physical ID 4-3-13 was found.
[ INFO] [1703074414.797665516]: Device with name Intel RealSense D405 was found.
[ INFO] [1703074414.798194275]: Device with port number 4-3 was found.
[ INFO] [1703074414.798226689]: Device USB type: 3.2
[FATAL] [1703074414.798278625]: Unsupported device! Product ID: 0x0B5B
[camera/realsense2_camera_manager-2] process has died [pid 69326, exit code 1, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/fgebrayel/.ros/log/3237a120-9f31-11ee-8167-657b29b0e20a/camera-realsense2_camera_manager-2.log].
log file: /home/fgebrayel/.ros/log/3237a120-9f31-11ee-8167-657b29b0e20a/camera-realsense2_camera_manager-2*.log

Why does this problem occur, and what should I do in this case? 
Thanks in advance!

MartyG-RealSense, 2023-12-20:
Hi @FadiGEBRAYEL  The D405 camera model was initially not supported by the RealSense ROS1 wrapper.  Though development of this wrapper has now ceased, Intel added D405 support to the ROS1 wrapper a couple of months ago at the time of writing this.  

If you installed the wrapper longer than a couple of months ago, I recommend removing the 2.3.2 wrapper and then reinstalling it to see whether this resolves your problem.

If that does not work, a custom 'forked' version of the 2.3.2 wrapper created by a RealSense user that includes D405 support can be found at https://github.com/rjwb1/realsense-ros/tree/development

FadiGEBRAYEL, 2023-12-20:
thanks for your advices! I successfully resolved the issue by uninstalling the ros-noetic-realsense2-camera package and installing a modified wrapper for D405 compatibility with ROS 1 from this Git repository: git clone --branch development https://github.com/rjwb1/realsense-ros.git. Now, both the D405 and D435 cameras are working smoothly.
If it is not a good way to do it let me know please
Thanks once again!

MartyG-RealSense, 2023-12-20:
It is fine to do it this way.  The ROS1 wrapper has ceased development and only the ROS2 wrapper is actively updated now, so you are not going to miss any new updates if you continue to use the modified fork.

MartyG-RealSense, 2023-12-28:
Hi @FadiGEBRAYEL  Do you require further assistance with this case, please?  Thanks!

FadiGEBRAYEL, 2023-12-28:
no further help is needed , thank you

MartyG-RealSense, 2023-12-29:
You are very welcome.  Thanks very much for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-12-29 by MartyG-RealSense