realsenseai/realsense-ros #2955: ROS2 point cloud stuck after a few seconds with message 'Hardware Notification:Left MIPI error,1.70373e+12,Error,Hardware Error'

kenloumixx, 2023-12-28:
Hello,
thanks for the amazing ROS wrapper. 

camera : D435i
librealsense : 3.2.2 (from source)
realsense-ros : 2.48.0
ROS version : ROS2 galactic 

With the above setting, I successfully checked the RGB image streaming from camera with rviz2.
However, when I set 'enable_pointcloud' to true, the warning appears and the rviz2 display gets stuck

```powershell
❯ ros2 launch realsense2_camera rs_launch.py

[INFO] [launch]: All log files can be found below /root/.ros/log/2023-12-28-02-06-29-827901-nx-22435
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [22506]
[realsense2_camera_node-1] [INFO] [1703729190.668566800] [RealSenseCameraNode]: RealSense ROS v3.2.2
[realsense2_camera_node-1] [INFO] [1703729190.668804628] [RealSenseCameraNode]: Built with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO] [1703729190.668855605] [RealSenseCameraNode]: Running with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO] [1703729191.233569808] [RealSenseCameraNode]: Device with serial number 117322074667 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1703729191.233709938] [RealSenseCameraNode]: Device with physical ID 2-3.3-5 was found.
[realsense2_camera_node-1] [INFO] [1703729191.233754931] [RealSenseCameraNode]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1703729191.235367564] [RealSenseCameraNode]: Device with port number 2-3.3 was found.
[realsense2_camera_node-1] [INFO] [1703729191.235453198] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1703729191.265505895] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [INFO] [1703729191.300041415] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-1] [INFO] [1703729191.300131912] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1703729191.300184009] [RealSenseCameraNode]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1703729191.300219081] [RealSenseCameraNode]: Device Serial No: 117322074667
[realsense2_camera_node-1] [INFO] [1703729191.300250218] [RealSenseCameraNode]: Device physical port: 2-3.3-5
[realsense2_camera_node-1] [INFO] [1703729191.300279434] [RealSenseCameraNode]: Device FW version: 5.15.1
[realsense2_camera_node-1] [INFO] [1703729191.300307915] [RealSenseCameraNode]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1703729191.300373356] [RealSenseCameraNode]: Enable PointCloud: On
[realsense2_camera_node-1] [INFO] [1703729191.300406188] [RealSenseCameraNode]: Align Depth: On
[realsense2_camera_node-1] [INFO] [1703729191.300434381] [RealSenseCameraNode]: Sync Mode: On
[realsense2_camera_node-1] [INFO] [1703729191.301960197] [RealSenseCameraNode]: Device Sensors: 
[realsense2_camera_node-1] [INFO] [1703729191.436837232] [RealSenseCameraNode]: Stereo Module was found.
[realsense2_camera_node-1] [INFO] [1703729191.501134500] [RealSenseCameraNode]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1703729191.501832399] [RealSenseCameraNode]: Motion Module was found.
[realsense2_camera_node-1] [INFO] [1703729191.501945105] [RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1703729191.501995442] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1703729191.502037555] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1703729191.503225125] [RealSenseCameraNode]: Add Filter: pointcloud
[realsense2_camera_node-1] [INFO] [1703729191.505126947] [RealSenseCameraNode]: num_filters: 2
[realsense2_camera_node-1] [INFO] [1703729191.505195076] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1703729194.790100050] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1703729194.792883710] [RealSenseCameraNode]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[realsense2_camera_node-1] [INFO] [1703729194.804312466] [RealSenseCameraNode]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[realsense2_camera_node-1] [INFO] [1703729194.809306240] [RealSenseCameraNode]: setupPublishers...
[realsense2_camera_node-1] [INFO] [1703729194.846063906] [RealSenseCameraNode]: setupStreams...
[realsense2_camera_node-1]  28/12 02:06:35,147 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1703729195.163396930] [RealSenseCameraNode]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO] [1703729195.167074652] [RealSenseCameraNode]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1703729195.265403622] [RealSenseCameraNode]: 
[realsense2_camera_node-1]  28/12 02:06:35,266 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,277 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,287 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,298 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,309 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,320 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,334 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,346 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,360 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,371 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1703729195.458459235] [RealSenseCameraNode]: publishing ordered pointcloud.
[realsense2_camera_node-1]  28/12 02:06:35,502 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,513 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,526 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,537 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,550 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/12 02:06:35,561 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[realsense2_camera_node-1]  28/12 02:06:35,574 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[... cut ...]
[realsense2_camera_node-1]  28/12 02:06:35,724 WARNING [547667040576] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [WARN] [1703729196.036334179] [RealSenseCameraNode]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1]  28/12 02:06:36,367 ERROR [547423750464] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
[realsense2_camera_node-1] [WARN] [1703729197.795401580] [RealSenseCameraNode]: Hardware Notification:Left MIPI error,1.70373e+12,Error,Hardware Error
```

MartyG-RealSense, 2023-12-28:
Hi @kenloumixx  Although you are correct that librealsense **2.48.0** should be used with ROS2 wrapper **3.2.2**, the launch log states that the librealsense version that is actually installed on your computer is the much newer **2.54.2** (currently the latest version at the time of writing this).  As the 3.2.2 wrapper is designed specifically for 2.48.0, this version mis-match may be the cause of your problem.

[image]

********

 If you need to use ROS2 Galactic and cannot update to Humble or Iron, there is a much more modern ROS2 wrapper that is Galactic compatible, which is **4.54.1**.  The 3.x series of ROS2 wrapper, which includes 3.2.2, became obsolete a long time ago and only the 4.x ROS2 wrapper series is now actively updated.

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.51.1

Wrapper 4.54.1 should be matched with librealsense version **2.54.1**.

kenloumixx, 2024-01-03:
Thank you so much for the reply!

closed 2024-01-03 by kenloumixx