realsenseai/realsense-ros #2960: ros2 topic echo gets stuck after a few seconds

kenloumixx, 2024-01-03:
Hello,

I was trying to see the output of the command `ros2 topic echo /camera/color/image_raw`

After showing some img data messages for a few seconds, the terminal shows no updates and stops at the old final message. 

I use librealsense 2.54.2 and realsense-ros 4.54.1. I downloaded librealsense 2.54.1 but the installed version is 2.54.2 (I don't know why..

Below is the warning messages that I see,

Thank you!

```shell
❯ ros2 launch realsense2_camera rs_launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-01-03-08-32-25-064298-ubuntu-1065
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [1068]
[realsense2_camera_node-1] [INFO] [1704270745.665800153] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1704270745.666308368] [camera.camera]: Built with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO] [1704270745.666422421] [camera.camera]: Running with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO] [1704270746.103732850] [camera.camera]: Device with serial number 135222251064 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1704270746.113296931] [camera.camera]: Device with physical ID 2-1.1-4 was found.
[realsense2_camera_node-1] [INFO] [1704270746.113460202] [camera.camera]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-1] [INFO] [1704270746.115195129] [camera.camera]: Device with port number 2-1.1 was found.
[realsense2_camera_node-1] [INFO] [1704270746.117647015] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1704270746.122359069] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1704270746.136031239] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1704270746.136260113] [camera.camera]: Device Name: Intel RealSense D455
[realsense2_camera_node-1] [INFO] [1704270746.136733926] [camera.camera]: Device Serial No: 135222251064
[realsense2_camera_node-1] [INFO] [1704270746.136842091] [camera.camera]: Device physical port: 2-1.1-4
[realsense2_camera_node-1] [INFO] [1704270746.136930543] [camera.camera]: Device FW version: 5.13.0.50
[realsense2_camera_node-1] [INFO] [1704270746.137016371] [camera.camera]: Device Product ID: 0x0B5C
[realsense2_camera_node-1] [INFO] [1704270746.137102295] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1704270749.388506781] [camera.camera]: Set Exposure.1 to 7500
[realsense2_camera_node-1] [INFO] [1704270749.692549809] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1704270750.398758923] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1704270750.404992773] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1704270750.405625089] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1704270750.458124615] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1704270750.603919942] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1]  03/01 08:32:30,876 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:30,893 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:30,906 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:30,917 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:30,931 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:30,943 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:30,954 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:30,965 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:30,977 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:30,988 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,000 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,011 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1704270751.024389737] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1704270751.025251280] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1704270751.041800252] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [WARN] [1704270751.109428781] [camera.camera]: 
[realsense2_camera_node-1] [INFO] [1704270751.169965246] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1704270751.176303837] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  03/01 08:32:31,460 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,471 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,483 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,494 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,607 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,617 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,630 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,645 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,656 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,667 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:32:31,680 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... 25 lines cut ...]
[... cut ...]
[realsense2_camera_node-1]  03/01 08:34:05,243 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:34:05,254 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  03/01 08:34:05,268 WARNING [546937235776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

MartyG-RealSense, 2024-01-03:
Hi @kenloumixx  2.54.2 was a release with bug-fixes only rather than major changes, so it should be okay to use it with the 4.54.1 wrapper.

The warning message **control_transfer returned error, index: 768, error: Resource temporarily unavailable**, when it generates continuously like this, indicates a communication problem between the camera and computer / computing device.  

Does the same problem occur if launching with **ros2 run** instead of rs_launch.py?

`ros2 run realsense2_camera realsense2_camera_node`

kenloumixx, 2024-01-04:
@MartyG-RealSense Yes, when I run `ros2 run` and `ros2 launch`, in both cases the error occurs with about 20-30 warning messages and then stops to occur. I run the command `ros2 topic echo /camera/color/image_raw` and now it works well, showing multiple images without stopping. 

I think the problem has been solved still the error message remains. 
How can I solve this communication problem? 
I connected C type USB to the camera and USB-A type to the jetson board.

MartyG-RealSense, 2024-01-04:
If performance is normal despite the warning messages in the log and the warnings stop then they can be safely ignored.

kenloumixx, 2024-01-05:
@MartyG-RealSense Okay I got it! Thank you :)

closed 2024-01-05 by kenloumixx