realsenseai/realsense-ros #2961: How to control the fps of RGB / Depth image?

kenloumixx, 2024-01-03:
Hello,

I want to check the fps of RGB, depth map and point cloud. 

I checked the `rs_launch.py` but the fps is only used in gyro_fps, accel_fps and pose_fps.

How can I check the fps of the values without using `ros2 topic hz` command?

And also, when I check `ros2 topic hz /camera/color/image_raw`, sometimes it shows the values around 13 and the other day around 19, even in the same arguments. 

I'd like to ask what affects the values of topic hz in this situation. 

Thank you!

MartyG-RealSense, 2024-01-03:
Hi @kenloumixx  The parameters for the other stream types are depth_fps, color_fps and infra_fps.

If both depth and color are enabled then sometimes the FPS of one of the streams - typically the color - may drop.  Disabling the **auto_exposure_priority** parameter may help with this.  Please try doing so by inputting the command below after launch, during runtime.

`ros2 param set /camera/camera rgb_camera.auto_exposure_priority false`

kenloumixx, 2024-01-04:
@MartyG-RealSense Thank you. 
I use librealsense 2.54.2 and realsense-ros 4.54.1
I checked the arguments below in rs_launch.py and saw that there's no depth_fps, color_fps and infra_fps. 
```python
# Copyright 2023 Intel Corporation. All Rights Reserved.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

"""Launch realsense2_camera node."""
import os
import yaml
from launch import LaunchDescription
import launch_ros.actions
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.substitutions import LaunchConfiguration

configurable_parameters = [{'name': 'camera_name',                  'default': 'camera', 'description': 'camera unique name'},
                           {'name': 'serial_no',                    'default': "''", 'description': 'choose device by serial number'},
                           {'name': 'usb_port_id',                  'default': "''", 'description': 'choose device by usb port id'},
                           {'name': 'device_type',                  'default': "''", 'description': 'choose device by type'},
                           {'name': 'config_file',                  'default': "''", 'description': 'yaml config file'},
                           {'name': 'unite_imu_method',             'default': "0", 'description': '[0-None, 1-copy, 2-linear_interpolation]'},
                           {'name': 'json_file_path',               'default': "''", 'description': 'allows advanced configuration'},
                           {'name': 'log_level',                    'default': 'info', 'description': 'debug log level [DEBUG|INFO|WARN|ERROR|FATAL]'},
                           {'name': 'output',                       'default': 'screen', 'description': 'pipe node output [screen|log]'},
                           {'name': 'depth_module.profile',         'default': '0,0,0', 'description': 'depth module profile'},                           
                           {'name': 'enable_depth',                 'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'rgb_camera.profile',           'default': '0,0,0', 'description': 'color image width'},
                           {'name': 'rgb_camera.enable_auto_exposure', 'default': 'true', 'description': 'enable/disable auto exposure for color image'},
                           {'name': 'enable_color',                 'default': 'true', 'description': 'enable color stream'},
                           {'name': 'enable_infra1',                'default': 'false', 'description': 'enable infra1 stream'},
                           {'name': 'enable_infra2',                'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'infra_rgb',                    'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'enable_fisheye1',              'default': 'true', 'description': 'enable fisheye1 stream'},
                           {'name': 'enable_fisheye2',              'default': 'true', 'description': 'enable fisheye2 stream'},
                           {'name': 'enable_confidence',            'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'gyro_fps',                     'default': '0', 'description': "''"},                           
                           {'name': 'accel_fps',                    'default': '0', 'description': "''"},                           
                           {'name': 'enable_gyro',                  'default': 'false', 'description': "''"},                           
                           {'name': 'enable_accel',                 'default': 'false', 'description': "''"},                           
                           {'name': 'enable_pose',                  'default': 'true', 'description': "''"},                           
                           {'name': 'pose_fps',                     'default': '200', 'description': "''"},                           
                           {'name': 'pointcloud.enable',            'default': 'true', 'description': ''}, 
                           {'name': 'pointcloud.stream_filter',     'default': '2', 'description': 'texture stream for pointcloud'},
                           {'name': 'pointcloud.stream_index_filter','default': '0', 'description': 'texture stream index for pointcloud'},
                           {'name': 'enable_sync',                  'default': 'true', 'description': "''"},                           
                           {'name': 'align_depth.enable',           'default': 'true', 'description': "''"},                           
                           {'name': 'colorizer.enable',             'default': 'false', 'description': "''"},
                           {'name': 'clip_distance',                'default': '-2.', 'description': "''"},                           
[... cut ...]
def generate_launch_description():
    return LaunchDescription(declare_configurable_parameters(configurable_parameters) + [
        OpaqueFunction(function=launch_setup)
    ])
``` 
And also, could you explain the meaning of the `rgb_camera.auto_exposure_priority`?
If you mean the big difference of the topic hz of `camera/color/image_raw` is due to the alignment with depth image, then is it correct that the frequency of the depth image stream is not always the same?

thank you again for your help!

kenloumixx, 2024-01-04:
And as I run the `ros2 topic hz` command, I noticed that the frequency gets slower. At first it was 14 fps but after 2~3 minutes not it is 12. I'd like to ask if this is normal.

MartyG-RealSense, 2024-01-04:
When **auto_exposure_priority** is true, it allows a sensor to dynamically adjust the frame rate depending on lighting conditions.  

When auto_exposure = true and auto_exposure_priority = false, a constant FPS for both streams is enforced instead of permitting the FPS to vary.

When depth and color are both enabled, there does not need to be alignment in order for an FPS drop to occur.

It is not normal for FPS to drop over time but it does sometimes happen.

I cannot think of an alternative to `ros2 topic hz `for checking the FPS rate of a particular stream, unfortunately.

MartyG-RealSense, 2024-01-12:
Hi @kenloumixx  Do you require further assistance with this case, please?  Thanks!

kenloumixx, 2024-01-15:
@MartyG-RealSense Hi, I want to ask more about the decreasing of the fps. 

You said that it sometimes happen. Could you explain more about why this happens? Since I am working with jetson device and the fps issue strongly matters.

MartyG-RealSense, 2024-01-15:
This issue has previously been investigated by my Intel RealSense colleagues on the ROS wrapper development team but reasons for performance problems on Jetson compared to desktop / laptop PCs have not been found, unfortunately.  Raspberry Pi boards, which like Jetson are based on Arm architecture, have also experienced similar problems.

closed 2024-02-09 by MartyG-RealSense