realsenseai/realsense-ros #2962: No stream match for pointcloud chosen texture Process - Color

kenloumixx, 2024-01-04:
Hi, 

I am trying to obtain point cloud information using `rs_launch.py` with below setting. 

When I run the command with `ros2 topic hz /camera/depth/color/points`, it shows some messages with 2~3 hz while `/camera/color/image_raw` shows 9hz.

But after few hz messages, the command `ros2 topic hz /camera/depth/color/points` stops and the terminal that I ran `rs_launch.py` shows the numerous error of `No stream match for pointcloud chosen texture Process - Color`. 

How can I solve this error and after solving this issue, how can I raise the fps of the pointcloud? 

thank you!

MartyG-RealSense, 2024-01-04:
Hi @kenloumixx  The message **No stream match for pointcloud chosen texture Process - Color** means that color frames are being dropped.

Because the message is accompanied by **control_transfer returned error, index: 768, error: Resource temporarily unavailable**, it suggests that the problem may be with the USB port or the USB cable.  Are you using the official 1 meter long USB cable supplied with the camera, please?

kenloumixx, 2024-01-08:
@MartyG-RealSense Yes, I'm using the official cable supplied with the camera. 
And I also flipped up and down the micro usb cable and re-insert it into the camera but the warning still occurs. 
I add initial_reset:=true args into the launch command but it stops at the line 'Resetting device...' and never goes on.

MartyG-RealSense, 2024-01-08:
Does the FPS improve if you disable the RGB option **auto_exposure_priority** by inputting the instruction below into the terminal after launch has completed?

`ros2 param set /camera/camera rgb_camera.auto_exposure_priority false`

kenloumixx, 2024-01-09:
@MartyG-RealSense I also ran the command `ros2 param set /camera/camera rgb_camera.auto_exposure_priority false` but it does not significantly increase the pointcloud fps..

kenloumixx, 2024-01-15:
@MartyG-RealSense The result of running the command `ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true` was the same low fps. Meanwhile, I changed my DDS from cyclonedds to fastrtps, then the fps increased over 10! I should have share my DDS setting haha thank you for your support :)

closed 2024-01-15 by kenloumixx

MartyG-RealSense, 2024-01-15:
You are very welcome.  I'm pleased to hear that you found a solution.  Thanks very much for the update!