realsenseai/realsense-ros #2965: unable to generate pointcloud in rviz (melodic)

alwisy64, 2024-01-07:
Hello!

I am trying to generate a pointcloud by running `roslauch realsense2_camera rs_camera.launch filters:=pointcloud`
I can see the depthcloud in rviz but cannot see a pointcloud nor do I see any messages being published to /camera/depth/color/points.

I saw in [this comment](https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-876615270) that downgrading librealsense2/realsense-ros can solve the issue but I am unsure how to do this or if there is an easier solution.

[image]

Thanks!

MartyG-RealSense, 2024-01-07:
Hi @alwisy64  Please see if you can successfully view a pointcloud in the RealSense ROS1 wrapper using an **RGBD** launch instead of rs_camera.  Instructions for doing so can be found at https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-1468076630

alwisy64, 2024-01-08:
Hi Marty,

Thanks for your reply. I was able to install and launch using rs_rgbd.launch (I tried with and without a pointcloud filter) but neither were able to generate a pointcloud and I did not see anything being published to depth_image_proc.

MartyG-RealSense, 2024-01-08:
There does not seem be any RGB color information in your image, only depth data.  

Which RealSense camera model are you using, please?

In the log of your launch, which name does the camera have on the 'Device with name' line?

[image]

alwisy64, 2024-01-09:
I am using an Intel RealSense L515

MartyG-RealSense, 2024-01-09:
Have you tried unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).  If you have not tried this yet, does doing so make any difference?

alwisy64, 2024-01-10:
I hadn't tried that before, but it did not seem to make a difference just now.

MartyG-RealSense, 2024-01-10:
Are there any RGB color messages on /camera/camera/color/image_raw

alwisy64, 2024-01-19:
No, I don’t see any messages being published there either

MartyG-RealSense, 2024-01-19:
The rs_camera launch file should provide a pointcloud in RViz like the one in the image below when the pointcloud filter is enabled.

[image]

I note that in the bottom half of your RViz side panel, the color image is not shown like it is in the one above.

What happens if you click the **Add** button in the side panel and then select the **By Topic** option and the topic **/camera/color/image_raw**

MartyG-RealSense, 2024-01-26:
Hi @alwisy64  Do you require further assistance with this case, please?  Thanks!

alwisy64, 2024-01-26:
Hi, sorry for the delayed reply. After trying rs_camera.launch with the pointcloud filter enabled, these were the topics that were available to add. I didn't have the choice of /camera/color/image_raw. I also tried /camera/depth/color/points/PointCloud2, which doesn't seem to generate anything, but /camera/depth/image_rect_raw is able to create the image on the bottom left.

[image]

MartyG-RealSense, 2024-01-27:
Which RealSense camera model are you using, please?

alwisy64, 2024-01-27:
I'm using the L515

MartyG-RealSense, 2024-01-27:
Thanks for the reminder about your camera model.

Do you have access to the RealSense Viewer tool?  If you do, is the RGB color stream able to be displayed?

alwisy64, 2024-01-30:
Yes, I've tested the camera using the viewer tool and the color stream works fine there. If it makes a difference, I don't necessarily need the color output, just to be able to create and visualize the pointcloud data.

MartyG-RealSense, 2024-01-31:
If you do not necessarily need the color output, are you able to texture the pointcloud with the **infrared** stream instead of the color stream by using the roslaunch command below?

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_INFRARED`

MartyG-RealSense, 2024-02-02:
The infrared topics are disabled by default in the ROS wrapper.  Are you able to get the **infra1** topic and texture a pointcloud with it if you add **enable_infra1:=true** to your roslaunch instruction, please?

alwisy64, 2024-02-04:
I gave it a shot, but it did not seem to make a difference still. I don't see any additional topics available on rviz. Here is the roslaunch command I used to make sure I didn't miss something.

[image]
[image]

MartyG-RealSense, 2024-02-05:
The lack of color in RViz, no infra1 topic when enabled and no pointcloud when using rs_rgbd.launch suggests that there is a problem in this ROS1 wrapper installation.

What methods did you use to install the librealsense SDK and RealSense ROS1 wrapper please?  For example, if you installed librealsense first and then installed the ROS1 wrapper from packages with `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` then you could end up with two conflicting librealsense installations on the same computer.  This is because the apt-get instruction installs both librealsense and the wrapper at the same time.

alwisy64, 2024-02-12:
I'm not completely sure since I installed it a while ago. I don't think I installed librealsense separately though, I think I just used `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` to install the wrapper. Is there a way to check that I don't have multiple conflicting installations?

Also, I'm not running Linux natively. I had been using VirtualBox but switched to VMWare Workstation. Now I'm able to get color output from the camera, but I still am unable to generate a pointcloud or use the infra1 topic.

MartyG-RealSense, 2024-02-12:
If you had installed librealsense from both source code and packages (which wold count as two separate conflicting installations) then the easiest way to tell would be to launch the RealSense Viewer tool and see if a red error message **Multiple realsense udev-rules were found! ** appears in the top corner of the Viewer window.

alwisy64, 2024-02-20:
Hi, sorry for the delay again. It seems switching virtual machines resolved most of the issues for me. No further assistance needed, thanks!

closed 2024-02-20 by alwisy64