realsenseai/realsense-ros #2969: Collision and visual do not align for d435

samibouziri, 2024-01-09:
Hello,

I noticed that the mesh and collision box do not align for the the d435 as depicted in the image bellow.
[image]
The origin of that problem is that the x coordinate of the origin of the collision box for link ${name}_link seems to be wrong. It should be -d435_cam_depth/2+d435_zero_depth_to_glass + d435_glass_to_front instead of 0.
Same goes for x coordinate of the visual if we are not using the mesh.

MartyG-RealSense, 2024-01-10:
Hi @samibouziri  Are you using a URDF, please?  If you are then the above image looks unusual for a D435.  A D435 URDF should look like the one at https://github.com/IntelRealSense/realsense-ros/issues/459#issuecomment-420505591

[image]

samibouziri, 2024-01-10:
Yes I am using this [urdf](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_description/urdf/_d435.urdf.xacro) I am just enabling the visualization of the mesh and the corresponding collision box in the same picture to demonstrate the offset.
here is the mesh alone:
[image]
here is the collision box alone:
[image]
and here are the two overlapping:
[image]
Sorry for the confusion

MartyG-RealSense, 2024-01-10:
It's no problem at all.  :)

The mesh image looks correct.  base_link corresponds to the tripod screw thread on the base of the camera and camera_link corresponds to the center-line of the left infrared sensor on the front of the camera.

On RealSense cameras with a glass front such as D435, depth is measured from the front glass and not from the sensor inside the casing.  The depth measurement is 0 at the glass front.  To find the 'ground truth' value (the depth when measured from the sensor) on a D435 camera, -3.2 mm is added to the depth measurement.

samibouziri, 2024-01-10:
I am not talking about the frames or the measurements. I am talking about the position of the **collision box**. The **collision box** of the camera should encompass the camera mesh which is not the case in the picture above. 
The collision box should not be centered around the camera frame (x coordinate of the origin of the collision box 0) as the camera is not positioned in the center of the camera.
In other words when visualizing both the mesh and the collision box I am expecting to see something like this :
[image]
and **not** something like this:
[image]

MartyG-RealSense, 2024-01-10:
I have not seen this particular problem in past RealSense ROS cases, unfortunately.

Are you using ROS1 or ROS2?  You linked to the ROS2 wrapper's D435 URDF.  If you are using the ROS1 wrapper then it has a D435 URDF here:

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy/realsense2_description/urdf

samibouziri, 2024-01-10:
I am using ROS1 but the problem should be the same for ROS2 since the URDFs are the same and the meshes are the same. 
Anyway if you want to align the collision box with the mesh like shown just replace the collision tag which is now:

<collision>
        <origin xyz="0 ${-d435_cam_depth_py} 0" rpy="0 0 0"/>
        <geometry>
          <box size="${d435_cam_depth} ${d435_cam_width} ${d435_cam_height}"/>
        </geometry>
</collision>

by:

<collision>
        <origin xyz="${-d435_cam_depth / 2 + d435_zero_depth_to_glass + d435_glass_to_front instead} ${-d435_cam_depth_py} 0" rpy="0 0 0"/>
        <geometry>
          <box size="${d435_cam_depth} ${d435_cam_width} ${d435_cam_height}"/>
        </geometry>
</collision>

MartyG-RealSense, 2024-01-10:
It's great to hear that you achieved a solution.  Thanks so much for sharing it!

closed 2024-01-11 by samibouziri