realsenseai/realsense-ros #2970: How to reduce the effect of the sunlight and object color? [Depth Camera, L515)

kevinDrawn, 2024-01-10:
Hello @MartyG-RealSense,

I am encountering two problems with the depth camera (model: L515) that I'm using for a mobile manipulator. It is utilized both in the morning and at night, (indoor of the hospital). The depth values of the same relative place differ depending on the sunlight passing through the window, which changes with the time of day.

The second issue involves a black-colored sofa and refrigerator. When I view these objects through the depth map, they are not well recognized, especially in the black areas. 

How can I improve the precision quality? Could you recommend a filter, or should I consider changing the camera model?

Thank you.

MartyG-RealSense, 2024-01-10:
A strong light source on a black surface should be less likely to result in a glare that blinds the camera than a lighter surface such as white.

kevinDrawn, 2024-01-10:
I totally understand.

Lastly, where can i change the configuration of the variable light conditions?

Thank you a lot!

kevinDrawn, 2024-01-10:
Thank you so much!

In https://github.com/IntelRealSense/librealsense/issues/6757, it is recommended to use the D435 camera, which has IR recognition. So, can the D435 model provide more stable precision on black or gray surfaces than L515?

MartyG-RealSense, 2024-01-10:
Difficulty in reading gray / black surfaces is a general physics problem rather than a RealSense-specific one. However, using a 400 Series camera such as D435 may provide an advantage in regard to its built-in infrared projector component, which can project a pattern of IR dots onto a surface to aid the camera's depth analysis.

[image]

kevinDrawn, 2024-01-11:
Thank you!! I'll try it later.

closed 2024-01-28 by MartyG-RealSense