realsenseai/realsense-ros #2972: Issue with RGB Data Retrieval in librealsense 2.54.2 and realsense-ros 4.54.1

gladgaj, 2024-01-12:
We are currently using librealsense version 2.54.2 and realsense-ros version 4.54.1 in our project. The launch command used is as follows: ros2 launch realsense2_camera rs_launch.py.

However, we are encountering an issue where the RGB data is not being retrieved successfully, leading to the inability to generate a point cloud. The following error messages are observed:

Copy code
```
[realsense2_camera_node-1] [INFO] [1705034723.018501752] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [WARN] [1705034723.029958202] [camera.camera]:
[realsense2_camera_node-1] [INFO] [1705034723.038908011] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] 12/01 04:45:23,345 WARNING [546660407616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] 12/01 04:45:23,658 WARNING [546660407616] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[realsense2_camera_node-1] 12/01 04:45:23,669 WARNING [546660407616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] 12/01 04:45:24,674 ERROR [546173860160] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
[realsense2_camera_node-1] [WARN] [1705034749.883937757] [camera.camera]: No stream match for pointcloud chosen texture Process - Color

```
## Steps to Reproduce:

Execute the command: ros2 launch realsense2_camera rs_launch.py.
Observe the error messages related to RGB data retrieval.

## Expected Behavior:
RGB data should be successfully retrieved, allowing for the proper generation of a point cloud.

## Actual Behavior:
RGB data retrieval is failing, resulting in the inability to generate a point cloud.

## Environment:

Operating System: ubuntu 18.04
ROS Version: ROS2 humble
librealsense Version: 2.54.2
realsense-ros Version: 4.54.1

Please let us know if you need any further information or clarification on the matter. Thank you for your assistance in resolving this RGB data retrieval issue.

MartyG-RealSense, 2024-01-12:
Hi @gladgaj  ROS2 Humble is intended to be used with Ubuntu 22,04 rather than 18.04, as noted in the following item from the RealSense ROS2 wrapper instructions.

[image]

The best ROS2 version match for Ubuntu 18.04 is **Eloquent**.  

The 4.54.1 wrapper does not support Eloquent, though the earlier 4.51.1 wrapper for librealsense 2.51.1 did support it.

In regard to the message **No stream match for pointcloud chosen texture Process - Color**, this indicates that color frames are missing, possibly due to frame drops.

I note that you are launching with 'ros2 launch realsense2_camera rs_launch.py'.  That launch instruction will not publish a pointcloud, as the pointcloud needs to be enabled in the launch instruction:

`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true`

gladgaj, 2024-01-15:
Hi, thanks for the reply.
I apologize for the mistake I wrote previous question. Can you reply for **the below setting** and the configuration?

## ROS version: ROS2 galatic (Building from source -> no problem with the Ubuntu version)

## Configuration for the rs_launch.py file

```
# Copyright 2023 Intel Corporation. All Rights Reserved.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     <http://www.apache.org/licenses/LICENSE-2.0>
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

"""Launch realsense2_camera node."""
import os
import yaml
from launch import LaunchDescription
import launch_ros.actions
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.substitutions import LaunchConfiguration

configurable_parameters = [{'name': 'camera_name',                  'default': 'camera', 'description': 'camera unique name'},
                           {'name': 'serial_no',                    'default': "''", 'description': 'choose device by serial number'},
                           {'name': 'usb_port_id',                  'default': "''", 'description': 'choose device by usb port id'},
                           {'name': 'device_type',                  'default': "''", 'description': 'choose device by type'},
                           {'name': 'config_file',                  'default': "''", 'description': 'yaml config file'},
                           {'name': 'unite_imu_method',             'default': "0", 'description': '[0-None, 1-copy, 2-linear_interpolation]'},
                           {'name': 'json_file_path',               'default': "''", 'description': 'allows advanced configuration'},
                           {'name': 'log_level',                    'default': 'info', 'description': 'debug log level [DEBUG|INFO|WARN|ERROR|FATAL]'},
                           {'name': 'output',                       'default': 'screen', 'description': 'pipe node output [screen|log]'},
                           {'name': 'depth_module.profile',         'default': '0,0,0', 'description': 'depth module profile'},
                           {'name': 'enable_depth',                 'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'rgb_camera.profile',           'default': '0,0,0', 'description': 'color image width'},
                           {'name': 'rgb_camera.enable_auto_exposure', 'default': 'true', 'description': 'enable/disable auto exposure for color image'},
                           {'name': 'enable_color',                 'default': 'true', 'description': 'enable color stream'},
                           {'name': 'enable_infra1',                'default': 'false', 'description': 'enable infra1 stream'},
                           {'name': 'enable_infra2',                'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'infra_rgb',                    'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'enable_fisheye1',              'default': 'true', 'description': 'enable fisheye1 stream'},
                           {'name': 'enable_fisheye2',              'default': 'true', 'description': 'enable fisheye2 stream'},
                           {'name': 'enable_confidence',            'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'gyro_fps',                     'default': '0', 'description': "''"},
                           {'name': 'accel_fps',                    'default': '0', 'description': "''"},
                           {'name': 'enable_gyro',                  'default': 'false', 'description': "''"},
                           {'name': 'enable_accel',                 'default': 'false', 'description': "''"},
                           {'name': 'enable_pose',                  'default': 'true', 'description': "''"},
                           {'name': 'pose_fps',                     'default': '200', 'description': "''"},
                           {'name': 'pointcloud.enable',            'default': 'true', 'description': ''},
                           {'name': 'pointcloud.stream_filter',     'default': '2', 'description': 'texture stream for pointcloud'},
                           {'name': 'pointcloud.stream_index_filter','default': '0', 'description': 'texture stream index for pointcloud'},
                           {'name': 'enable_sync',                  'default': 'true', 'description': "''"},
                           {'name': 'align_depth.enable',           'default': 'true', 'description': "''"},
                           {'name': 'colorizer.enable',             'default': 'false', 'description': "''"},
                           {'name': 'clip_distance',                'default': '-2.', 'description': "''"},
                           {'name': 'linear_accel_cov',             'default': '0.01', 'description': "''"},
                           {'name': 'initial_reset',                'default': 'false', 'description': "''"},
[... cut ...]
[realsense2_camera_node-1]  15/01 01:22:06,076 ERROR [546962381120] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
[realsense2_camera_node-1]  15/01 01:22:13,227 WARNING [547381819712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  15/01 01:22:17,244 WARNING [547381819712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

MartyG-RealSense, 2024-01-15:
Wrapper 4.54.1 does not support Galactic, only Humble and Iron.  Galactic should also be used with Ubuntu 20.04 rather than 18.04.

Wrapper 4.51.1 supports Galactic.  This wrapper should be used with librealsense 2.51.1.

What happens if you try the launch instruction below with your current RealSense ROS installation?

`ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true initial_reset:=true`

closed 2024-01-28 by MartyG-RealSense