realsenseai/realsense-ros #2973: Humble apt/source conflict?

martinerk0, 2024-01-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455                        | 
| Firmware Version                   | 5.15.1                         | 
| Operating System & Version |   Ubuntu 22.04.3 LTS  | 
| Kernel Version (Linux Only)    |  5.15.122-tegra                                                                    | 
| Platform                                 | Jetson Orin Nano devkit                         |
| Librealsense SDK Version       | latest main branch from source                                           | 
| Language                               | -       | 
| Segment			         |  -                          | 
| ROS Distro			         |  humble                                     | 
| RealSense ROS Wrapper Version   |   latest main branch from source /  apt                                                  | 

### Issue Description
I don't know if it's bug but surely some people might get confused, I installed SDK by script https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh on Orin, but then when I wanted to install ros wrapper by apt:
`sudo apt install ros-humble-realsense2-*` as in Step 3 of installation, it also installs `ros-humble-librealsense2` which I guess is SDK for humble? And then you get conflicts in the sense https://github.com/IntelRealSense/librealsense/issues/12566 where the viewer (and everything) runs only with sudo.

MartyG-RealSense, 2024-01-12:
Hi @martinerk0  Yes, there is the potential for conflicts between two librealsense installations if librealsense is installed first and then a ROS wrapper install procedure that installs both librealsense and the wrapper at the same time is used.

If the RealSense Viewer tool is used then the ideal method to follow is to install librealsense first from packages or source code, and then build the ROS wrapper from source afterwards to ensure that librealsense is not installed a second time.

MartyG-RealSense, 2024-01-20:
Hi @martinerk0  Do you require further assistance with this case, please?  Thanks!

martinerk0, 2024-01-21:
@MartyG-RealSense can I ask what is the purpose of `ros-humble-librealsense2`?

The documentation is still misleading, if in step 2 people install librealsense2 by apt or source, and then in step 3 by apt `ros-humble-librealsense2` they will have these weird permission problems due to both packages/libs installed.

MartyG-RealSense, 2024-01-21:
`ros-humble-librealsense2` installs the ROS2 wrapper and necessary files using Debian packages instead of compiling the wrapper from source code.

I understand and appreciate that the instructions can be somewhat unclear in regards to performing a package installation with Option 1.  The same is true for the ROS1 wrapper and its Method 1 package instructions.

graham-fleming, 2024-01-24:
Will the instructions be updated? also having problems installing on a similar environment. 

Three steps with 5 different options plus these extra instruction pages is unclear.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md
https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

MartyG-RealSense, 2024-01-24:
I am not aware of any plans to adjust how the ROS wrapper's package / source instructions are presented.

The librealsense SDK needs be able to work on multiple different computing hardware platforms but one guide will not work for all of those platforms.  This is why there are different sets of instructions.  

The `installation_jetson.md` instructions are for installation on Nvidia Jetson computing boards only, whilst the `libuvc_installation.sh` instructions help to enable a successful installation when other methods have been tried and not worked and are especially suited for installing on Raspberry Pi boards with the Ubuntu operating system.

If you are using a Jetson board then the only official instruction page that should be used is `installation_jetson.md`.  There are also some unofficial guides contributed by Jetson users, such as the source code installation one at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

MartyG-RealSense, 2024-01-31:
Hi @martinerk0 and @graham-fleming  Do you require further assistance with this case, please?  Thanks!

MartyG-RealSense, 2024-02-20:
Case closed due to no further comments received.

closed 2024-02-20 by MartyG-RealSense

martinerk0, 2024-03-11:
I think documentation should be reviewed, since more people are having same problems https://github.com/IntelRealSense/librealsense/issues/12735

What do you think @MartyG-RealSense ?

MartyG-RealSense, 2024-03-11:
Hi @martinerk0  I have hughlighted this issue to the RealSense ROS team for their consideration.

SamerKhshiboun, 2024-03-29:
Readme was updated in #3056

MartyG-RealSense, 2024-03-30:
Thanks very much @SamerKhshiboun