realsenseai/realsense-ros #2974: Ubable to transform PointCloud in rviz

Karin-Sugi, 2024-01-17:
Hi, I'm trying to show PointCloud taken by fixed camera in rviz, but "Ubable to transform" is returned. I use realsense 435i and ros melodic.

In details, i placed two models (realsense camera and robot) in rviz. The camera detects robot's position with apriltag and all rocations are saved with tf2. I'd like to see PointCloud data with those models, but the rviz returns `Ubable to transform [/camera_depth_optical_frame] to [world]`.

However, when i check the tree with rqt, the `camera_depthoptical_frame` is connected to the `world` as you can see the image below...  Do you have any idea??
  
Actually, I've already checked both PointCloud is published correctly and all tf tree (including robot) is connected. Also, both works fine when i run it independently.

[image]

Karin-Sugi, 2024-01-18:
Hi, thanks for your quick reply!

I set publish_tf false, but it doesn't work. My launch file is below. I run realsense by including `rs_camera.launch`. Also, both robot_description and camera_description are loaded by rviz.
```
<?xml version="1.0" encoding="UTF-8"?>
<launch>
  <!-- configure camera input -->
  <arg name="camera" default="camera" />
  <arg name="camera_name" default="/$(arg camera)/color" />
  <arg name="image_topic" default="image_raw" />
  <arg name="queue_size" default="1" />

  <!-- apriltag_ros continuous detection node -->
  <node pkg="apriltag_ros" type="apriltag_ros_continuous_node" name="apriltag_ros_continuous_node" clear_params="true" output="screen" launch-prefix="$(arg launch_prefix)">
    <!-- Remap topics -->
    <remap from="image_rect" to="$(arg camera_name)/$(arg image_topic)" />
    <remap from="camera_info" to="$(arg camera_name)/camera_info" />
    <param name="publish_tag_detections_image" type="bool" value="true" /><!-- default: false -->
    <param name="queue_size" type="int" value="$(arg queue_size)" />

    <!-- load parameters (incl. tag family, tags, etc.) -->
    <rosparam command="load" file="$(find hauser_localization)/config/settings.yaml"/>
    <rosparam command="load" file="$(find hauser_localization)/config/tags.yaml"/>
  </node>

 <!-- run realsense with point cloud -->
  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera" value="$(arg camera)" />
    <arg name="filters" value="pointcloud"/>
    <arg name="publish_tf" value="false"/>
    <!-- <arg name="unite_imu_method" value="linear_interpolation"/>
    <arg name="enable_gyro" value="true"/>
    <arg name="enable_accel" value="true"/> -->
  </include>

  <!-- Show on rviz -->
  <include file="$(find hauser_viz)/launch/rviz/multi_robots.launch">
    <arg name="camera_name" value="$(arg camera_name)" />
  </include>
 
  <!-- Calcurate robot attitude -->
  <node pkg="hauser_localization" type="tags_center" name="tags_center" output="screen" />
  <node pkg="hauser_localization" type="tag2robot" name="tag2robot" output="screen" />

```

Actually, I found another strange message in rviz as you can see in following image. It says `Transform OK` and `Failed to transform`. What do they mean??
[image]
 to`

Thanks in advance,
Karin

Karin-Sugi, 2024-01-18:
Hi, when i delete camera_descriptio, new warning message was returned on terminal saying `[ WARN] [1705547684.766363468]: Invalid argument "/camera_depth_optical_frame" passed to canTransform argument source_frame in tf2 frame_ids cannot start with a '/' like:`.

The ` / ` on the top causes this problem? Then, how can i fix it?

MartyG-RealSense, 2024-01-18:
I do not know the answers to these questions unfortunately as my knowledge of TF is limited, and I carried out extensive research but could not find much relevant information either.  I'm very sorry.

A person in a case with a non-RealSense camera at the link below suggested dealing with a missing transform between /world and /camera_depth_optical_frame by creating a **static transform publisher**.

https://groups.google.com/g/moveit-users/c/KCIPoSP3dB0/m/hkul9NqTBAAJ

MartyG-RealSense, 2024-01-24:
Hi @Karin-Sugi  Bearing in mind the comment above, do you require further assistance with this case please?  Thanks!

closed 2024-01-31 by MartyG-RealSense