realsenseai/realsense-ros #2975: D455 URDF

anath93, 2024-01-17:
| Camera Model                       | D455                          
| Firmware Version                   |  5.15.1                         
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.15-rt                                                                     
| Platform                                 | Nvidia Jetson Orin                           
| Librealsense SDK Version       |  2.55.0                                                    
| Segment			         | Robot                           
| ROS Distro			         |  Galactic                                         
| RealSense ROS Wrapper Version   |  4.54.1                                            

### Issue Description
I am trying to add D455 URDF to my model, but don't think the offsets are right, what is the origin point on this camera as camera link in rviz is show towards RGB sensor but considering that the offset is wrong, please see image below 

[image]

[image]

[image]

MartyG-RealSense, 2024-01-17:
Hi @anath93  The origin of depth is the center-line of the left infrared sensor.  The physical position of the RGB sensor can differ though depending on the camera model.  

For example, on the D415 model the RGB sensor is on the right end of the front of the camera, and on D455 it is near to the center of the camera's front.  The left infrared sensor on D455 is on the right end.  

So the axes on the end of the camera in your image above represent the depth origin.

[image]

anath93, 2024-01-17:
@MartyG-RealSense Is this red circled the origin for adding URDF ?

[image]

MartyG-RealSense, 2024-01-17:
The circled sensor is the D455's RGB sensor.  camera_link is the left infrared sensor (on the end of the camera right-side on D455).

anath93, 2024-01-17:
@MartyG-RealSense Thank you for confirming that, so this is below 

[image]

Is it the middle of the camera or is there any offset from the glass ?

MartyG-RealSense, 2024-01-17:
In the D455 URDF file, the origin of the mesh is stated as being between the two infrared camera axes.  As the right infrared sensor is on the left end, that position would likely correspond to the camera center.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_description/urdf/_d455.urdf.xacro#L76

anath93, 2024-01-17:
@MartyG-RealSense Thank you !

closed 2024-01-17 by anath93