realsenseai/realsense-ros #2976: Understanding signal processing of L515

Daya-AWR1642, 2024-01-17:
Hi 
I had completed my experiment with L515 and wanted to understand the signal processing of L515. Still, unfortunately, very few publications used L515, and none of them explained the mathematical processing of Lidar Signal processing. 

I want the mathematical model reference 
can you please

MartyG-RealSense, 2024-01-17:
Hi @Daya-AWR1642  There is not a mathematical reference available to precisely describe the signal processing of the L515 camera model.  However, the official User Guide PDF document for the L515 camera model  - **L515_User_Guide_v1.0.pdf** - may be a helpful reference to better understand the L515's principles.  It can be downloaded as a PDF from the link below.

https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide

Daya-AWR1642, 2024-01-17:
Yeah, I knew this. Can you please suggest any IEEE papers, please

MartyG-RealSense, 2024-01-17:
The most suitable IEEE paper referencing the L515 that I could find is this one:

https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=10122518

Daya-AWR1642, 2024-01-17:
Haa, I have gone through this already. Anyway, thank you for your response. I'll try my own.

Daya-AWR1642, 2024-01-18:
Hi,
I have a few doubts about L515
1) is this direct detection lidar or coherent detection lidar?
(I guess is direct detection)
2) can I assume L515 getting processed like 

[image]
[image]
[image]
[image]
[image]

MartyG-RealSense, 2024-01-19:
Detailed information about the operating principles of the L515 model (which is now retired from the RealSense product range) is not available and so it is unfortunately not possible to provide confident answers to the above questions.

However, the L515 data sheet document provides the following description:

"The MEMS is used to scan the IR laser beam over the entire field of view (FOV).  The L515 vision ASIC will process the data from the reflected beam captured by the photodiode and will output a depth point representing the accurate distance of a specific point in the scene from the camera".

This description makes it sound like L515 is based on **direct** detection (measuring amplitude changes of the reflected light) rather than coherent (measuring changes in the phase of the reflected light).  I do not have the technical information to confirm this for certain though.

MartyG-RealSense, 2024-01-26:
Hi @Daya-AWR1642  Do you require further assistance with this case, please?  Thanks!

Daya-AWR1642, 2024-01-26:
No thank you, our team completed and submitted a research paper with L515.
And Special thank you for assisting each time.

closed 2024-01-26 by Daya-AWR1642

MartyG-RealSense, 2024-01-26:
You are very welcome.  I'm pleased that I could help.  Thanks very much for the update!