realsenseai/realsense-ros #2977: No image received in RIVZ

CYJ00, 2024-01-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D415                                                                                | 
| Operating System & Version |  Ubuntu 18                                                                      | 
| Kernel Version (Linux Only)    |  4.9.201                                                                           | 
| Platform                                 | NVIDIA Jetson Nano                                                        
| Segment			         |  Unitree GO1                                                                  | 
| ROS Distro			         |  Melodic                                                                         | 

I installed using sudo apt-get install ros-melodic-realsense2-camera

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello. First, sorry for my poor English. English is not my first language.

When I run `roslaunch realsense2_camera rs_camera.launch`, the `[ERROR] hwmon command 0x75( 0 1df 0 27f ) failed(response -6= Invalid parameter)` occures. But, the process is working after the error.
When I check the rostopic, I can find the topic, /camera/depth/color/points and /camera/color/image_raw.
But in rostopic echo /camera/depth/color/points and rostopic echo  /camera/color/image_raw there is no data.
But in rostopic echo /camera/depth/image_rect_raw, I can find the data. (It works well.)
I need the RGB data and point cloud. 
When I check in desktop in the docker(my desktop OS is ubuntu 20.04 and I used docker container(melodic).) , there was no problem.
So, definitely not hardware problem. When I use other camera, the same error occurs.
How can I fix it?

MartyG-RealSense, 2024-01-18:
Hi @CYJ00  Your English is very good, so don't worry.  :)

The pointcloud is not enabled by default.  To enable it, please add the pointcloud filter to your launch command.

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

You can also generate a pointcloud with an **RGBD** launch by following the instructions at https://github.com/IntelRealSense/librealsense/issues/10223#issuecomment-1152864261

CYJ00, 2024-01-19:
`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`
This wasn't working...
But the solution, https://github.com/IntelRealSense/librealsense/issues/10223#issuecomment-1152864261 is worked!
Thank you!

MartyG-RealSense, 2024-01-19:
You are very welcome.  It's excellent to hear that an RGBD launch worked for you.  Thanks very much for the update!

closed 2024-01-22 by CYJ00