realsenseai/realsense-ros #2978: realsense-ros-2.3.1 not building on raspi-4

ta-jetson, 2024-01-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400, T200 } | 
| Operating System & Version |   Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)    |  5.4.0-1100-raspi                                        | 
| Platform                                 |Raspberry Pi -4 model B+ |
| SDK Version                            |  2.48.0                        |  
|ROS           | Noetic |

### Issue Description

Hey @MartyG-RealSense , 
Hope you're doing well.
I am trying to build realsense-ros 2.3.1 and librealsense 2.48.0 to use my realsense cameras [D435i and T265] on raspi-4. While the librealsense worked and I can successfully use my realsense-viewer as well as pyrealsense2 (thanks to your help with my issue  : Not able to use pyrealsense2. "No Module named pyrealsense2" #12377 ).
Building realsense-ros-2.3.1 has issues. These are the steps I followed :

mkdir ~/catkin_realsenseros_ws/src
cd ~/catkin_realsenseros_ws/src

Here I extracted the realsense-ros-2.3.1.zip from realsense-ros releases and  tried two approaches :
Approach -1:
cd ..
catkin build realsense2_camera -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release -j1

Approach-2:
catkin_init_workspace
cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release

Both gave the same error as below:
user@raspberrypi:~/catkin_realsense_ws$ catkin build realsense2_camera -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release

--------------------------------------------------------------------------------------------------------------------------------------

Profile:                     default
Extending:             [env] /opt/ros/noetic
Workspace:                   /home/user/catkin_realsense_ws

Build Space:        [exists] /home/user/catkin_realsense_ws/build
Devel Space:        [exists] /home/user/catkin_realsense_ws/devel
Install Space:      [unused] /home/user/catkin_realsense_ws/install
Log Space:         [missing] /home/user/catkin_realsense_ws/logs
Source Space:       [exists] /home/user/catkin_realsense_ws/src
DESTDIR:            [unused] None

Devel Space Layout:          linked
Install Space Layout:        None

Additional CMake Args:       -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False

Buildlisted Packages:        None
Skiplisted Packages:         None

Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.

[build] Found 2 packages in 0.0 seconds.                                       
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 3.9 seconds ]             
Starting  >>> realsense2_camera                                                
_______________________________________________________________________________________________
[... 189 lines cut ...]
      |        ^~~~~~~
make[2]: *** [CMakeFiles/realsense2_camera.dir/build.make:76: CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:246: CMakeFiles/realsense2_camera.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
cd /home/user/catkin_realsense_ws/build/realsense2_camera; catkin build --get-env realsense2_camera | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...............................................................................................
Failed     << realsense2_camera:make               [ Exited with code 2 ]                      
Failed    <<< realsense2_camera                    [ 52.3 seconds ]                            
[build] Summary: 1 of 2 packages succeeded.                                                    
[build]   Ignored:   1 packages were skipped or are skiplisted.                                
[build]   Warnings:  None.                                                                     
[build]   Abandoned: None.                                                                     
[build]   Failed:    1 packages failed.                                                        
[build] Runtime: 56.2 seconds total.                                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

I tried again by adding these lines to the CMakelists.txt of realsense2_camera:
set(CMAKE_CXX_STANDARD_14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
add_compile_options(-std=c++14)

No change (Same error). 
Could you (@MartyG-RealSense ) please suggest on how to proceed?

MartyG-RealSense, 2024-01-18:
Hi @ta-jetson  The T265 camera model is no longer supported by Intel, so I can only assist in relation to D435i.

The **eleccelerator** guide at the link below for installing the RealSense ROS1 wrapper on Raspberry Pi 4 may be a helpful reference.

https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense#Installing_ROS_Package:_realsense-ros

ta-jetson, 2024-01-18:
Hey @MartyG-RealSense , 
I was following the reference you provided and tried to checkout v2.3.1. I got the following error.

-----------------------------------------------

tsalla@raspberrypi:~/catkin_realsense_ws/src$ git clone https://github.com/IntelRealSense/realsense-ros.git
Cloning into 'realsense-ros'...
remote: Enumerating objects: 16812, done.
remote: Counting objects: 100% (1919/1919), done.
remote: Compressing objects: 100% (753/753), done.
remote: Total 16812 (delta 1331), reused 1560 (delta 1094), pack-reused 14893
Receiving objects: 100% (16812/16812), 48.31 MiB | 5.25 MiB/s, done.
Resolving deltas: 100% (10742/10742), done.
tsalla@raspberrypi:~/catkin_realsense_ws/src$ cd realsense-ros/
tsalla@raspberrypi:~/catkin_realsense_ws/src/realsense-ros$ git checkout v2.3.1
error: pathspec 'v2.3.1' did not match any file(s) known to git

ta-jetson, 2024-01-18:
Hey @MartyG-RealSense , 

This is what git checkout gave me

git checkout `git tag v2.3.1`
Your branch is up to date with 'origin/ros2-development'.

I want ROS1, not ROS2. Nevertheless I followed the documentation until catkin_make and got the following error:

~/catkin_realsense_ws$ catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
Base path: /home/tsalla/catkin_realsense_ws
Source space: /home/tsalla/catkin_realsense_ws/src
Build space: /home/tsalla/catkin_realsense_ws/build
Devel space: /home/tsalla/catkin_realsense_ws/devel
Install space: /home/tsalla/catkin_realsense_ws/install
####
#### Running command: "cmake /home/tsalla/catkin_realsense_ws/src -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/tsalla/catkin_realsense_ws/devel -DCMAKE_INSTALL_PREFIX=/home/tsalla/catkin_realsense_ws/install -G Unix Makefiles" in "/home/tsalla/catkin_realsense_ws/build"
####
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/tsalla/catkin_realsense_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: False
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - realsense2_camera_msgs (plain cmake)
-- ~~  - realsense2_description (plain cmake)
-- ~~  - realsense2_camera (plain cmake)
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/tsalla/catkin_realsense_ws/build/CMakeFiles/CMakeOutput.log".
Invoking "cmake" failed

MartyG-RealSense, 2024-01-18:
It may be necessary to install the latest available ROS1 version, 2.3.2, though it is not ideal as it was intended for use with the next librealsense version after 2.48.0, which was 2.50.0.

First, delete the entire **catkin_ws\src** catkin workspace folder to ensure that any previous ROS wrapper installation is removed.

Then do this:

`git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1``

ta-jetson, 2024-01-18:
Hey @MartyG-RealSense ,

Thank you, your suggestion helped. 

Solution to issue:
I just followed the documentation in Intel Realsense ROS [https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy](url), with the branch ROS1-legacy.

Thanks again @MartyG-RealSense

closed 2024-01-18 by ta-jetson

MartyG-RealSense, 2024-01-18:
That's great to hear, @ta-jetson - thanks very much for the update!