realsenseai/realsense-ros #2981: ddynamic_reconfigure not working for ROS Melodic

jolags313, 2024-01-23:
Hello, I am trying to use the ROS1 legacy wrapper for ROS Melodic, and I have been unable to install the ddynamic-reconfigure package to have everything working properly. When I try to install it using `sudo apt-get install ros-melodic-ddynamic-reconfigure`, I get the error `E: Unable to locate package ros-melodic-ddynamic-reconfigure`, even after I run ` sudo apt-get update`. 

I then tried to install it from source from https://github.com/pal-robotics/ddynamic_reconfigure/tree/kinetic-devel. Although this built, when I ran rs_camera.launch, I got the following error

[ INFO] [1706045095.132528729]: Setting Dynamic reconfig parameters.
/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /home/conlab/slam_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE
[camera/realsense2_camera_manager-2] process has died [pid 15637, exit code 127, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/conlab/.ros/log/d97e1d46-ba35-11ee-9fab-ef2fad7123a9/camera-realsense2_camera_manager-2.log].
log file: /home/conlab/.ros/log/d97e1d46-ba35-11ee-9fab-ef2fad7123a9/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/conlab/.ros/log/d97e1d46-ba35-11ee-9fab-ef2fad7123a9/camera-realsense2_camera-3*.log

I'm guessing this is because the code is from the kinetic-devel branch, while I am using ROS Melodic. Is there a fix for this? I've looked through similar issues, but I haven't been able to use `sudo apt-get install ros-melodic-ddynamic-reconfigure` which is what appeared to work for others.

jolags313, 2024-01-23:
For some more context, I believe I have installed the correct versions of everything (librealsense 2.50.0 with firmware 5.13.0.50), and I would rather install the ROS wrapper from source and not using `sudo apt-get install` since I would like to change some of the code.

MartyG-RealSense, 2024-01-23:
Hi @jolags313  I recommend completely deleting the catkin workspace folder that you built the wrapper from source in so that previous installations of the wrapper are completely removed before attempting another build.  The official source code build instructions for the ROS1 wrapper can be found at the link below.

https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/README.md

At the point in the instructions where it suggests cloning ddynamic_reconfigure into your catkin workspace, ignore this and instead install ddynamic_reconfigure using `sudo apt install ros-noetic-ddynamic-reconfigure`

jolags313, 2024-01-23:
Is if fine if I just delete all of the wrapper's folders in the build/devel/src folders? I have other things in the workspace, so I'd rather not delete them. Also, I had built the workspace using catkin build, not catkin_make, so I have some questions regarding the last part below 

```
catkin_init_workspace
cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
```
Here, would I be able to just navigate into the realsense2_camera + description packages and use `catkin clean --this` in place of `catkin_make clean` and then `catkin build` the entire workspace afterwards?

MartyG-RealSense, 2024-01-23:
I am not familiar with installing the wrapper with catkin build instead of catkin_make and so I do not have advice to offer about using it instead of catkin_make or how / where to clean, unfortunately.  I do apologize.

MartyG-RealSense, 2024-02-02:
Hi @jolags313  Bearing in mind the information in the comment above, do you require further assistance with this case please?  Thanks!

closed 2024-02-09 by MartyG-RealSense