realsenseai/realsense-ros #2982: HW not ready error after fixing an IMU issue on Jetson

TakShimoda, 2024-01-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | {  D435i  }                          | 
| Firmware Version                   | (Open RealSense Viewer --> 5.15.1)                          | 
| Operating System & Version |   { Linux (Ubuntu 20 } | 
| Kernel Version (Linux Only)    |  (4.9.253)                                                                       | 
| Platform                                 | NVIDIA Jetson Nano                           |
| Librealsense SDK Version       |  { 2.51.1 }                                                    | 
| Language                               |  {python }         | 
| Segment			         |  {Robot }                             | 
| ROS Distro			         |  {foxy }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1}                                                  | 

### Issue Description
Hello, I'm running on a Jetson nano with the library versions specified above. I installed the realsense SDK as per the [instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) for Jetson, then installed the ros2 wrapper for foxy. 

I was previously encountering an issue with setting the ros2 parameters, when trying to run the command ros2 launch `realsense2_camera rs_launch.py unite_imu_method:=1 enable_gyro:=true enable_accel:=true` . It would give the following error in the output:

`[ERROR] [1705959753.179528268] [camera.camera]: /tmp/binarydeb/ros-foxy-realsense2-camera-4.51.1/src/rs_node_setup.cpp:344:An exception has been thrown: Failed to set frequency 1. device path: /sys/devices/70090000.xusb/usb2/2-1/2-1.1/2-1.1:1.5/0003:8086:0B3A.000A/HID-SENSOR-200073.1.auto/iio:device1/in_accel_sampling_frequency Last Error: Permission denied`

After looking online for the same/similar issue, I found that [udev-rules](https://github.com/IntelRealSense/librealsense/issues/3242#issuecomment-464775017) weren't in place, and tried re-installing the SDK, and also manually did `sudo apt-get install librealsense2-udev-rules`  just in case.

It seemed to have fixed the issue, as I can receive the /camera/imu topic and those error messages are gone. But now, every time I launch `rs_launch.py`, I encounter the error message:

`[realsense2_camera_node-1] [ERROR] [1706044263.652393112] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_exposure_limit:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)`

There is also a related warning which was always there even before fixing the IMU issue:

`[realsense2_camera_node-1] [WARN] [1706044263.676038384] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.`

As far as I understand, the realsense requires a lot of power and doesn't come across this issue on a PC like it does with the Jetson Nano, and so far, I can receive all the relevant topics (image_raw, imu, etc.). If so, is this an issue, or can I ignore it? I didn't have this error message before fixing the imu error but now it always shows this message. I followed all the steps done [here](https://github.com/IntelRealSense/realsense-ros/issues/1633) but they didn't work (and it seems the user installed from source for the fix but I have installed the debians).

Thanks.

MartyG-RealSense, 2024-01-23:
Hi @TakShimoda  The **auto_exposure_limit** error usually occurs because there is a mismatch between the librealsense SDK version and the firmware driver version installed in the camera.  With the 2.51.1 version of librealsense, the firmware version installed in the camera should be **5.13.0.50**.  If a newer firmware such as 5.15.1.0 is used with librealsense 2.51.1 then the auto_exposure_limit error occurs.  Using firmware 5.13.0.50 should cause the error to disappear.

TakShimoda, 2024-01-24:
Hello MartyG,

Thank you for the quick response. Going off the [firmware downloads page](https://dev.intelrealsense.com/docs/firmware-releases), I downloaded firmware 5.13.0.50 as recommended for SDK 2.51.1.  I couldn't find official guides on downgrading firmware, but I found links such as [here](https://github.com/IntelRealSense/librealsense/issues/8186#issuecomment-762807389). From this specific answer, there's no option on the realsense-viewer V2.54.2 to **Install Recommended Firmware** but there is **Update Firmware**. From here, would I select the .bin file from the downloaded firmware folder? I've never changed the firmware before so I just wanted to make sure this was the correct process.

Regards,

MartyG-RealSense, 2024-01-24:
Yes, you are correct.  First, please download the firmware bin file for 5.13.0.50 from the firmware downloads page.  Then go to the Update Firmware option in realsense-viewer.  A file browser window will pop up.  

Navigate to the location on your computer that you downloaded the bin file to and select it.  The firmware update to version 5.13.0.50 will then automatically begin.

TakShimoda, 2024-01-24:
Thank you MartyG. I'm just having trouble right now with my Jetson, but when I get it running I'll try the firmware update and close this issue if it works.

Regards,

MartyG-RealSense, 2024-01-24:
You are very welcome.  I look forward to your next update.  Good luck!

TakShimoda, 2024-01-25:
Thank you, the firmware change fixed the problem

closed 2024-01-25 by TakShimoda

MartyG-RealSense, 2024-01-25:
You are very welcome.  It's excellent to hear that a firmware update resolved the problem.  Thanks very much for the update!