realsenseai/realsense-ros #2985: ROS Realsense package crashes every time.

kennethhoffmann, 2024-01-31:
Hi all, having the following issue:
Any help is appreciated! Happy to provide more info. 
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | {D435 / D435i}                          | 
| Firmware Version                   | See below                        | 
| Operating System & Version |   { Linux } | 
| Kernel Version (Linux Only)    |  4.9 (Jetson/Nvidia Firmware                                                                     | 
| Platform                                 | Jetson                          |
| Librealsense SDK Version       |  { 2.50.0}                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  noetic                      | 
| RealSense ROS Wrapper Version   | Unsure                                               | 

### Issue Description

ROS realsense node is not launching. No combination of initial reset or different image size etc is helping.
###

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://ubuntu.local:11311

process[camera/realsense2_camera_manager-1]: started with pid [8654]
process[camera/realsense2_camera-2]: started with pid [8655]
[ INFO] [1706667101.473538242]: Initializing nodelet with 4 worker threads.
[ INFO] [1706667101.811989338]: RealSense ROS v2.3.2
[ INFO] [1706667101.812088036]: Built with LibRealSense v2.50.0
[ INFO] [1706667101.812157880]: Running with LibRealSense v2.50.0
[ INFO] [1706667101.943948297]:  
 30/01 18:11:42,010 WARNING [546995954032] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ INFO] [1706667102.245022570]: Device with serial number 109622073443 was found.

[ INFO] [1706667102.245141007]: Device with physical ID 2-1.2-3 was found.
[ INFO] [1706667102.245197882]: Device with name Intel RealSense D435I was found.
[ INFO] [1706667102.246528664]: Device with port number 2-1.2 was found.
[ INFO] [1706667102.246635591]: Device USB type: 3.2
[ INFO] [1706667102.246733612]: Resetting device...
[ INFO] [1706667108.396506611]:  
[ INFO] [1706667108.647037914]: Device with serial number 109622073443 was found.

[ INFO] [1706667108.651994529]: Device with physical ID 2-1.2-3 was found.
[ INFO] [1706667108.652064581]: Device with name Intel RealSense D435I was found.
[ INFO] [1706667108.653065779]: Device with port number 2-1.2 was found.
[ INFO] [1706667108.653137498]: Device USB type: 3.2
[ INFO] [1706667108.669456456]: getParameters...
[ INFO] [1706667109.525862866]: setupDevice...
[ INFO] [1706667109.525965210]: JSON file is not provided
[ INFO] [1706667109.526043179]: ROS Node Namespace: camera
[ INFO] [1706667109.526100731]: Device Name: Intel RealSense D435I
[ INFO] [1706667109.526151199]: Device Serial No: 109622073443
[ INFO] [1706667109.526199324]: Device physical port: 2-1.2-3
[ INFO] [1706667109.526246772]: Device FW version: 05.15.01.00
[ INFO] [1706667109.526292085]: Device Product ID: 0x0B3A
[ INFO] [1706667109.526336824]: Enable PointCloud: Off
[ INFO] [1706667109.526380418]: Align Depth: Off
[ INFO] [1706667109.526423804]: Sync Mode: Off
[ INFO] [1706667109.526526668]: Device Sensors: 
[ INFO] [1706667109.630185627]: Stereo Module was found.
[ INFO] [1706667109.699477242]: RGB Camera was found.
[ INFO] [1706667109.700363388]: Motion Module was found.
[ INFO] [1706667109.700574950]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1706667109.700716565]: num_filters: 0
[ INFO] [1706667109.700825523]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1706667115.022163769]: Done Setting Dynamic reconfig parameters.
[ INFO] [1706667115.023559394]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1706667115.026335956]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1706667115.026458561]: setupPublishers...
[ INFO] [1706667115.043128248]: Expected frequency for depth = 30.00000
[ INFO] [1706667115.068919863]: Expected frequency for color = 30.00000
[ INFO] [1706667115.095987675]: setupStreams...
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/uvify/.ros/log/b623c9ac-bfa0-11ee-a93d-ec2e98ca42cd/camera-realsense2_camera-2*.log
 30/01 18:11:55,348 WARNING [546652017008] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1706667115.398961791]: SELECTED BASE:Depth, 0
[ WARN] [1706667115.468862676]: 
 30/01 18:11:55,470 WARNING [546652017008] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1706667115.515453250]: RealSense Node Is Up!
 30/01 18:11:55,520 WARNING [546652017008] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 30/01 18:11:55,571 WARNING [546652017008] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
^C[camera/realsense2_camera_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Log:
  GNU nano 4.8                                          /home/uvify/.ros/log/b623c9ac-bfa0-11ee-a93d-ec2e98ca42cd/camera-realsense2_camera-2-stdout.log                                                    
^[[0m[ INFO] [1706667101.437632356]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:^[[0m
^[[0m[ INFO] [1706667101.478116107]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...^[[0m
^[[0m[ INFO] [1706667101.505035013]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.^[[0m
^[[0m[ INFO] [1706667114.896394654]: Bond broken, exiting^[[0m

MartyG-RealSense, 2024-01-31:
Hi @kennethhoffmann  I note that your camera has firmware driver version **5.15.1.0** installed.  Using this firmware with librealsense version 2.50.0 will cause errors as it is designed for 2.52.2.  Please downgrade your firmware version to **5.13.0.50**, the version that is designed for 2.50.0.

kennethhoffmann, 2024-01-31:
Unfortunately that does not make a difference. I think the problem is on the os/ubuntu side and may require debugging there. Are there any other logs that may be helpful?

MartyG-RealSense, 2024-02-01:
There is not another log that would help in this particular situation, unfortunately.

How did you install the librealsense SDK and the RealSense ROS1 wrapper, please?  

For example, if you installed librealsense first and then installed the ROS wrapper with **Method 1** (apt-get) afterwards then because Method 1 installs both librealsense and the wrapper at the same time you could end up with two conflicting librealsense installations on the same computer.

closed 2024-02-16 by MartyG-RealSense