realsenseai/realsense-ros #2986: Health Check?

taeyeonggim, 2024-01-31:
I have a question about realsense. Is the realsense has a health check topic or something?

I think that we should check the frame delayed or something else...
As soon as can repeat ASAP plz! Have a great day!

MartyG-RealSense, 2024-01-31:
Hi @taeyeonggim  The RealSense ROS wrapper publishes a **/diagnostics** topic.  The /diagnostics topic includes information regarding the device temperatures and actual frequency of the enabled streams.

taeyeonggim, 2024-01-31:
@MartyG-RealSense 
First of all, thanks for the answer, is it possible to get that information using the pyrealsense2 library, and if so, do you have any example code?

taeyeonggim, 2024-01-31:
Thank you for your quick response!!! I was able to resolve the issue thanks to you! Have a great day! @MartyG-RealSense 
Please close the issue if you saw this comment!
[image]

MartyG-RealSense, 2024-01-31:
You are very welcome.  I'm pleased that I could help.  :)

closed 2024-01-31 by MartyG-RealSense