realsenseai/realsense-ros #2991: Issue using realsense-ros on Raspberry Pi 4

henryburon, 2024-02-03:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                          | 
| Firmware Version                   | 5.15.1                          | 
| Operating System & Version |   Ubuntu 22.04.3 LTS | 
| Kernel Version (Linux Only)    |  5.15.0-1045-raspi                                                                      | 
| Platform                                 | Raspberry Pi 4 Model B 8 GB                           |
| Librealsense SDK Version       |  2.54.1                                                    | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Iron                                           | 
| RealSense ROS Wrapper Version   |  4.54.1                                                | 

### Issue Description

Unable to launch realsense2_camera on Raspberry Pi 4 B.

### Details

I followed these instruction for installing librealsense: https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide
and I believe I was able to install it correctly (I didn't see any errors).

I am able to successfully use ```realsense-viewer``` on the RPi and launch the Realsense Viewer with the video feed.

To install the ROS wrapper, I used: ```apt install ros-iron-realsense2-*```

However, when I run ```ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x15 pointcloud.enable:=true``` I get several errors and I am not able to view the image from the camera in RViz.

### Error

terra@rpi:~$ ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x15 pointcloud.enable:=true

[INFO] [launch]: All log files can be found below /home/terra/.ros/log/2024-02-03-16-39-49-143023-rpi-4487
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [4490]
[realsense2_camera_node-1] [INFO] [1706999989.814240144] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1706999989.814543067] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1706999989.814613659] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1]  03/02 16:40:03,415 ERROR [281472588177856] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  03/02 16:40:03,482 WARNING [281472588177856] (rs.cpp:312) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1707000003.482951324] [camera.camera]: Device 1/7 failed with exception: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  03/02 16:40:03,501 ERROR [281472588177856] (librealsense-exception.h:52) /dev/video10 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  03/02 16:40:03,501 ERROR [281472588177856] (sensor.cpp:661) acquire_power failed: /dev/video10 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  03/02 16:40:03,584 WARNING [281472588177856] (rs.cpp:312) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1707000003.584946812] [camera.camera]: Device 2/7 failed with exception: /dev/video10 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  03/02 16:40:03,605 ERROR [281472588177856] (librealsense-exception.h:52) /dev/video12 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  03/02 16:40:03,605 ERROR [281472588177856] (sensor.cpp:661) acquire_power failed: /dev/video12 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  03/02 16:40:03,686 WARNING [281472588177856] (rs.cpp:312) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1707000003.687035486] [camera.camera]: Device 3/7 failed with exception: /dev/video12 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  03/02 16:40:03,707 ERROR [281472588177856] (librealsense-exception.h:52) /dev/video18 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  03/02 16:40:03,707 ERROR [281472588177856] (sensor.cpp:661) acquire_power failed: /dev/video18 is no video capture device Last Error: Invalid argument
[realsense2_camera_node-1]  03/02 16:40:03,788 WARNING [281472588177856] (rs.cpp:312) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1707000003.788820829] [camera.camera]: Device 4/7 failed with exception: /dev/video18 is no video capture device Last Error: Invalid argument
[ERROR] [realsense2_camera_node-1]: process has died [pid 4490, exit code -11, cmd '/opt/ros/iron/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_b3wgdj_1 --params-file /tmp/launch_params_vfoxek1l'].

Let me know if there is any other information that I can provide. Thank you!

MartyG-RealSense, 2024-02-04:
Hi @henryburon  As you apparently installed librealsense from source code first using the guide that you linked to before you installed the ROS wrapper with **apt install**, it is possible that you have two conflicting installations of librealsense on your computer that is negatively affecting the ROS wrapper launch.

Please try uninstalling your source code build of librealsense by going to the **build** directory of the source code folder and inputting the command **sudo make uninstall && make clean** to uninstall the source code build of librealsense and clean up CMake without starting a new SDK installation after the clean-up.

Are you able to launch the ROS2 wrapper after performing the above librealsense source code uninstall process, please?

henryburon, 2024-02-05:
Hi Marty, thanks for your reply. I ran the uninstall command, but am still getting several errors when I launch the ROS2 wrapper.

# Error

terra@rpi:~$ ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x15 pointcloud.enable:=true

[INFO] [launch]: All log files can be found below /home/terra/.ros/log/2024-02-05-14-05-05-501808-rpi-3122
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [3125]
[realsense2_camera_node-1] [INFO] [1707163553.557459175] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1707163553.557791619] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1707163553.557856897] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1707163553.641128451] [camera.camera]: Device with serial number 233522077626 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1707163553.641318488] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1707163553.641375396] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1707163553.642196136] [camera.camera]: Device with port number 1-1.1 was found.
[realsense2_camera_node-1] [INFO] [1707163553.642319080] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1707163553.642376025] [camera.camera]: Device 233522077626 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1707163553.645933616] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1707163553.648947874] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1707163553.649059151] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1707163553.649108188] [camera.camera]: Device Serial No: 233522077626
[realsense2_camera_node-1] [INFO] [1707163553.649153040] [camera.camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1707163553.649198985] [camera.camera]: Device FW version: 5.15.1
[realsense2_camera_node-1] [INFO] [1707163553.649242281] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1707163553.649283892] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1707163553.706744125] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1707163553.713890418] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1707163553.716646750] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1707163553.716999028] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1707163553.763047692] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1707163553.820459147] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1]  05/02 14:05:58,856 ERROR [281472996598208] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1] [INFO] [1707163558.856932528] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1707163558.858036991] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [WARN] [1707163558.867282080] [camera.camera]: 
[realsense2_camera_node-1] [WARN] [1707163558.867469043] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[realsense2_camera_node-1] [INFO] [1707163558.887590536] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1]  05/02 14:06:03,976 ERROR [281472996598208] (librealsense-exception.h:52) xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error
[realsense2_camera_node-1]  05/02 14:06:03,976 ERROR [281472619438528] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1]  05/02 14:06:03,976 ERROR [281472619438528] (error-handling.cpp:96) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1] [ERROR] [1707163563.976473463] [camera.camera]: ./src/rs_node_setup.cpp:370:An exception has been thrown: xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error
[realsense2_camera_node-1] Failed to start device: xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error
[realsense2_camera_node-1] [INFO] [1707163563.977393982] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1707163563.996804402] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1]  05/02 14:06:09,096 ERROR [281472988144064] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[... cut ...]
[ERROR] [realsense2_camera_node-1]: process[realsense2_camera_node-1] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [realsense2_camera_node-1]: sending signal 'SIGKILL' to process[realsense2_camera_node-1]
[ERROR] [realsense2_camera_node-1]: process has died [pid 3125, exit code -9, cmd '/opt/ros/iron/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_v8z3shx5 --params-file /tmp/launch_params_uxb2qhs4'].

MartyG-RealSense, 2024-02-06:
Is there any improvement if you add **initial_reset:=true** onto the end of your `ros2 launch` instruction to reset the camera at launch?

henryburon, 2024-02-06:
No, I get the same error as in my previous reply.

Since I may have two conflicting instances of librealsense, would it be worth it to completely remove both librealsense and the ROS2 wrapper from my RPi, and reinstall (wrapper first, then librealsense with SDK installation)?

In addition, are the instructions at https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide what I should be using? I followed them exactly.

Thank you.

MartyG-RealSense, 2024-02-07:
A clean installation of everything can sometimes solve errors that could not otherwise be resolved, because something that was causing a conflict has been removed.

If the librealsense SDK and ROS wrapper are being installed separately then librealsense should always be installed first and the ROS wrapper second.

The guide that you linked to is intended for x86 / x64 architecture computers such as desktop and laptop PCs rather than the Arm architecture that Raspberry Pi has.

Installing librealsense on Raspberry Pi is typically a difficult process.  The **libuvc backend** build script at the link below usually works well with Pi though.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

henryburon, 2024-02-11:
Hi Marty, thanks for your reply again.

I uninstalled/deleted librealsense and the ros2 wrapper and reinstalled using the instructions you sent. The librealsense installation went smoothly, and I was able to use ```realsense-viewer``` on the RPi and get the live video.

However, I got the same errors when launching the ros2 wrapper:

terra@rpi:~$ ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x15 pointcloud.enable:=true initial_reset:=true
[INFO] [launch]: All log files can be found below /home/terra/.ros/log/2024-02-11-16-07-42-961483-rpi-3668
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [3671]
[realsense2_camera_node-1] [INFO] [1707689310.869618749] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1707689310.869933893] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1707689310.870004244] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1707689310.956751597] [camera.camera]: Device with serial number 233522077626 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1707689310.956953539] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1707689310.957011816] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1707689310.957836473] [camera.camera]: Device with port number 1-1.1 was found.
[realsense2_camera_node-1] [INFO] [1707689310.958004971] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1707689310.958066192] [camera.camera]: Device 233522077626 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1707689310.958112858] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1707689317.101677057] [camera.camera]: Device with serial number 233522077626 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1707689317.101854907] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1707689317.101914462] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1707689317.102658713] [camera.camera]: Device with port number 1-1.1 was found.
[realsense2_camera_node-1] [INFO] [1707689317.102788137] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1707689317.102897247] [camera.camera]: Device 233522077626 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1707689317.106265338] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1707689317.107924208] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1707689317.108044873] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1707689317.108096057] [camera.camera]: Device Serial No: 233522077626
[realsense2_camera_node-1] [INFO] [1707689317.108142205] [camera.camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1707689317.108187242] [camera.camera]: Device FW version: 5.15.1
[realsense2_camera_node-1] [INFO] [1707689317.108230426] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1707689317.108272370] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1707689317.167947997] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1707689317.174947565] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1707689317.177183835] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1707689317.177590071] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1707689317.220299246] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1707689317.276699503] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1]  11/02 16:08:42,314 ERROR [281473139401152] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1] [INFO] [1707689322.315679674] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1707689322.316769624] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [WARN] [1707689322.323241737] [camera.camera]: 
[realsense2_camera_node-1] [WARN] [1707689322.323490975] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[realsense2_camera_node-1] [INFO] [1707689322.347605815] [camera.camera]: Starting Sensor: RGB Camera
[... cut ...]
[realsense2_camera_node-1]  11/02 16:09:07,914 ERROR [281472619438528] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1]  11/02 16:09:07,915 ERROR [281472619438528] (error-handling.cpp:96) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1]  11/02 16:09:13,034 ERROR [281472594076096] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out

MartyG-RealSense, 2024-02-12:
Are you able to test with ROS2 Humble instead of Iron to confirm whether or not Iron is the cause of the conflict?

garyservin, 2024-02-16:
I'm running ROS 2 Humble on a RPI4 with the debian packages and I see this same issue

MartyG-RealSense, 2024-02-17:
Thanks very much @garyservin for your report.

A Pi 4 user in November 2023 at https://github.com/IntelRealSense/realsense-ros/issues/2545#issuecomment-1815703525 had the same errors after they built librealsense from source code in RSUSB = true mode.  They tried with SDK 2.54.1 and 2.54.2 and experienced the error with both.

The librealsense SDK can be difficult to get working on Raspberry Pi boards and even if a working librealsense build is achieved then the ROS wrapper is likely to have problems, unfortunately.

henryburon, 2024-02-17:
Thanks for your responses, Marty.

I think the solution is to write a custom ROS2 node that fetches the image data using the librealsense API, and then publishes it on topics so its available for use throughout the project.

MartyG-RealSense, 2024-02-23:
Hi @henryburon  Do you require further assistance with this case, please?  Thanks!

henryburon, 2024-02-23:
No further assistance required, thank you!

MartyG-RealSense, 2024-02-24:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks!

closed 2024-02-24 by MartyG-RealSense

SuperNik17, 2024-03-12:
> Thanks for your responses, Marty.
> 
> I think the solution is to write a custom ROS2 node that fetches the image data using the librealsense API, and then publishes it on topics so its available for use throughout the project.

I've been having a lot of issues trying to get things working smoothly. The Raspberry Pi just doesn't seem efficient for what I'm trying to do. Right now, I'm using the DDS network to get the Realsense camera working on a companion computer and to publish the data across the network.

heyichen2333, 2024-05-01:
> Thanks for your responses, Marty.
> 
> I think the solution is to write a custom ROS2 node that fetches the image data using the librealsense API, and then publishes it on topics so its available for use throughout the project.

May I know which api did you use to solve this error? Here are the two api I used to stream color and depth and they still don't work.
enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30)
enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);

heyichen2333, 2024-05-01:
> If you are attempting to do more than stream depth and color with a RealSense camera then you may experience issues when doing so with Pi boards

may I know which API can safely launch the color stream and depth stream?