realsenseai/realsense-ros #2992: slam_toolbox with IntelRealsense publishing empty /map topic

lujan002, 2024-02-04:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455                          | 
| Firmware Version                   | 5.15.1                          | 
| Operating System & Version | Linux (Ubuntu 20.04.3 | 
| Kernel Version (Linux Only)    | 5.15.0-92-generic                                                                       | 
| Platform                                 | PC                        |
| Librealsense SDK Version       |   2.54.2                                                     | 
| Language                               |  n/a         | 
| Segment			         |  Robot                        | 
| ROS Distro			         |  Humble                                           | 
| RealSense ROS Wrapper Version   |  4.54.1                                                  | 

### Issue Description
Hello, 

I will preface that I am fairly new to ROS. I have been trying to use slam_toolbox using my IntelRealsense D455 depth camera in Ros2 Humble, but /map topic output is empty, which I beleive stems from the fact that either my camera or /tf topic is failing to publish an /odom topic, which slam_toolbox expects. 

My situation is similar to this user's: https://support.intelrealsense.com/hc/en-us/community/posts/19599334779283--D455-SLAM-using-Ros2 and issue #2387. 

I am using depthimage_to_laserscan to convert the camera's depth map to a /scan topic. My understanding is that slam_toolbox expects a /scan, /odom, and /tf topic, and since my /scan and /tf topics appear to be working below, I reason that the issue is to do with the lack of an /odom topic.

I can confirm that the /camera/imu topic is being published. **How can I "convert" this /imu topic to an /odom topic?** (Is this even the right approach?). 

My end goal is to incorporate the camera with a Clearpath Jackal Rover, which I imagine might be able to provide an /odom topic, but I imagine that there should be a way to get the realsense to publish it's own /odom topic for SLAM independet of a robot. 

If there is a SLAM package which works better with realsense and Ros2, I would love to know. 

ros2 topic list returns:

    /camera/accel/imu_info
    /camera/accel/metadata
    /camera/accel/sample
    /camera/aligned_depth_to_color/camera_info
    /camera/aligned_depth_to_color/image_raw
    /camera/aligned_depth_to_color/image_raw/compressed
    /camera/aligned_depth_to_color/image_raw/compressedDepth
    /camera/aligned_depth_to_color/image_raw/theora
    /camera/color/camera_info
    /camera/color/image_raw
    /camera/color/image_raw/compressed
    /camera/color/image_raw/compressedDepth
    /camera/color/image_raw/theora
    /camera/color/metadata
    /camera/depth/camera_info
    /camera/depth/color/points
    /camera/depth/image_rect_raw
    /camera/depth/image_rect_raw/compressed
    /camera/depth/image_rect_raw/compressedDepth
    /camera/depth/image_rect_raw/theora
    /camera/depth/metadata
    /camera/extrinsics/depth_to_accel
    /camera/extrinsics/depth_to_color
    /camera/extrinsics/depth_to_depth
    /camera/extrinsics/depth_to_gyro
    /camera/gyro/imu_info
    /camera/gyro/metadata
    /camera/gyro/sample
    /camera/imu
    /clicked_point
    /clock
    /goal_pose
    /initialpose
    /map
    /map_metadata
    /map_updates
    /parameter_events
    /pose
    /rosout
    /scan
    /slam_toolbox/feedback
    /slam_toolbox/graph_visualization
    /slam_toolbox/scan_visualization
    /slam_toolbox/update
    /tf
    /tf_static`

Rviz says "no map received"
[image]

rqt graph
[image]

Closer look at the TF
[image]

MartyG-RealSense, 2024-02-04:
Hi @lujan002  slam_toolbox and depthimage_to_laserscan are difficult to get working together in ROS2, unfortunately.

If possible to do so then I would recommend instead using  **ORB-SLAM** with ROS2 and running it in **Stereo** mode with your D455 stereo depth camera.

https://github.com/zang09/ORB_SLAM3_ROS2/tree/humble

MartyG-RealSense, 2024-02-13:
Hi @lujan002  Do you require further assistance with this case, please?  Thanks!

lujan002, 2024-02-13:
Hi Marty, I am having some issues setting up orb_slam_3. After following the instructions in the link you sent, I get this error when trying to build: 

CMake Error at CMakeLists.txt:23 (find_package):
  By not providing "FindSophus.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Sophus", but
  CMake did not find one.

  Could not find a package configuration file provided by "Sophus" with any
  of the following names:

    SophusConfig.cmake
    sophus-config.cmake

  Add the installation prefix of "Sophus" to CMAKE_PREFIX_PATH or set
  "Sophus_DIR" to a directory containing one of the above files.  If "Sophus"
  provides a separate development package or SDK, be sure it has been
  installed.

Not sure if this is your area of expertise, but any guidance would be helpful.

MartyG-RealSense, 2024-02-14:
Hi @lujan002  I carefully researched your problem with Sophus but there was very little available information about it unfortunately.

If Sophus could not be found after following the installation guide's troubleshooting advice:

https://github.com/zang09/ORB_SLAM3_ROS2/tree/humble?tab=readme-ov-file#troubleshootings

One user took the approach of installing Sophus by performing a **git clone** of its GitHub repository.

https://github.com/strasdat/Sophus

lujan002, 2024-02-20:
Hi Marty, tbh I haven't touched this issue for some time, I will let you know if I need assistance in the future.

MartyG-RealSense, 2024-02-20:
Okay, thanks for the update.  I will close the issue for now as you are not currently working on it.  Please feel free to re-open the issue at a future date if you need to or create a new issue.  Thanks again!