realsenseai/realsense-ros #2995: QOS Setting Pointcloud

anath93, 2024-02-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D455 }                          | 
| Firmware Version                   | 5.15.1                          | 
| Operating System & Version |   { Ubuntu 20.04 } | 
| Kernel Version (Linux Only)    | 5.15                                                                     | 
| Platform                                 | Jetson AGX Orin                           |
| Segment			         |  {Robot }                             | 
| ROS Distro			         |  Galactic                                           | 
| RealSense ROS Wrapper Version   |  4.54.1                                                 | 

### Issue Description
Pointcloud QOS not getting applied:

I looked into previous issues, unable to change the QoS on pointcloud data
[image]

[image]

https://github.com/IntelRealSense/realsense-ros/issues/1827

MartyG-RealSense, 2024-02-05:
Hi @anath93  Can the QoS mode be successfully set to a mode other than SENSOR_DATA by defining it in the launch instruction if you are not doing so already, please?

`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true depth_qos:=SYSTEM_DEFAULT`

anath93, 2024-02-05:
@MartyG-RealSense I tried that couple days ago, it does not work by setting in the launch file.

MartyG-RealSense, 2024-02-05:
And it also does not change the setting if you launch with **ros2 run** instead of ros2 launch?

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true -p depth_qos:=SYSTEM_DEFAULT`

anath93, 2024-02-05:
@MartyG-RealSense  No it does not.

anath93, 2024-02-05:
Yes that solves the issue, its just a simple bug. Manually changing in the .h file and building the workspace works. But would be better if that could be linked to 'depth_qos' .

anath93, 2024-02-06:
I see this is under ros2-development, but I am on 4.54.1, before I try and spend time on this, is this going to work for 4.54.1 ? meaning is there yaml parser in the source code side ?

MartyG-RealSense, 2024-02-06:
ros2_development is the name of the Master branch of the RealSense ROS2 wrapper.

Older versions of the wrapper already support the import of yaml configuration files.

anath93, 2024-02-08:
@MartyG-RealSense for 4.54.1, I got this info and maybe helpful from the person I am working with over NDA, the parameter is 'pointcloud.pointcloud_qos' for pointcloud qos.

closed 2024-02-08 by anath93

MartyG-RealSense, 2024-02-08:
@anath93  Thanks so much for sharing!