realsenseai/realsense-ros #2997: "Segmentation fault (core dumped)" installing sdk and wrapper from ros server

MFHuniovi, 2024-02-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | {D435i }                          | 
| Firmware Version                   | (5.15.1)                          | 
| Operating System & Version |   { Linux (Ubuntu  22.04.3 LTS) } | 
| Kernel Version (Linux Only)    |  ( 5.15.71)                                                                       | 
| Platform                                 | DEBIX Model A Industrial Single Board Computer 8gb                          |
| Librealsense SDK Version       |  { 2.54.1-1jammy.20240125.204242 }                                                    | 
| Language                               |  {}         | 
| Segment			         |  {Robot}                             | 
| ROS Distro			         |  {Humble }                                           | 
| RealSense ROS Wrapper Version   |  { 4.54.1-1jammy.20240126}                                                  | 

### Issue Description
I have installed from ros servers the sdk and the wrapper in principle correctly since no error came out but then any command I do always gives the same result "segmention fault (core dumped)".
"segmention fault
"error"
[image]

installation
[image]

MartyG-RealSense, 2024-02-06:
Hi @MFHuniovi  Thanks very much for the update.  Have you enabled the gyro and accel topics (which are disabled by default) and combined them together into a single `/imu `topic using the `unite_imu_method` parameter, please?

ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_gyro:=true -p enable_accel:=true -p unite_imu_method:=2

MFHuniovi, 2024-02-06:
Sorry I didn't put the command I use, it is the following: 
ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=false enable_accel:=true enable_gyro:=true unite_imu_method:=2 diagnostics_period:=3.141596
 I have enabled these two calculations because to calculate an odom I am using the robot_pose_ekf and I am passing it a single imu
 
 From the camera I would only need the imuand a lens to see in black and white for now, can you help me with the parameters? Is there any documentation that explains more in detail the parameters and if they affect more or less the performance of the cpu and memory?

MartyG-RealSense, 2024-02-13:
Hi @MFHuniovi  Do you require further assistance with this case, please?  Thanks!

MFHuniovi, 2024-02-19:
Hello, no thanks :D